espressif/esp-idf #2943: Feature Request : MCPWM module sync trigger on rising edge OR falling edge only

AadiMehta, 2019-01-13:
Hi,

Can we have functionality to trigger timer only in case of rising edge(or falling edge, any one) on sync pin and not both.

Thanks,

https://esp32.com/viewtopic.php?f=2&t=6033&p=26085&hilit=sync+pin#p26085

koobest, 2019-01-20:
Hi,@AadiMehta
I'm sorry for the later reply.  I have seen the image you posted in `viewtopic`. Can you explain how you want to use mcpwm?  It's a bit hard to guess  :- ). and what is the timer's counter mode, Count-Up, Count-Down and Up-Dowm?

koobest, 2019-01-20:
Hi,@AadiMehta

Thanks for you clear explanation, I understand now. I have also referenced to the TRM, I think the sync module is triggered on the rising edge. but due to the sync signal is 50Hz, the edge (H->L) of the waveform is very slow, the fluctuations between the critical levels of logic 0 and logic 1 cause the sync module triggered by mistake. this is my guess, of course, and I'll confirm it tomorrow. If you have a logic analyzer, you can try to parse the sync signal.

thanks !!

[image]

AadiMehta, 2019-01-20:
@koobest I dont think this is being caused due to fluctuations, but I will recheck for sure. Also I have pulled down the sync pin using **gpio_pulldown_en** function. But thanks for putting your time on this. Really appreciate it.

Currently to overcome the massive fluctuations I have used a cheat(there is still some flickering though). I know the duty cycle of sync signal(time length between rising edge and falling edge) as recorded using oscilloscope. I reduce the frequency to overcome timer reset at falling edge.

AadiMehta, 2019-02-19:
@koobest Sorry for the late reply. I have checked output of zero crossing(sync pin) again and its stable, giving about 100 pulses per second(2 zero crossing per pulse of 50Hz frequency).

Attaching with this comment various screenshots of DSO. Yellow channel is for PWM_OUT and red channel is SYNC INPUT.

for 100%
[image]

for 50%
[image]

for 25%
[image]

NOTE : I see in above case of 25% Output frequency has changed to ~100 Hz from ~50 Hz

AadiMehta, 2019-02-25:
Hey @koobest , any updates on the issue? Also please let me know if you require more info

AadiMehta, 2019-03-05:
Hi @koobest , any updates on this issue?

AadiMehta, 2019-03-22:
Also here is my general circuit:
[image]

AadiMehta, 2019-03-22:
@koobest @projectgus @david-cermak Can you please look into this issue related to MCPWM or redirect it to appropriate person?

Thanks

AadiMehta, 2019-03-25:
@david-cermak @projectgus Any updates on this issue?

koobest, 2019-03-26:
Hi, @AadiMehta

Sorry for the late reply, we are focusing on this issue and we need some more infos:
1.  Is the Yellow channel  your circuit output? 
2. The circuit is empty or loaded when you do this test?
3. Can you provide the output of the ESP32 MCPWM when the problem occurs (Timer reset is not normal)?

 thanks !!

AadiMehta, 2019-03-26:
@koobest 

> Is the Yellow channel your circuit output?

Its the PWM output which is connected to load

> The circuit is empty or loaded when you do this test?

Load is connected in this instance

> Can you provide the output of the ESP32 MCPWM when the problem occurs (Timer reset is not normal)?

Above DSO images corresponds to the problem. It occurs all the time, there is flickering on the output as seen on the DSO images above(reason for this flickering is not clear to me).

Here is my code for the mcpwm. 
[pwm.zip](https://github.com/espressif/esp-idf/files/3006762/pwm.zip)

negativekelvin, 2019-03-26:
Why are you using symmetric (up/down)?

koobest, 2019-03-26:
@AadiMehta,
I'm not sure what went wrong, can you please use the attachment to do a test? it use Software reset method.  I want to know is there still a problem?

thanks !!

[pwm.tar.gz](https://github.com/espressif/esp-idf/files/3007962/pwm.tar.gz)

AadiMehta, 2019-03-26:
@negativekelvin I have tried with all 3 counter modes, and UpDown counter mode is giving least flicker.

negativekelvin, 2019-03-26:
That sounds very non-scientific.  I think you want count up because you don't want the pulse centered you want it at the end.

AadiMehta, 2019-03-30:
@koobest Changing to software sync also doesn't resolve the issue. We are trying to create phasing based dimmers on top of ESP32. It still results in same visible flickering as earlier. This flickering is clearly visible for <75% duty cycle 

Please let me know what else we can try?

negativekelvin, 2019-03-30:
Show the PWM triac input signal on the scope together with the triac output

koobest, 2019-03-31:
@AadiMehta,
Please try the attachment again and check if it  behaver a little better.

thanks !!

[pwm.zip](https://github.com/espressif/esp-idf/files/3026147/pwm.zip)

koobest, 2019-04-11:
@AadiMehta 
Did you solve the problem with my last code? I want to know if it is a hardware issue.

thanks !!

AadiMehta, 2019-04-11:
Nope, there is very little improvement. The flicker is still very much present. I have tried capturing the sync signal using capture module of the mcpwm. This indicates that the zero crossing circuit is working fine.

```
I (64550) ESP32Project.gpio: CAP0 : 9875 us

I (64560) ESP32Project.gpio: CAP0 : 9875 us

I (64570) ESP32Project.gpio: CAP0 : 9875 us

I (64580) ESP32Project.gpio: CAP0 : 9875 us

I (64590) ESP32Project.gpio: CAP0 : 9875 us

I (64600) ESP32Project.gpio: CAP0 : 9875 us

I (64610) ESP32Project.gpio: CAP0 : 9875 us

I (64620) ESP32Project.gpio: CAP0 : 9875 us

I (64630) ESP32Project.gpio: CAP0 : 9875 us

I (64640) ESP32Project.gpio: CAP0 : 9875 us

I (64650) ESP32Project.gpio: CAP0 : 9875 us

I (64652) ESP32Project.gpio: CAP0 : 1125 us

I (64660) ESP32Project.gpio: CAP0 : 8750 us

I (64669) ESP32Project.gpio: CAP0 : 9875 us

I (64672) ESP32Project.gpio: CAP0 : 1125 us

I (64678) ESP32Project.gpio: CAP0 : 8750 us

I (64688) ESP32Project.gpio: CAP0 : 9875 us

I (64698) ESP32Project.gpio: CAP0 : 9875 us

I (64708) ESP32Project.gpio: CAP0 : 9875 us

I (64718) ESP32Project.gpio: CAP0 : 9875 us

I (64728) ESP32Project.gpio: CAP0 : 9875 us

I (64738) ESP32Project.gpio: CAP0 : 9875 us

I (64747) ESP32Project.gpio: CAP0 : 9875 us

I (64749) ESP32Project.gpio: CAP0 : 1125 us

I (64757) ESP32Project.gpio: CAP0 : 8750 us

I (64766) ESP32Project.gpio: CAP0 : 9875 us

I (64774) ESP32Project.gpio: CAP0 : 9875 us

I (64783) ESP32Project.gpio: CAP0 : 9875 us

```

koobest, 2019-04-12:
@AadiMehta
Which GPIOs are used?

AadiMehta, 2019-04-12:
@koobest We are using 16, 17, 21, 26, 32 & 33

AadiMehta, 2019-04-12:
@koobest Changing the direction of interrupt will increase the flicker, as the timer will reset at wrong timings in that case. Please let me know if you still want me to try that?

negativekelvin, 2019-04-13:
I would still like to see pwm output overlay with ac output waveform.  Also you didn't say whether your load is incandescent or led and whether you have any snubber circuit.

AadiMehta, 2019-04-19:
@negativekelvin load is incandescent bulb which is a small load(20W bulb).

markiv25, 2019-08-14:
@AadiMehta hey I am too stuck  with same problem did u get the solution?

ageurtse, 2019-08-31:
working on a same kind of dimmer, here also some flickering. 
see youtube movie to see flickering and to see scope image. 

[https://www.youtube.com/watch?v=79n3DwduI9w&feature=youtu.be](Youtube movie)
the lower part of the scope screen is the ZCD
the upper part is the MCPWM output, as you can see it is not a stable signal. 
And i can't figure out why not. 

Below is the code written in arduino:
```
#include "driver/mcpwm.h"

#define GPIO_SYNC0_IN 23       //pin for the external interrupt (Input with Pullup)                         
#define MOSFET 22          //pin to trigger the MOSFET (Output) pin22

void setup() {
  // put your setup code here, to run once:
  mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0A, MOSFET);

  mcpwm_pin_config_t pin_config; 
    pin_config.mcpwm_sync0_in_num  = GPIO_SYNC0_IN;
      
  mcpwm_config_t pwm_config;
    pwm_config.frequency = 60;    //frequency = 110Hz
    pwm_config.cmpr_a = 100;       //duty cycle of PWMxA = 60.0%
    pwm_config.counter_mode = MCPWM_UP_COUNTER;
    pwm_config.duty_mode = MCPWM_DUTY_MODE_0;
    
  mcpwm_set_pin(MCPWM_UNIT_0, &pin_config);
  mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config);   //Configure PWM0A & PWM0B with above settings
  mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, MCPWM_DUTY_MODE_0);
  mcpwm_start(MCPWM_UNIT_0, MCPWM_TIMER_0);
  mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_SELECT_SYNC0, 0);

  mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A,25);
}

void loop() {

}
```

markiv25, 2019-09-13:
but i have tried it with Mega using Interrupt its working flawless ,so i donts think its cuz of the noise!,its something else

negativekelvin, 2019-09-14:
https://esp32.com/viewtopic.php?f=19&p=49211#p49211

markiv25, 2019-09-24:
negative nothing helped

markiv25, 2019-09-25:
yes i have changed the ic

markiv25, 2019-09-25:
i am using 4n35

negativekelvin, 2019-09-25:
4n35 has transistor output not logic/schmitt

markiv25, 2019-09-25:
yes but the output is smooth

markiv25, 2019-09-25:
[image]
[image]

markiv25, 2019-09-25:
this circuit

wilhelmw201, 2020-06-22:
hi,
There will be an option to do so, but as of now you can OR the two flags to configure the MCPWM to capture on both edges. Use something like  `mcpwm_capture_enable(MCPWM_UNIT_0, MCPWM_SELECT_CAP0, MCPWM_POS_EDGE|MCPWM_NEG_EDGE, 0);`

closed 2020-08-20 by projectgus

bot1131357, 2020-09-12:
Hi @markiv25 
Based on the circuit you have posted, isn't DP2 getting a falling edge at zero crossing? 

As far as I can tell MCPWM SYNC only detects the rising edge. Did you have to [modify mcpwm.c](https://github.com/espressif/esp-idf/issues/2943#issuecomment-482465890) as suggested by @koobest ?

bot1131357, 2020-09-12:
Also, I believe is now:
 
`change gpio_matrix_in(gpio_num, ..., 0); to gpio_matrix_in(gpio_num, ..., 1);`
should be 
`change esp_rom_gpio_connect_in_signal(gpio_num, ..., 0); to esp_rom_gpio_connect_in_signal(gpio_num, ..., 1);`

```
/**
 * @brief Combine a GPIO input with a peripheral signal, which tagged as input attribute.
 *
 * @note There's no limitation on the number of signals that a GPIO can combine with.
 *
 * @param gpio_num GPIO number, especially, `GPIO_MATRIX_CONST_ZERO_INPUT` means connect logic 0 to signal
 *                                          `GPIO_MATRIX_CONST_ONE_INPUT` means connect logic 1 to signal
 * @param signal_idx Peripheral signal index (tagged as input attribute)
 * @param inv  Whether the GPIO input to be inverted or not
 */
void esp_rom_gpio_connect_in_signal(uint32_t gpio_num, uint32_t signal_idx, bool inv);
```

Is this currently the only way to allow detection on falling edge? By changing the source code in esp-idf?

elementgreen, 2021-09-24:
I know this is an old issue and closed, but it was one of the top results I kept coming to when searching for a solution for this same problem, so I thought I'd post my solution in case it helps anyone. In my case, it was indeed triggering on rising edge even when configured for falling edge only, by using mcpwm_sync_invert_gpio_synchro(), because of a noisy rising edge (chatter). The solution I settled on was to use the MCPWM capture subsystem instead of directly synchronizing on the GPIO zero crossing input. The capture ISR callback can then keep track of the last synchronization timestamp and reject additional captures which are still within the zero crossing pulse. The synchronization feature is still used to determine the phase shift of the triac pulse (and thus the duty cycle), but it is triggered by software within the capture ISR callback. I have not actually tested this yet with a real triac circuit, but my scope shows the PWM stable and locked on the zero crossing input. I haven't checked to see how many captures occur, which could impact system performance. I think the capture prescaler might be usable for limiting this though.

```
#include "soc/mcpwm_periph.h"
#include "driver/mcpwm.h"
#include "hal/misc.h"
#include "hal/mcpwm_ll.h"
#include "hal/mcpwm_hal.h"

#define ZERO_CROSS_GPIO         N       // Set this to the GPIO of the zero crossing input
#define OUTPUT_GPIO                   N       // Set this to the GPIO of the PWM output
#define SYNC_LOCKOUT_TICKS   (2 * 80000)     // Sync pulse duration lockout. 2ms in 80 MHz APB clock ticks
#define MIN_DUTY                          100   // Minimum duty in 10ths of a percent (100 = 10%)
#define MAX_DUTY                         900   // Maximum duty in 10ths of a percent (900 = 90%)

static volatile uint32_t acDimmerLastSyncTime;

static bool
ac_dimmer_capture_intr (mcpwm_unit_t mcpwm, mcpwm_capture_channel_id_t cap_channel,
                                         const cap_event_data_t *edata, void *user_data)
{
  mcpwm_dev_t *dev = MCPWM_LL_GET_HW (0);

  if (edata->cap_value - acDimmerLastSyncTime > SYNC_LOCKOUT_TICKS)
  {
    mcpwm_ll_timer_trigger_soft_sync (dev, MCPWM_TIMER_0);
    acDimmerLastSyncTime = edata->cap_value;
  }

  return false;
}

void
ac_dimmer_init (void)
{
  mcpwm_config_t pwm_config =
  {
    .frequency = 2 * 60,    // AC frequency * 2
    .cmpr_a = 1.0,            // Duty cycle of triac pulse in percent
    .cmpr_b = 0.0,
    .counter_mode = MCPWM_UP_COUNTER,
    .duty_mode = MCPWM_DUTY_MODE_0,
  };

  mcpwm_capture_config_t cap_conf =
  {
    .cap_edge = MCPWM_NEG_EDGE,
    .cap_prescale = 1,
    .capture_cb = ac_dimmer_capture_intr,
    .user_data = NULL,
  };

  mcpwm_init (MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config);

  // Start with PWM output disabled (set generator B output to low and assign to PWM output)
  mcpwm_set_signal_low (MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_GEN_B);
  mcpwm_gpio_init (MCPWM_UNIT_0, MCPWM0B, OUTPUT_GPIO);

  mcpwm_gpio_init (MCPWM_UNIT_0, MCPWM_CAP_0, ZERO_CROSS_GPIO);
  mcpwm_capture_enable_channel (MCPWM_UNIT_0, MCPWM_SELECT_CAP0, &cap_conf);
}

void
ac_dimmer_set_duty (int duty)
{
  mcpwm_sync_config_t sync_conf =
  {
    .sync_sig = MCPWM_SELECT_GPIO_SYNC0,
    .count_direction = MCPWM_TIMER_DIRECTION_UP,
  };

  if (duty <= MIN_DUTY)
  {
    mcpwm_set_signal_low (MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_GEN_B);
    mcpwm_gpio_init (MCPWM_UNIT_0, MCPWM0B, OUTPUT_GPIO);
  }
  else if (duty >= MAX_DUTY)
  {
    mcpwm_set_signal_high (MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_GEN_B);
    mcpwm_gpio_init (MCPWM_UNIT_0, MCPWM0B, OUTPUT_GPIO);
  }
  else
  {
    sync_conf.timer_val = duty;
    mcpwm_sync_configure (MCPWM_UNIT_0, MCPWM_TIMER_0, &sync_conf);
    mcpwm_gpio_init (MCPWM_UNIT_0, MCPWM0A, OUTPUT_GPIO);
  }
}
```

CC-MASTER05, 2023-03-13:
HI @elementgreen  can you please share whole code with your main function it will help up.
Thanks