espressif/esp-idf #2945: Undefined reference to poll

raveious, 2019-01-14:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 
    v3.2-beta1
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

I've noticed that there is a `poll.h` header in `newlib/platform_include/sys`, with no implementation. So just in case I missed it, I tried using it by taking the UART Select Example, and updating it to use `poll` instead of `select`, and I'm seeing an undefined reference to it. Am I missing something or is the implementation legitimately missing?

dobairoland, 2019-02-20:
Yes, you are right @raveious. I'm working on the implementation now.

closed 2019-02-26 by igrr

dobairoland, 2019-02-26:
Hello @raveious. This is now implemented in the master branch. You can test it there. The implementation will be backported to v3.2 ASAP.