espressif/esp-idf #2947: Q: app_main

grahamehorner, 2019-01-14:
Is/will it be possible to allow app_main to become a int app_main(int argc, char* argv[]) I have as I have a need to pass parameters into the user code, these parameters could/would be by way of the call function (kernel code) and be values from a file config.sys for example.  NOTE the file and values wouldn’t be accessible for the user code directly.

igrr, 2019-01-14:
Could you please explain in a bit more detail, what you would like to achieve? From the perspective of IDF, all the application minus the components provided by ESP-IDF constitutes "user code", do you mean something different by that? Where is the config.sys file located? What code would be responsible for obtaining arguments from a file and passing them to app_main?

grahamehorner, 2019-01-14:
@igrr I’ve traced the calls back to the caller of the app_main(void) the caller seems to accept void *args but doesn’t do anything with these, I was thinking it would be good to pass these up to the app_main if possible?

In my use case I need to secure some data that is contained in the boot partition, and partial accessibile to the app_main by way and args; not sure if this the correct way or if there are other options of securing, hiding sensitive configuration data?

igrr, 2019-01-14:
By boot partition, do you mean the bootloader?

The data which is contained in, or is available to the bootloader will also, in principle, be available to the application. This is because the application has access to all of the flash and RAM.

Also I do not entirely understand the model where you want the application to not have access to certain data, and at the same time pass the same data to the application via argv arguments.

I believe this may be an [XY problem](https://meta.stackexchange.com/a/66378). Could you please explain what you intend to do in the first place?

grahamehorner, 2019-01-14:
@igrr my use case I require to have the device have a security key/secret configured at time of flashing, the key shouldn’t be accessible to anyone but the application at runtime, the application will using the security key and data obtained from a service create a access token by which the application will be allowed to communicate.

igrr, 2019-01-19:
@grahamehorner This use case (storing keys/secrets at production time in a secure fashion) is supported by the combination of [flash encryption](https://docs.espressif.com/projects/esp-idf/en/latest/security/flash-encryption.html) and [NVS encryption](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_flash.html#nvs-encryption) features.
To generate NVS partition at production time, use the [NVS partition generator utility](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_partition_gen.html) or the [mass manufacturing utility](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/mass_mfg.html).

closed 2019-02-26 by igrr