espressif/esp-idf #2949: inform new client ip4 address during SYSTEM_EVENT_AP_STAIPASSIGNED

123swk123, 2019-01-15:
in the soft AP mode, when a new client connects SYSTEM_EVENT_AP_STAIPASSIGNED is raised and it would be good if it also has additional information about which IP address it has assigned to its new client in the event information.

some thing like below.
```cpp
    89	static void tcpip_adapter_dhcps_cb(u8_t client_ip[4])
    90	{
    91	    ESP_LOGI(TAG,"softAP assign IP to station,IP is: %d.%d.%d.%d",
    92	                client_ip[0],client_ip[1],client_ip[2],client_ip[3]);
    93	    system_event_t evt;
    94	    evt.event_id = SYSTEM_EVENT_AP_STAIPASSIGNED;
    95	    memcpy((char *) &evt.event_info.got_ip.ip_info.ip.addr, (char *) client_ip, sizeof(evt.event_info.got_ip.ip_info.ip.addr));
    96	    esp_event_send(&evt);
    97	}

```

TimXia, 2019-01-15:
Sorry， I can't catch you. Isn't IP address in the event information?

123swk123, 2019-01-15:
Yes in current implementation it just sends empty event, the modified code is proposed method

https://github.com/espressif/esp-idf/issues/2949#issuecomment-454297172

zhangyanjiaoesp, 2019-01-15:
@123swk123  Thanks, it's a good idea. We can add it in v4.0

closed 2019-03-13 by igrr