espressif/esp-idf #2950: wifi: unstable conection to AP on mobile phoneh

dibalavs, 2019-01-15:
Hi,
I open AP on my phone (Sony Xperia XZ1 (Android 8)) and try to connect ESP32 chip (rev 1) to my phone.

I use ESP-IDF 3.1.  Connection is very unstable, ESP chip connected only after number attempts. 
(My notebook and other devices are connected successfully)

this is code which I use to connection. No magic, just copy&paste from example:
 [GIST](https://gist.github.com/dibalavs/78dc4c833636e910a4ffe6cdd983607a) 

## Debug Logs

```
I (31) boot: ESP-IDF v3.1-91-gf11ac037c 2nd stage bootloader
I (31) boot: compile time 23:01:45
I (39) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1c3a4 (115620) map
I (135) esp_image: segment 1: paddr=0x0002c3cc vaddr=0x3ffb0000 size=0x03594 ( 13716) load
I (141) esp_image: segment 2: paddr=0x0002f968 vaddr=0x3ffb3594 size=0x00000 (     0) load
I (142) esp_image: segment 3: paddr=0x0002f970 vaddr=0x40080000 size=0x00400 (  1024) load
I (151) esp_image: segment 4: paddr=0x0002fd78 vaddr=0x40080400 size=0x00298 (   664) load
I (160) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x7e544 (517444) map
I (350) esp_image: segment 6: paddr=0x000ae564 vaddr=0x40080698 size=0x110bc ( 69820) load
I (379) esp_image: segment 7: paddr=0x000bf628 vaddr=0x400c0000 size=0x00418 (  1048) load
I (380) esp_image: segment 8: paddr=0x000bfa48 vaddr=0x50000000 size=0x001a0 (   416) load
I (397) boot: Loaded app from partition at offset 0x10000
I (397) boot: Disabling RNG early entropy source...
I (398) cpu_start: Pro cpu up.
I (401) cpu_start: Starting app cpu, entry point is 0x400810a0
I (0) cpu_start: App cpu up.
I (412) heap_init: Initializing. RAM available for dynamic allocation:
D (419) heap_init: New heap initialised at 0x3ffae6e0
I (424) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (430) heap_init: New heap initialised at 0x3ffb8ff8
I (435) heap_init: At 3FFB8FF8 len 00027008 (156 KiB): DRAM
I (441) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (447) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (454) heap_init: New heap initialised at 0x40091754
I (459) heap_init: At 40091754 len 0000E8AC (58 KiB): IRAM
I (465) cpu_start: Pro cpu start user code
D (497) clk: RTC_SLOW_CLK calibration value: 3313412
D (503) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (504) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (504) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (509) cpu_start: Starting scheduler on PRO CPU.
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (0) cpu_start: Starting scheduler on APP CPU.
D (524) heap_init: New heap initialised at 0x3ffe0440
D (534) heap_init: New heap initialised at 0x3ffe4350
D (544) intr_alloc: Connected src 16 to int 12 (cpu 0)
I (544) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (554) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (564) main: Setup
I (564) main: Wakeup was not caused by deep sleep (0)

I (574) CNT: InitGPIO
I (574) main: Check reason to connect: need_init:1, need_sync:0

I (584) transp: Wifi init start

D (584) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
D (634) nvs: nvs_open_from_partition misc 1
D (634) nvs: nvs_get_str_or_blob log
I (634) wifi: wifi driver task: 3ffc00e4, prio:23, stack:3072, core=0
I (634) wifi: wifi firmware version: 8c44f46
I (634) wifi: config NVS flash: enabled
I (644) wifi: config nano formating: enabled
I (644) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (654) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (664) nvs: nvs_open_from_partition nvs.net80211 1
D (674) nvs: nvs_get opmode 1
D (674) nvs: nvs_get_str_or_blob sta.ssid
D (674) nvs: nvs_get_str_or_blob sta.mac
D (684) nvs: nvs_get sta.authmode 1
D (684) nvs: nvs_get_str_or_blob sta.pswd
D (684) nvs: nvs_get_str_or_blob sta.pmk
D (694) nvs: nvs_get sta.chan 1
D (694) nvs: nvs_get auto.conn 1
D (694) nvs: nvs_get bssid.set 1
D (704) nvs: nvs_get_str_or_blob sta.bssid
D (704) nvs: nvs_get sta.lis_intval 2
D (714) nvs: nvs_get sta.phym 1
D (714) nvs: nvs_get sta.phybw 1
D (714) nvs: nvs_get_str_or_blob sta.apsw
D (724) nvs: nvs_get_str_or_blob sta.apinfo
D (724) nvs: nvs_get sta.scan_method 1
D (734) nvs: nvs_get sta.sort_method 1
D (734) nvs: nvs_get sta.minrssi 1
D (734) nvs: nvs_get sta.minauth 1
D (744) nvs: nvs_get_str_or_blob ap.ssid
D (744) nvs: nvs_get_str_or_blob ap.mac
D (744) nvs: nvs_get_str_or_blob ap.passwd
D (754) nvs: nvs_get_str_or_blob ap.pmk
D (754) nvs: nvs_get ap.chan 1
D (754) nvs: nvs_get ap.authmode 1
D (764) nvs: nvs_get ap.hidden 1
D (764) nvs: nvs_get ap.max.conn 1
D (764) nvs: nvs_get bcn.interval 2
D (774) nvs: nvs_get ap.phym 1
D (774) nvs: nvs_get ap.phybw 1
D (774) nvs: nvs_get ap.sndchan 1
D (784) nvs: nvs_get lorate 1
D (784) nvs: nvs_set_blob sta.mac 6
D (804) nvs: nvs_set_blob ap.mac 6
I (804) wifi: Init dynamic tx buffer num: 32
I (804) wifi: Init data frame dynamic rx buffer num: 32
I (804) wifi: Init management frame dynamic rx buffer num: 32
I (804) wifi: Init static rx buffer size: 1600
I (814) wifi: Init static rx buffer num: 10
I (814) wifi: Init dynamic rx buffer num: 32
D (824) nvs: nvs_set_blob sta.ssid 36
D (824) nvs: nvs_set_blob sta.pswd 65
D (824) nvs: nvs_set bssid.set 1 0
D (834) nvs: nvs_set sta.lis_intval 2 3
D (834) nvs: nvs_set sta.scan_method 1 0
D (834) nvs: nvs_set sta.sort_method 1 0
D (844) nvs: nvs_set sta.minrssi 1 -127
D (844) nvs: nvs_set sta.minauth 1 0
D (854) nvs: nvs_set_blob sta.apsw 2
D (854) nvs: nvs_set_blob sta.apinfo 700
D (864) RTC_MODULE: Wi-Fi takes adc2 lock.
D (864) phy_init: loading PHY init data from application binary
D (864) nvs: nvs_open_from_partition phy 0
D (874) nvs: nvs_get cal_version 4
D (874) nvs: nvs_get_str_or_blob cal_mac
D (874) nvs: nvs_get_str_or_blob cal_data
D (894) nvs: nvs_close 3
I (964) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (964) wifi: mode : sta (3c:71:bf:03:3e:44)
D (964) event: SYSTEM_EVENT_STA_START
I (964) transp: wifi_init_sta finished.
I (964) transp: Start connect to AP: my_phone, err:0

I (974) transp: connect to ap SSID:my_phone password:**********
I (1084) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (3054) wifi: state: init -> auth (b0)
I (3054) wifi: state: auth -> init (1c0)
I (3054) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
D (3054) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:my_phone, ssid_len:8, bssid:28:3f:69:c8:55:fb, reason:202
D (3064) tcpip_adapter: if0 start ip lost tmr: enter
D (3074) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc4f0c interval=120 ip=0
E (3084) transp: retry to connect to the AP: my_phone, err:0

D (5494) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:my_phone, ssid_len:8, bssid:00:00:00:00:00:00, reason:201
D (5494) tcpip_adapter: if0 start ip lost tmr: enter
D (5504) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc4f0c interval=120 ip=0
E (5514) transp: retry to connect to the AP: my_phone, err:0

I (5634) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5634) wifi: state: init -> auth (b0)
I (5634) wifi: state: auth -> assoc (0)
I (5644) wifi: state: assoc -> run (10)
I (5684) wifi: connected with my_phone, channel 1
D (5684) nvs: nvs_set sta.chan 1 1
D (5684) nvs: nvs_set_blob sta.apinfo 700
I (5694) wifi: pm start, type: 1

D (5694) event: SYSTEM_EVENT_STA_CONNECTED, ssid:my_phone, ssid_len:8, bssid:28:3f:69:c8:55:fb, channel:1, authmode:3
D (5704) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (5704) tcpip_adapter: if0 start ip lost tmr: enter
D (5714) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc4f0c interval=120 ip=0
D (5724) tcpip_adapter: dhcp client start successfully
D (6624) tcpip_adapter: if0 dhcpc cb
D (6624) tcpip_adapter: if0 ip changed=1
D (6624) event: SYSTEM_EVENT_STA_GOT_IP, ip:192.168.43.138, mask:255.255.255.0, gw:192.168.43.1
I (6624) event: sta ip: 192.168.43.138, mask: 255.255.255.0, gw: 192.168.43.1
```

liuzfesp, 2019-01-15:
HI @dibalavs, could you check whether this issue exists in latest IDF release/v3.1?

dibalavs, 2019-01-15:
Hello, I cheked on the latest v3.1 branch:
esp idf: commit 7fe18ef334d096219cbc4ba4e1feeadc830c58b5
esp32_wifi_lib: commit f5ce27713bfca49e86e2bdc55d95fd2efa8cef0c

The result a little better. So previously ESP connected to my phone after 2-3 re-tries, Now connected after 1-2 re-tries.
Connection to my home wi-fi router is OK (now and previously).

Logs:
```
D (583) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
D (633) nvs: nvs_open_from_partition misc 1
D (633) nvs: nvs_get_str_or_blob log
I (633) wifi: wifi driver task: 3ffbfcdc, prio:23, stack:3072, core=0
I (633) wifi: wifi firmware version: 22ffa86
I (643) wifi: config NVS flash: enabled
I (643) wifi: config nano formating: enabled
I (643) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (653) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (663) nvs: nvs_open_from_partition nvs.net80211 1
D (673) nvs: nvs_get opmode 1
D (673) nvs: nvs_get_str_or_blob sta.ssid
D (673) nvs: nvs_get_str_or_blob sta.mac
D (683) nvs: nvs_get sta.authmode 1
D (683) nvs: nvs_get_str_or_blob sta.pswd
D (693) nvs: nvs_get_str_or_blob sta.pmk
D (693) nvs: nvs_get sta.chan 1
D (693) nvs: nvs_get auto.conn 1
D (703) nvs: nvs_get bssid.set 1
D (703) nvs: nvs_get_str_or_blob sta.bssid
D (703) nvs: nvs_get sta.lis_intval 2
D (713) nvs: nvs_get sta.phym 1
D (713) nvs: nvs_get sta.phybw 1
D (713) nvs: nvs_get_str_or_blob sta.apsw
D (723) nvs: nvs_get_str_or_blob sta.apinfo
D (733) nvs: nvs_get sta.scan_method 1
D (733) nvs: nvs_get sta.sort_method 1
D (733) nvs: nvs_get sta.minrssi 1
D (733) nvs: nvs_get sta.minauth 1
D (743) nvs: nvs_get_str_or_blob ap.ssid
D (743) nvs: nvs_get_str_or_blob ap.mac
D (753) nvs: nvs_get_str_or_blob ap.passwd
D (753) nvs: nvs_get_str_or_blob ap.pmk
D (753) nvs: nvs_get ap.chan 1
D (763) nvs: nvs_get ap.authmode 1
D (763) nvs: nvs_get ap.hidden 1
D (763) nvs: nvs_get ap.max.conn 1
D (773) nvs: nvs_get bcn.interval 2
D (773) nvs: nvs_get ap.phym 1
D (773) nvs: nvs_get ap.phybw 1
D (783) nvs: nvs_get ap.sndchan 1
D (783) nvs: nvs_get lorate 1
D (783) nvs: nvs_set_blob sta.mac 6
D (803) nvs: nvs_set_blob ap.mac 6
I (803) wifi: Init dynamic tx buffer num: 32
I (803) wifi: Init data frame dynamic rx buffer num: 32
I (803) wifi: Init management frame dynamic rx buffer num: 32
I (803) wifi: Init static rx buffer size: 1600
I (813) wifi: Init static rx buffer num: 10
I (813) wifi: Init dynamic rx buffer num: 32
D (823) nvs: nvs_set_blob sta.ssid 36
D (823) nvs: nvs_set_blob sta.pswd 65
D (833) nvs: nvs_set bssid.set 1 0
D (833) nvs: nvs_set sta.lis_intval 2 3
D (833) nvs: nvs_set sta.scan_method 1 0
D (843) nvs: nvs_set sta.sort_method 1 0
D (843) nvs: nvs_set sta.minrssi 1 -127
D (843) nvs: nvs_set sta.minauth 1 0
D (853) nvs: nvs_set_blob sta.apsw 2
D (853) nvs: nvs_set_blob sta.apinfo 700
D (863) RTC_MODULE: Wi-Fi takes adc2 lock.
D (863) phy_init: loading PHY init data from application binary
D (863) nvs: nvs_open_from_partition phy 0
D (873) nvs: nvs_get cal_version 4
D (873) nvs: nvs_get_str_or_blob cal_mac
D (883) nvs: nvs_get_str_or_blob cal_data
D (893) nvs: nvs_close 3
I (963) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (973) wifi: mode : sta (3c:71:bf:03:3e:44)
D (973) event: SYSTEM_EVENT_STA_START
I (973) transp: wifi_init_sta finished.
I (973) transp: Start connect to AP: my_phone, err:0

I (983) transp: connect to ap SSID:my_phone password:**********
I (1093) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (3053) wifi: state: init -> auth (b0)
I (3063) wifi: state: auth -> init (1c0)
I (3063) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
D (3063) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:my_phone, ssid_len:8, bssid:28:3f:69:c8:55:fb, reason:202
D (3073) tcpip_adapter: if0 start ip lost tmr: enter
D (3073) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc4b04 interval=120 ip=0
E (3083) transp: retry to connect to the AP: my_phone, err:0

D (5503) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:my_phone, ssid_len:8, bssid:00:00:00:00:00:00, reason:201
D (5503) tcpip_adapter: if0 start ip lost tmr: enter
D (5503) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc4b04 interval=120 ip=0
E (5513) transp: retry to connect to the AP: my_phone, err:0

I (5633) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (5633) wifi: state: init -> auth (b0)
I (5643) wifi: state: auth -> assoc (0)
I (5653) wifi: state: assoc -> run (10)
I (5683) wifi: connected with my_phone, channel 1
D (5683) nvs: nvs_set sta.chan 1 1
D (5683) nvs: nvs_set_blob sta.apinfo 700
I (5693) wifi: pm start, type: 1

D (5693) event: SYSTEM_EVENT_STA_CONNECTED, ssid:my_phone, ssid_len:8, bssid:28:3f:69:c8:55:fb, channel:1, authmode:3
D (5693) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (5703) tcpip_adapter: if0 start ip lost tmr: enter
D (5703) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc4b04 interval=120 ip=0
D (5723) tcpip_adapter: dhcp client start successfully
D (5913) phy_init: wifi mac time delta: 102501
D (6103) phy_init: wifi mac time delta: 74105
```

liuzfesp, 2019-01-16:
@dibalavs I suspect it is a compatibility issue in ESP32, we are now debugging this issue.

esp-HarveyRong, 2019-11-07:
Hi @dibalavs Thank you very much for your feedback, I will debug the problem, I will update it in time.

chegewara, 2019-11-16:
Hi,
im not sure its the same issue, but i have project with deep sleep + ULP. When sleep time is relatively short (tested with 30 minutes) there is no problem, but with deep sleep set to 12 hours esp32 cant connect to AP after wake up. From logs we can see it is connecting but cant get IP and radio switch to light sleep mode, even if i set wifi power save to none `esp_wifi_set_ps(WIFI_PS_NONE);`:
```
I (1213) wifi: wifi driver task: 3ffd1cc8, prio:23, stack:3584, core=0
I (1215) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1225) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1236) wifi: wifi firmware version: d57a2ef
I (1238) wifi: config NVS flash: disabled
I (1242) wifi: config nano formating: disabled
I (1246) wifi: Init dynamic tx buffer num: 32
I (1250) wifi: Init data frame dynamic rx buffer num: 8
I (1255) wifi: Init management frame dynamic rx buffer num: 8
I (1260) wifi: Init management short buffer num: 32
I (1265) wifi: Init static tx buffer num: 6
I (1270) wifi: Init static rx buffer size: 1600
I (1273) wifi: Init static rx buffer num: 8
I (1277) wifi: Init dynamic rx buffer num: 8
I (1509) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 1
I (1511) wifi: mode : sta (24:0a:c4:xx)
I (1513) BLYNK_AWS: app_event_handler: event: 2
I (1637) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (3663) wifi: state: init -> auth (b0)
I (3669) wifi: state: auth -> assoc (0)
cycles: 14811137
I (3676) wifi: state: assoc -> run (10)
I (3749) wifi: connected with sipeed, aid = 1, channel 1, BW20, bssid = 26:18:1d:97:ea:0a, security type: 3, phy: bgn, rssi: -57
I (3750) wifi: pm start, type: 1

I (3757) BLYNK_AWS: app_event_handler: event: 4
I (3837) wifi: AP's beacon interval = 102400 us, DTIM period = 2
I (5714) wifi: state: run -> init (0)
I (5716) wifi: pm stop, total sleep time: 1580962 us / 1963292 us

I (5722) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
```

Just realized i might have set power save in wrong place, before wifi init.

esp-HarveyRong, 2020-03-31:
Hi @dibalavs,
I used my phone hotspot (Xiaomi 9) to reproduce the same problem, but I found by grabbing 802.11 packets, the cause of the problem was because the phone hotspot did not send Auth frames. 
At the same time I used another mobile (Apple 8p), he can connect stably.
So if you can provide the 802.11 packets that are causing the problem, it will be very helpful to analyze the problem.
The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets: 
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

dibalavs, 2020-04-01:
Hi.
I do not have Macbook, but on My windows laptop it looks like wifi adapter cannot work in monitor mode. In wireshark I can see only packets which are go to  my host (from my host) computer or broadcast packets, but no low-level 802.11 packets.

esp-HarveyRong, 2020-04-01:
Hi @dibalavs,
Can you try to reproduce the problem with the latest IDF version/v3.1?(https://github.com/espressif/esp-idf/commit/94e2981d17447c38c9c9b1b359c21236de96fdb7)

liuzfesp, 2020-04-15:
Hi @dibalavs based on our analysis and test, this issue has been fix by: be115ec3. So you can just upgrade your IDF to latest v3.1 to do a test.

We think the problem only happens in following scenario:

- ESP32 STA connects to the AP (some special AP, such as ) successfully
- Power reset or reboot ESP32
- Reconnect ESP32 to the AP, 

The root cause is, when reset board, ESP32 has no chance to send deauthentication frame to hotspot. So hotspot does not know ESP32 has disconnected from it. After reset, ESP32 send authentication request frame with sequence number zero. As hotspot still keeps the connection entity of ESP32, it will reply nothing to ESP32. As a result, ESP32 can not connect to hotspot until hotspot deauth ESP32 after a period of time or restart hotspot.

I would like to close this issue, just feel free to reopen it if you still have this issue after upgrading your IDF.

Alvin1Zhang, 2020-04-16:
@dibalavs Thanks for reporting,  will close this issue for now, feel free to reopen it if you still have this issue after upgrading your IDF. Thanks.

closed 2020-06-01 by Alvin1Zhang