espressif/esp-idf #2951: Nothing receiving on rmt_nec_tx_rx example.

keesus, 2019-01-15:
- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): 
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):  v3.3-dev-132-gbb4714671

- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 5.2.0
- Operating System: macOS
- Power Supply:  USB

Hi, I am in trouble with IR functions.

I am testing rmt_nec_tx_rx example with the git docs easily find.
(https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt_nec_tx_rx)

But nothing shows to my monitor while pushing  my remote control button.

Hardwares are all checked.

I tried to give more size to ringbuffer, failed.

Any help will be appreciated.

Thanks.

koobest, 2019-01-16:
Hi. @keesus
Can you provide your log?

keesus, 2019-01-16:
> Hi. @keesus
> Can you provide your log?

Hi @koobest 

Thanks for your reply.

I (0) cpu_start: App cpu up.
I (238) heap_init: Initializing. RAM available for dynamic allocation:
I (244) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (251) heap_init: At 3FFB2F08 len 0002D0F8 (180 KiB): DRAM
I (257) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (263) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (270) heap_init: At 4008A190 len 00015E70 (87 KiB): IRAM
I (276) cpu_start: Pro cpu start user code
I (294) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (495) NEC: RMT TX DATA
I (12565) NEC: RMT TX DATA
I (24635) NEC: RMT TX DATA

-------------

Simple log with showing continuing TX message.
I expected some message show while i push my remote button, but nothing happens.

koobest, 2019-01-16:
Can you get the log if doing the self-test ?

keesus, 2019-01-16:
@koobest 

Yes. At first i did self test and it was fine. I checked correct data on my monitor.

koobest, 2019-01-16:
Hi, can you help check the following 2 points?
1. The ESP32 RMT module can only receive signals without a carrier. Can you check if the infrared receiver tube has filtered out the carrier?
2. What is your active level ?(#define RMT_RX_ACTIVE_LEVEL) It depends on your hardware. 

thanks !!

keesus, 2019-01-16:
@koobest 

A1. I guess my receiver filters the carrier frequency. For a reference, I attached the data sheet of the receiver i'm using.
[ROM-WT138LM(N)..pdf](https://github.com/espressif/esp-idf/files/2763163/ROM-WT138LM.N.pdf)

A2. it is defined 0 in my code.

Thanks for your reply.

koobest, 2019-01-16:
Hi, keesus
Please change this macro, and check if it  works.
```
#define NEC_BIT_MARGIN         20 ----> 100

```

thanks !!

koobest, 2019-01-16:
Hi,
I think this is caused by the ending of the receive task. because it did not receieing data for a long time . 
```
static void rmt_example_nec_rx_task()
{
    int channel = RMT_RX_CHANNEL;
    nec_rx_init();
    RingbufHandle_t rb = NULL;
    //get RMT RX ringbuffer
    rmt_get_ringbuf_handle(channel, &rb);
    rmt_rx_start(channel, 1);
    while(rb) {
        size_t rx_size = 0;
        rmt_item32_t* item = (rmt_item32_t*) xRingbufferReceive(rb, &rx_size, 1000);
        if(item) {
            ....
            vRingbufferReturnItem(rb, (void*) item);
        } else {
            break;  // Posible break the loop from here ??  can add log here.
        }
    }
    vTaskDelete(NULL);
}
```

koobest, 2019-01-16:
Hi, sorry forget to give you a solution. :-)
```
rmt_item32_t* item = (rmt_item32_t*) xRingbufferReceive(rb, &rx_size, (portTickType)portMAX_DELAY);
```
You can keep this issue on until we have fixed this issue in IDF. 

thanks !!

keesus, 2019-01-16:
@koobest 

Perfect !! 

I wonder why i could not find any discussion about this but you gave an answer.

I will leave it opened for another wanderer.

Thanks !!

closed 2019-01-16 by keesus