espressif/esp-idf #2952: Getting error in make menuconfig

TheIronMechanic, 2019-01-15:
Hi,
I did everything according to the get started page, and even uninstalled everything and retried.
I open a MSYS2 MINGW32 Shell, copy the hello_world example to my  homefolder (~esp/), cd to it and try to run "make menuconfig", but I keep getting following issue:

```
Lucas@DESKTOP MINGW32 ~/esp/hello_world
$ make menuconfig
DEFCONFIG
/home/Lucas/esp/esp-idf/Kconfig:59: can't open file "Toolchain\msys32\home\Lucas\esp\esp-idf\components\partition_table\Kconfig.projbuild"
make: *** Keine Regel vorhanden, um das Ziel „defconfig“,
  benötigt von „menuconfig“, zu erstellen.  Schluss.
```

when I try running make clean I get this issue:
`Lucas@DESKTOP MINGW32 ~/esp/hello_world
$ make clean
DEFCONFIG
/home/Lucas/esp/esp-idf/Kconfig:59: can't open file "Toolchain\msys32\home\Lucas\esp\esp-idf\components\partition_table\Kconfig.projbuild"
make: *** Keine Regel vorhanden, um das Ziel „defconfig“,
  benötigt von „/home/Lucas/esp/hello_world/sdkconfig“, zu erstellen.  Schluss.
`

I'm on the latest verion from Github on windows 10

Somewhere I found that I need to edit .bash_profile and add
`export PATH="$PATH:$HOME/esp/xtensa-esp32-elf/bin"
export IDF_PATH="$HOME/esp/esp-idf"`

I tried removing it but it didnt change my issue

can anybody help me?

[image]

wayneuroda, 2019-01-16:
I have had similar issues like that before (in November 2018) which stopped me from upgrading my ESP-IDF.

Today I have followed all the steps again - but I skipped the python requirements step and everything seems to be working for me. Maybe try that? Otherwise, sorry I don't know.

extracted esp32_win32_msys2_environment_and_toolchain-20181001.zip
created esp folder
cloned esp-idf (as of Jan 16 2019)
Setup IDF_PATH="c:/dev/esp/jan2019/home/Wayne/esp/esp-idf"
copied hello world project
make menuconfig - OK
make... it builds!

TheIronMechanic, 2019-01-17:
I tried it exactly in your order. Now I'm getting:

`Lucas@DESKTOP MINGW32 ~/esp/hello_world
$ make menuconfig
Makefile:8: 2/esp32_win32_msys2_environment_and_toolchain-20181001/msys32/home/L                  ucas/esp/esp-idf/make/project.mk: No such file or directory
make: *** Keine Regel, um „2/esp32_win32_msys2_environment_and_toolchain-2018100                  1/msys32/home/Lucas/esp/esp-idf/make/project.mk“ zu erstellen.  Schluss.
`

but the file is there:

`Lucas@DESKTOP MINGW32 ~/esp/esp-idf/make
$ ls
common.mk  component_common.mk  component_wrapper.mk  ldgen.mk  project.mk  project_config.mk
`

How is this possible?

igrr, 2019-01-18:
Do you have a space somewhere in the path? The path in your message starts from '2':

2/esp32_win32_msys2_environment_and_toolchain-20181001/msys32/home/L ucas/esp/esp-idf/make/project.mk

TheIronMechanic, 2019-01-18:
Thanks a lot
It was the space.

this thing drove me crazy for 3 days in a row...

closed 2019-01-18 by TheIronMechanic

patrickctrf, 2020-05-22:
> Hello, there.
> I'm facing a problem to make the menuconfig. I´m trying everything and nothing helps. below you can see the error:
> Makefile:8: C:/msys32/home/9469/esp/esp-idf/make/project.mk: No such file or directory
> make: *** No rule to make target 'C:/msys32/home/9469/esp/esp-idf/make/project.mk'. Stop.
> 
> Some important information:
> 
> 9469@EC00966 MINGW32 /esp/hello_world
> $ echo $IDF_PATH
> C:/msys32/home/9469/esp/esp-idf
> 
> 9469@EC00966 MINGW32 /esp/hello_world
> $ ls /c/msys32
> autorebase.bat InstallationLog.txt msys2.ico
> autorebasebase1st.bat maintenancetool.dat msys2.ini
> clang32 maintenancetool.exe msys2_shell.cmd
> clang64 maintenancetool.ini network.xml
> components.xml mingw32 opt
> dev mingw32.exe sh.exe.stackdump
> dir mingw32.ini tmp
> esp mingw64 usr
> esp32-autotoolchain-msys2.sh mingw64.exe var
> etc mingw64.ini
> home msys2.exe
> 
> If anyone can help I really appreciated.
> Thanks in advance.

Did you remember to initialize "esp-idf" submodule? If you have Git installed, run the command bellow in your repository root directory:

git submodule update --init  --recursive