espressif/esp-idf #2954: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

leewo0, 2019-01-16:
This error occurs about 5-6 times per 12 hours(interval is random). Maybe more before I changed the timeout value from 300ms to 500ms in menuconfig. And it seems that this issue has no direct relationship with my application(application not running on CPU0?). I think this is caused by interrupt disable in ISR of CPU0 as 'Core 0 was running in ISR context:' and 'vTaskEnterCritical' according to the backtrace(each time the backtrace varies but these two msg seem always exist).
My application goes like this: both ble and wifi-sta are enabled. A uart is used with its driver installed and a task is created to deal with uart event queues. MQTT/BLE is used to receive data and then transfer with uart. A MQTT send task is created to receive queue objects, then send. Each task has hang mechanism. When it's running, uart continously receives data and send to MQTT send task by queue(about 1000Bytes/sec). MQTT receives data and transfer with uart(14Bytes/sec).
ESP-IDF is updated (commit b173cf28174b17f7df96c298f83c16a00d22e312).CPU clock is 240MHz.Interrupt watchdog settings are default except timeout value changed to 500ms.
sdkconfig:
CONFIG_INT_WDT=y
CONFIG_INT_WDT_TIMEOUT_MS=500
CONFIG_INT_WDT_CHECK_CPU1=y
CONFIG_TASK_WDT=y
CONFIG_TASK_WDT_PANIC=
CONFIG_TASK_WDT_TIMEOUT_S=5
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0=y
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=y
I have read #2672 and #2530. CONFIG_INT_WDT_CHECK_CPU1 is enabled.
What causes this error? I don't want to shut down the interrupt watchdog which could be worse for my application.
case 1:
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40094375  PS      : 0x00060834  A0      : 0x80088be3  A1      : 0x3ffbe2b0  
0x40094375: vTaskExitCritical at /home/tommy/esp/esp-idf/components/freertos/tasks.c:3537

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffbe308  A5      : 0x00000000  
A6      : 0x3ffdd328  A7      : 0x00060021  A8      : 0x80094375  A9      : 0x3ffbe2a0  
A10     : 0x00000003  A11     : 0x00060823  A12     : 0x00060821  A13     : 0x00000001  
A14     : 0x00060821  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core 0 was running in ISR context:
EPC1    : 0x40094255  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40094375
0x40094255: vTaskEnterCritical at /home/tommy/esp/esp-idf/components/freertos/tasks.c:3537

0x40094375: vTaskExitCritical at /home/tommy/esp/esp-idf/components/freertos/tasks.c:3537

Backtrace: 0x40094375:0x3ffbe2b0 0x40088be0:0x3ffbe2d0 0x4008a6fd:0x3ffbe2f0 0x40089dd9:0x3ffbe330 0x4008d0ca:0x3ffbe350 0x4008246d:0x3ffbe380 0x4000bfed:0x00000000
0x40094375: vTaskExitCritical at /home/tommy/esp/esp-idf/components/freertos/tasks.c:3537

0x40088be0: wifi_int_restore_wrapper at /home/tommy/esp/esp-idf/components/esp32/wifi_os_adapter.c:198

0x4008a6fd: pp_post at ??:?

0x40089dd9: lmacPostTxComplete at ??:?

0x4008d0ca: wDev_ProcessFiq at ??:?

0x4008246d: _xt_lowint1 at /home/tommy/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

Core 1 register dump:
PC      : 0x401b50de  PS      : 0x00060434  A0      : 0x800d4161  A1      : 0x3ffbc370  
0x401b50de: esp_pm_impl_waiti at /home/tommy/esp/esp-idf/components/esp32/pm_esp32.c:487

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x80093b9e  A5      : 0x3ffbc2a0  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d4026  A9      : 0x3ffbc340  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80093b9e  A13     : 0x3ffdbf30  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x401b50de:0x3ffbc370 0x400d415e:0x3ffbc390 0x40094951:0x3ffbc3b0 0x40093995:0x3ffbc3d0
0x401b50de: esp_pm_impl_waiti at /home/tommy/esp/esp-idf/components/esp32/pm_esp32.c:487

0x400d415e: esp_vApplicationIdleHook at /home/tommy/esp/esp-idf/components/esp32/freertos_hooks.c:86

0x40094951: prvIdleTask at /home/tommy/esp/esp-idf/components/freertos/tasks.c:3537

0x40093995: vPortTaskWrapper at /home/tommy/esp/esp-idf/components/freertos/port.c:403

case 2:
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40088be3  PS      : 0x00060e34  A0      : 0x8008a700  A1      : 0x3ffbe2d0  
0x40088be3: wifi_int_restore_wrapper at /home/tommy/esp/esp-idf/components/esp32/wifi_os_adapter.c:198

A2      : 0x3ffb60f8  A3      : 0x00000000  A4      : 0x3ffbe308  A5      : 0x3ffb5dc0  
A6      : 0x00000007  A7      : 0x00000000  A8      : 0x80088be3  A9      : 0x3ffbe2b0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffbe308  A13     : 0x00000000  
A14     : 0x3ffdd314  A15     : 0x00060021  SAR     : 0x0000001f  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core 0 was running in ISR context:
EPC1    : 0x40094255  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40088be3
0x40094255: vTaskEnterCritical at /home/tommy/esp/esp-idf/components/freertos/tasks.c:3537

0x40088be3: wifi_int_restore_wrapper at /home/tommy/esp/esp-idf/components/esp32/wifi_os_adapter.c:198

Backtrace: 0x40088be3:0x3ffbe2d0 0x4008a6fd:0x3ffbe2f0 0x40089dd9:0x3ffbe330 0x4008d0ca:0x3ffbe350 0x4008246d:0x3ffbe380 0x4000bfed:0x00000000
0x40088be3: wifi_int_restore_wrapper at /home/tommy/esp/esp-idf/components/esp32/wifi_os_adapter.c:198

0x4008a6fd: pp_post at ??:?

0x40089dd9: lmacPostTxComplete at ??:?

0x4008d0ca: wDev_ProcessFiq at ??:?

0x4008246d: _xt_lowint1 at /home/tommy/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

Core 1 register dump:
PC      : 0x401b50de  PS      : 0x00060e34  A0      : 0x800d4161  A1      : 0x3ffbc370  
0x401b50de: esp_pm_impl_waiti at /home/tommy/esp/esp-idf/components/esp32/pm_esp32.c:487

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x80093b9e  A5      : 0x3ffbc2a0  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d4026  A9      : 0x3ffbc340  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80093b9e  A13     : 0x3ffdbeb0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x401b50de:0x3ffbc370 0x400d415e:0x3ffbc390 0x40094951:0x3ffbc3b0 0x40093995:0x3ffbc3d0
0x401b50de: esp_pm_impl_waiti at /home/tommy/esp/esp-idf/components/esp32/pm_esp32.c:487

0x400d415e: esp_vApplicationIdleHook at /home/tommy/esp/esp-idf/components/esp32/freertos_hooks.c:86

0x40094951: prvIdleTask at /home/tommy/esp/esp-idf/components/freertos/tasks.c:3537

0x40093995: vPortTaskWrapper at /home/tommy/esp/esp-idf/components/freertos/port.c:403

liuzfesp, 2019-01-16:
HI @leewo0 this is a known WiFi/BT coexist issue, we already find out the root cause: the interrupt watchdog happens because the WiFi Transmitting interrupt status is not cleared when both WiFi/BT are in sleep status . The solution are still under investigation, I guess we will get a solution and merge it into IDF in two weeks.

leewo0, 2019-01-16:
> HI @leewo0 this is a known WiFi/BT coexist issue, we already find out the root cause: the interrupt watchdog happens because the WiFi Transmitting interrupt status is not cleared when both WiFi/BT [...]

Thank you for your reply. I'm glad you have found out the cause.

TimXia, 2019-02-01:
@leewo0 The bug has been fixed. Please update IDF to check it.

JosuGZ, 2019-02-11:
> @leewo0 The bug has been fixed. Please update IDF to check it.

> The Fix commit ID in master: [bf1955a](https://github.com/espressif/esp-idf/commit/bf1955aac819f501431f6d2230885ce8a4d4e6b8)

Being on that particular commit, I'm experiencing the same issue

TimXia, 2019-02-14:
@JosuGZ Could you please provide the whole log?

PRitu21, 2019-05-28:
Hi, 

I am also experiencing the same issue.
Device Version is : v4.0-dev-217-g5b2247df2

CORRUPT HEAP: Bad tail at 0x3ffd7199. Expected 0xbaad5678 got 0xbaad5600
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008ce50  PS      : 0x00060f34  A0      : 0x8008ddc5  A1      : 0x3ffd5a30
A2      : 0x3ffb0078  A3      : 0x00000000  A4      : 0x3ffc0a50  A5      : 0x00000001
A6      : 0x00060d23  A7      : 0x00000001  A8      : 0x0000cdcd  A9      : 0x0000cdcd
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x00000070  A13     : 0x00000001
A14     : 0x00060f23  A15     : 0x3ffb7584  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffec

ELF file SHA256: 7dc60b2b9414700e91ab41a9576633ffe41896e54ee51acc7206cd0d98c32f8a

Backtrace: 0x4008ce50:0x3ffd5a30 0x4008ddc2:0x3ffd5a50 0x4008e15a:0x3ffd5a70 0x4008e4c4:0x3ffd5a90 0x4008e5f5:0x3ffd5ad0 0x40083016:0x3ffd5af0 0x400830ed:0x3ffd5b20 0x400eede6:0x3ffd5b40 0x400ed98f:0x3ffd5b60 0x4000bd83:0x3ffd5b80 0x4000117d:0x3ffd5ba0 0x400592fe:0x3ffd5bc0 0x4005937a:0x3ffd5be0 0x40058bbf:0x3ffd5c00 0x400d9955:0x3ffd5c30 0x400df13e:0x3ffd5c60 0x400df281:0x3ffd5f70 0x40082eb5:0x3ffd5fa0 0x400d3a1a:0x3ffd5ff0 0x400d46ea:0x3ffd6090 0x4008def5:0x3ffd65d0

Core 1 register dump:
PC      : 0x4008f4e2  PS      : 0x00060934  A0      : 0x8008d897  A1      : 0x3ffe6130
A2      : 0x3ffc3fbc  A3      : 0x3ffdc2c0  A4      : 0x3ffb005c  A5      : 0x00000001
A6      : 0x00060923  A7      : 0x3ffc9590  A8      : 0x3ffdc2c0  A9      : 0x3ffdc2c0
A10     : 0x00000002  A11     : 0x00000002  A12     : 0xbaad5678  A13     : 0x00000001
A14     : 0x00060923  A15     : 0x3ffc17c0  SAR     : 0x00000018  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 7dc60b2b9414700e91ab41a9576633ffe41896e54ee51acc7206cd0d98c32f8a

Backtrace: 0x4008f4e2:0x3ffe6130 0x4008d894:0x3ffe6150 0x4008e8b0:0x3ffe6170 0x4008e919:0x3ffe61b0 0x40082fbd:0x3ffe61d0 0x400830ad:0x3ffe6200 0x400dc026:0x3ffe6220 0x400d5b91:0x3ffe6530 0x400904dd:0x3ffe6580 0x400906a2:0x3ffe65b0 0x40090e8a:0x3ffe65d0 0x400d36a7:0x3ffe6600 0x400d36cd:0x3ffe6620 0x400d3ab4:0x3ffe6640 0x4008def5:0x3ffe66f0

Please suggest what does this error mean?

Waiting for your response.

Thanks
Ritu.

JosuGZ, 2019-05-28:
You have a memory corruption problem, It seems that you are accessing the n+1 element of an array with n elements.

Are you using the idf-monitor?

PRitu21, 2019-05-29:
No, I am not using idf-monitor. I am using putty to check the logs, but I can use idf-monitor if you would ask to.
Below is the code which I have recently added, after which I have started getting this error. Please check the code if you find any thing suspicious.

[spiffs.txt](https://github.com/espressif/esp-idf/files/3230815/spiffs.txt)

Thanks
Ritu

JosuGZ, 2019-05-29:
The idf-monitor will resolve memory addresses from the serial output thus seeding some light about where your problem lies:

`make -j4 flash monitor ESPPORT=<your port here>`

JosuGZ, 2019-05-29:
> @JosuGZ Could you please provide the whole log?

I thought I did provide a log last time, my bad, sorry. I can confirm I'm no longer experiencing this problem as it was me the one taking too long to exit a critical section.

Alvin1Zhang, 2019-08-23:
@JosuGZ Thanks for the information, and @leewo0 thanks for reporting the issue. Feel free to reopen if you still have this issue, thanks.

AmrutaCh, 2020-01-09:
Something similar issues I am facing. I am running a TIMER0 and its resulting in GuruMeditation error.
Can anyone help?

[GuruMeditationError.txt](https://github.com/espressif/esp-idf/files/4039360/GuruMeditationError.txt)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4039366/sdkconfig.txt)

umer-ilyas, 2020-01-16:
I am facing this issue when i enable light sleep in power management 
```
    esp_pm_config_esp32_t cfg = {
        .max_freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ,
        .min_freq_mhz = 10,
        .light_sleep_enable = true,
    };
```

when i diable light_sleep_enable microcontroller does not restart 

using eps-idf - v3.3.1
my application is using wifi - mqtt and uart 

```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008284e  PS      : 0x00060034  A0      : 0x800d358d  A1      : 0x3ffaf9c0
0x4008284e: esp_timer_impl_set_alarm at F:/msys32/home/umeri/esp/esp-idf/components/esp32/esp_timer_esp32.c:367

A2      : 0x040804a7  A3      : 0x00000000  A4      : 0x3ffb4204  A5      : 0x00000b07
A6      : 0x00000003  A7      : 0x00060e23  A8      : 0x3ff47024  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00001e14  A12     : 0x00000014  A13     : 0x016cf426
A14     : 0x016cf43a  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 6b807dddf8f6cc517e89f921ef66f3d18c23f631c40872605ccc2db82e75a5f6

Backtrace: 0x4008284e:0x3ffaf9c0 0x400d358a:0x3ffaf9e0 0x400d35bb:0x3ffafa00 0x4008f6b9:0x3ffafa20
0x4008284e: esp_timer_impl_set_alarm at F:/msys32/home/umeri/esp/esp-idf/components/esp32/esp_timer_esp32.c:367

0x400d358a: timer_process_alarm at F:/msys32/home/umeri/esp/esp-idf/components/esp32/esp_timer.c:435

0x400d35bb: timer_task at F:/msys32/home/umeri/esp/esp-idf/components/esp32/esp_timer.c:435

0x4008f6b9: vPortTaskWrapper at F:/msys32/home/umeri/esp/esp-idf/components/freertos/port.c:403

Core 1 register dump:
PC      : 0x4008e528  PS      : 0x00060b34  A0      : 0x800828d5  A1      : 0x3ffb0bb0
0x4008e528: uxPortCompareSet at F:/msys32/home/umeri/esp/esp-idf/components/freertos/tasks.c:4409
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at F:/msys32/home/umeri/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at F:/msys32/home/umeri/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at F:/msys32/home/umeri/esp/esp-idf/components/freertos/tasks.c:4201

A2      : 0x3ffb0078  A3      : 0x00000140  A4      : 0x000000c6  A5      : 0x00000000
A6      : 0x00000020  A7      : 0x00000000  A8      : 0x0000cdcd  A9      : 0x0000abab
A10     : 0xb33fffff  A11     : 0x0000cdcd  A12     : 0x00000005  A13     : 0x00000001
A14     : 0x00060b23  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

ELF file SHA256: 6b807dddf8f6cc517e89f921ef66f3d18c23f631c40872605ccc2db82e75a5f6

Backtrace: 0x4008e528:0x3ffb0bb0 0x400828d2:0x3ffb0bd0 0x40081c9a:0x3ffb0bf0 0x40081e00:0x3ffb0c10 0x40081f31:0x3ffb0c50 0x400820c9:0x3ffb0c70 0x40081e3f:0x3ffb0c90 0x40081f68:0x3ffb0cb0 0x400831ab:0x3ffb0cd0 0x4000bfed:0x3ffc0580 0x4008e62e:0x3ffc0590 0x4008ecac:0x3ffc0
5b0 0x4008f6b9:0x3ffc05d0
0x4008e528: uxPortCompareSet at F:/msys32/home/umeri/esp/esp-idf/components/freertos/tasks.c:4409
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at F:/msys32/home/umeri/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at F:/msys32/home/umeri/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at F:/msys32/home/umeri/esp/esp-idf/components/freertos/tasks.c:4201

0x400828d2: esp_timer_impl_update_apb_freq at F:/msys32/home/umeri/esp/esp-idf/components/esp32/esp_timer_esp32.c:367

0x40081c9a: on_freq_update at F:/msys32/home/umeri/esp/esp-idf/components/esp32/pm_esp32.c:594

0x40081e00: do_switch at F:/msys32/home/umeri/esp/esp-idf/components/esp32/pm_esp32.c:594

0x40081f31: esp_pm_impl_switch_mode at F:/msys32/home/umeri/esp/esp-idf/components/esp32/pm_esp32.c:594

0x400820c9: esp_pm_lock_acquire at F:/msys32/home/umeri/esp/esp-idf/components/esp32/pm_locks.c:120

0x40081e3f: leave_idle at F:/msys32/home/umeri/esp/esp-idf/components/esp32/pm_esp32.c:594
[... 52 lines cut ...]
I (416) esp_image: segment 6: paddr=0x000fe958 vaddr=0x400c0000 size=0x00064 (   100) load
I (416) esp_image: segment 7: paddr=0x000fe9c4 vaddr=0x50000000 size=0x00004 (     4) load
I (434) boot: Loaded app from partition at offset 0x10000
I (434) boot: Disabling RNG early entropy source...
I (434) cpu_start: Pro cpu up.
I (438) cpu_start: Application information:
I (443) cpu_start: Project name:     solar
I (448) cpu_start: App version:      e8f708b-dirty
I (453) cpu_start: Compile time:     Jan 16 2020 17:00:58
I (459) cpu_start: ELF file SHA256:  6b807dddf8f6cc51...
I (465) cpu_start: ESP-IDF:          v3.3.1-1-gc62c5f2cc-dirty
I (472) cpu_start: Starting app cpu, entry point is 0x40081340
0x40081340: call_start_cpu1 at F:/msys32/home/umeri/esp/esp-idf/components/esp32/cpu_start.c:269

I (458) cpu_start: App cpu up.
I (482) heap_init: Initializing. RAM available for dynamic allocation:
I (489) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (495) heap_init: At 3FFBD7B8 len 00022848 (138 KiB): DRAM
I (501) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (508) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (514) heap_init: At 40092460 len 0000DBA0 (54 KiB): IRAM
I (520) cpu_start: Pro cpu start user code
I (538) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

KonstantinKondrashov, 2020-01-16:
Hi @umer-ilyas!
It is a bug related to esp_timer. Could you try this patch [7304.zip](https://github.com/espressif/esp-idf/files/4073170/7304.zip)?
How often does this problem occur? (regularly, sometimes)

umer-ilyas, 2020-01-17:
thanks @KonstantinKondrashov 
it happens somewhere between 30 minutes to few hours 

i ll test your patch and let you know what happens

umer-ilyas, 2020-01-18:
sorry this patch could not help out

KonstantinKondrashov, 2020-01-18:
Hi @umer-ilyas!
Really, that patch does not cover your case. I have worked on it and probably I have a solution for it. Could you try another patch [fix_timer.zip](https://github.com/espressif/esp-idf/files/4081117/fix_timer.zip)
Thanks.

umer-ilyas, 2020-01-20:
i applied both patches and now it seems that it is working 
wdt have not triggered in last few hours 
thanks

closed 2020-01-23 by espressif-bot