espressif/esp-idf #2958: Wi-fi supported security modes

Victor-ghcn, 2019-01-17:
Hi, 

We have a project using ESP32 of which we need to specify what WIFI security modes are supported and we are given the following list:
-WPA Personal
-WPA2 Personal
-WPA Enterprise
-WPA2 Enterprise
-Other, Please List:

The following is from the online documentation:
(https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/wifi/esp_wifi.html)
"Various security modes for the above (WPA, WPA2, WEP, etc.)"

We are not sure which one(s) we should select, could you please help?

Thank you for your attention!
Victor

liuzfesp, 2019-01-22:
HI @Victor-ghcn, you can get all the ESP32 supported security (auth) mode from following definition:
typedef enum {
    WIFI_AUTH_OPEN = 0,         /**< authenticate mode : open */
    WIFI_AUTH_WEP,              /**< authenticate mode : WEP */
    WIFI_AUTH_WPA_PSK,          /**< authenticate mode : WPA_PSK */
    WIFI_AUTH_WPA2_PSK,         /**< authenticate mode : WPA2_PSK */
    WIFI_AUTH_WPA_WPA2_PSK,     /**< authenticate mode : WPA_WPA2_PSK */
    WIFI_AUTH_WPA2_ENTERPRISE,  /**< authenticate mode : WPA2_ENTERPRISE */
    WIFI_AUTH_MAX
} wifi_auth_mode_t;

For ESP32 station, you don't need to configure any auth mode, for ESP32 soft-AP, the recommended auth mode is WIFI_AUTH_WPA_WPA2_PSK.

Victor-ghcn, 2019-01-24:
Hi liuzfesp, Noted with thanks!

closed 2019-01-24 by Victor-ghcn