espressif/esp-idf #2962: Esp-idf or ESP32 chip supports iBeacon scan and advertising simultaneously ?

muratsplat, 2019-01-17:
Hello Guys 

## Environment

- Development Kit:   EASY KIT ESP-32
- Kit version : C1
- Module or chip used:  [ESP32-WROOM-32]
- IDF version: v3.3-beta1-223-ga62cbfec9
- Build System:         [Make]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [macOS]
- Power Supply:         [USB]

## Question 

I looked at `https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/ble_ibeacon/main/esp_ibeacon_api.h` file and There is a  comment in there. It says "Because current ESP IDF version doesn't support scan and adv simultaneously so iBeacon sender and receiver should not run simultaneously" on line 31 of the file.

But I tried to work  scan and adv simultaneously. It works  looks well.  I care that  ESP-32 and the framework supports  scan and adv simultaneously now ?

chegewara, 2019-01-17:
Yes, currently esp32 supports scan and adv at the same time.

muratsplat, 2019-01-18:
So it is ready for production ?

Weijian-Espressif, 2019-01-21:
@muratsplat, ESP32 supports scan and adv at the same time now.

sencersoylu, 2020-04-18:
How can do that at sime time do you have any example ?

michaelleuer, 2020-04-22:
Yes please I would love an example!

gengyuchao, 2020-04-22:
Hello there is an example at the path `esp-idf/examples/bluetooth/bluedroid/coex/gattc_gatts_coex` in the master branch shows how to use both adv and scan.

michaelleuer, 2020-04-22:
@gengyuchao  Thank you! I will take a look now =) You Rock!

ilcontefabio, 2020-05-14:
Hi
can i have some confirmation about adv and scan working together?

I'm using something like that ...

void bleTask(void *unused)
{
	while (1)
	{
		ESP_LOGI(TAG, " start adv and scan \n");
		esp_ble_gap_set_scan_params(&ble_scan_params);
		ESP_ERROR_CHECK(esp_ble_gap_config_adv_data_raw(adv_raw_data, 30));
		vTaskDelay(adv_int * 1000 / portTICK_PERIOD_MS);
		esp_ble_gap_stop_advertising();
		ESP_LOGI(TAG, " stop adv \n");
		vTaskDelay((scan_int - adv_int) * 1000 / portTICK_PERIOD_MS);
		esp_ble_gap_stop_scanning();
		ESP_LOGI(TAG, " stop scan ..... \n");
	}
}

It seems to be work .. but sometime the scan period see many adv and sometine a few ... in addiction the rssi red is not very stable ...

We are thinkingcoud be some coexistency problem 

Thans
Fabio

chegewara, 2020-05-14:
@ilcontefabio it will work more stable when you set scan interval shorter than scan window. This way scan wont starve 100% of radio time.

ilcontefabio, 2020-05-14:
> @ilcontefabio it will work more stable when you set scan interval shorter than scan window. This way scan wont starve 100% of radio time.

thanks
but all timeis togarantee to see al the advertise packet ... but you would write interval larger than window?
is it ok to work with advertise and scanactive together?

chegewara, 2020-05-15:
> is it ok to work with advertise and scanactive together?

Yes, it is very OK to use scan and adv together.

> but all timeis togarantee to see al the advertise packet ... but you would write interval larger than window?

Ok, understand you want to have all time scan to guarantee, and how is it going now? Like you said, you can see sometimes more and sometimes less packets. This is hardware restriction. You have only 1 radio which can do 1 thing at a time in very strict timing, so you can either scan or advertise, but it is done in very short timeframes so, you can run both in your app and controller with take care of toggling radio.
Your only job is to set:
- advertising interval, longer is better, because radio has more time for scan
- scan window and interval, even if you set just a bit shorter interval than window, it will give controller time to run advertising

Im not saying you have to do this, i am just saying it will help you better use esp32 bluetooth according to my knowledge.

ilcontefabio, 2020-05-15:
> Yes, it is very OK to use scan and adv together.
>
>
> Ok, understand you want to have all time scan to guarantee, and how is it going now? Like you said, you can see sometimes more and sometimes less [...]

many many thanks for your description
now I'm working with 

static esp_ble_scan_params_t ble_scan_params = {
	.scan_type = BLE_SCAN_TYPE_ACTIVE,
	.own_addr_type = BLE_ADDR_TYPE_PUBLIC,
	.scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
	.scan_interval = 80,
	.scan_window = 80};

static esp_ble_adv_params_t ble_adv_params = {
	.adv_int_min = 32,
	.adv_int_max = 64,
	.adv_type = ADV_TYPE_NONCONN_IND,
	.own_addr_type = BLE_ADDR_TYPE_PUBLIC,
	.channel_map = ADV_CHNL_ALL,
	.adv_filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
};

that is not so god as you said ...

From your experience, which is the best configuration for a scan and adv running together?

And how has to be the scan duration to have a good number of adv scanned packet to average the rssi measure?

thanks