espressif/esp-idf #2963: UART chunks fail when wifi reconnect happens

hchaudhary1, 2019-01-17:
## Environment

- Development Kit: Custom PICO
- IDF version: master 15Jan2019, a62cbfec9ab681bddf8f8e45d9949b1b1f67a2ec
- Build System: Make
- Compiler version: 5.2.0
- Operating System: Windows
- Power Supply:  Battery --> DCDC, 3.3VDC

## Problem Description

For this test we are sending chuncks of 4000 byte UART chucks to the ESP32 from another processor. When there is a wifi disconnect->reconnect, then UART gets corrupted. UART is running at 115200.

### Expected Behavior
UART RX should always work, regardless of wifi modem state.

### Actual Behavior
UART RX packets are missed/corrupted when wifi reconnects.

### Steps to repropduce
Run example code to reproduce issue.

Attached zip
1. sample code adapted from "station_example_main"
2. sdkconfig
3. screenshot of logs in failure case

[failue.zip](https://github.com/espressif/esp-idf/files/2770156/failue.zip)

koobest, 2019-01-21:
Hi, @hchaudhary1 
I run your code, but don't reproduce your issue, the following is my log output, which is a bit different from yours.  I gave a wrong WiFi password to do this test. 

Can you help to check when WiFi reconnects, will the flow control pin  (RTS) output high? (you need enable flow control in configuration).

```
I (16651) wifi: state: run -> init (3c0)
I (16651) wifi: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (16651) wifi: flush txq
I (16651) wifi: stop sw txq
I (16661) wifi: lmac stop hw txq
I (16701) wifi: mode : sta (30:ae:a4:80:06:78)
<0x1b>[0;31mE (16701) wifi station: retry to connect to the AP<0x1b>[0m
<0x1b>[0;31mE (16701) wifi station: connect to the AP fail
<0x1b>[0m
<0x1b>[0;32mI (16891) uart: Recieved SUCCESFULLY,   81<0x1b>[0m
I (17061) wifi: new:<13,0>, old:<1,0>, ap:<255,255>, sta:<13,0>, prof:1
<0x1b>[0;32mI (17241) uart: Recieved SUCCESFULLY,   112<0x1b>[0m
<0x1b>[0;32mI (17591) uart: Recieved SUCCESFULLY,   76<0x1b>[0m
<0x1b>[0;32mI (17931) uart: Recieved SUCCESFULLY,   56<0x1b>[0m
I (18051) wifi: state: init -> auth (b0)
I (18061) wifi: state: auth -> assoc (0)
I (18081) wifi: state: assoc -> run (10)
<0x1b>[0;32mI (18291) uart: Recieved SUCCESFULLY,   100<0x1b>[0m
<0x1b>[0;32mI (18631) uart: Recieved SUCCESFULLY,   122<0x1b>[0m
<0x1b>[0;32mI (18971) uart: Recieved SUCCESFULLY,   86<0x1b>[0m
<0x1b>[0;32mI (19331) uart: Recieved SUCCESFULLY,   56<0x1b>[0m
<0x1b>[0;32mI (19671) uart: Recieved SUCCESFULLY,   110<0x1b>[0m
<0x1b>[0;32mI (20011) uart: Recieved SUCCESFULLY,   74<0x1b>[0m
<0x1b>[0;32mI (20371) uart: Recieved SUCCESFULLY,   54<0x1b>[0m
<0x1b>[0;32mI (20711) uart: Recieved SUCCESFULLY,   98<0x1b>[0m
<0x1b>[0;32mI (21061) uart: Recieved SUCCESFULLY,   120<0x1b>[0m
<0x1b>[0;32mI (21411) uart: Recieved SUCCESFULLY,   84<0x1b>[0m
<0x1b>[0;32mI (21751) uart: Recieved SUCCESFULLY,   54<0x1b>[0m
<0x1b>[0;32mI (22101) uart: Recieved SUCCESFULLY,   108<0x1b>[0m
<0x1b>[0;32mI (22451) uart: Recieved SUCCESFULLY,   72<0x1b>[0m
<0x1b>[0;32mI (22791) uart: Recieved SUCCESFULLY,   52<0x1b>[0m
I (22901) wifi: state: run -> init (3c0)
I (22901) wifi: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (22901) wifi: flush txq
I (22901) wifi: stop sw txq
I (22901) wifi: lmac stop hw txq
I (22941) wifi: mode : sta (30:ae:a4:80:06:78)
<0x1b>[0;31mE (22941) wifi station: retry to connect to the AP<0x1b>[0m
<0x1b>[0;31mE (22941) wifi station: connect to the AP fail
<0x1b>[0m
<0x1b>[0;32mI (23141) uart: Recieved SUCCESFULLY,   54<0x1b>[0m
I (23301) wifi: new:<13,0>, old:<1,0>, ap:<255,255>, sta:<13,0>, prof:1
<0x1b>[0;32mI (23491) uart: Recieved SUCCESFULLY,   118<0x1b>[0m
<0x1b>[0;32mI (23841) uart: Recieved SUCCESFULLY,   82<0x1b>[0m
<0x1b>[0;32mI (24181) uart: Recieved SUCCESFULLY,   52<0x1b>[0m
I (24291) wifi: state: init -> auth (b0)
I (24291) wifi: state: auth -> assoc (0)
I (24311) wifi: state: assoc -> run (10)
<0x1b>[0;32mI (24531) uart: Recieved SUCCESFULLY,   106<0x1b>[0m
<0x1b>[0;32mI (24881) uart: Recieved SUCCESFULLY,   70<0x1b>[0m
<0x1b>[0;32mI (25221) uart: Recieved SUCCESFULLY,   50<0x1b>[0m
<0x1b>[0;32mI (25571) uart: Recieved SUCCESFULLY,   52<0x1b>[0m
<0x1b>[0;32mI (25921) uart: Recieved SUCCESFULLY,   116<0x1b>[0m
<0x1b>[0;32mI (26261) uart: Recieved SUCCESFULLY,   80<0x1b>[0m
<0x1b>[0;32mI (26621) uart: Recieved SUCCESFULLY,   50<0x1b>[0m
<0x1b>[0;32mI (26961) uart: Recieved SUCCESFULLY,   104<0x1b>[0m
<0x1b>[0;32mI (27301) uart: Recieved SUCCESFULLY,   68<0x1b>[0m
<0x1b>[0;32mI (27661) uart: Recieved SUCCESFULLY,   90<0x1b>[0m
<0x1b>[0;32mI (28001) uart: Recieved SUCCESFULLY,   50<0x1b>[0m
<0x1b>[0;32mI (28341) uart: Recieved SUCCESFULLY,   114<0x1b>[0m
<0x1b>[0;32mI (28701) uart: Recieved SUCCESFULLY,   78<0x1b>[0m
<0x1b>[0;32mI (29041) uart: Recieved SUCCESFULLY,   48<0x1b>[0m
I (29141) wifi: state: run -> init (3c0)
I (29141) wifi: new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (29141) wifi: flush txq
I (29141) wifi: stop sw txq
I (29141) wifi: lmac stop hw txq
I (29181) wifi: mode : sta (30:ae:a4:80:06:78)
<0x1b>[0;31mE (29181) wifi station: retry to connect to the AP<0x1b>[0m
<0x1b>[0;31mE (29181) wifi station: connect to the AP fail
<0x1b>[0m
<0x1b>[0;32mI (29391) uart: Recieved SUCCESFULLY,   102<0x1b>[0m
I (29541) wifi: new:<13,0>, old:<1,0>, ap:<255,255>, sta:<13,0>, prof:1
<0x1b>[0;32mI (29741) uart: Recieved SUCCESFULLY,   66<0x1b>[0m
<0x1b>[0;32mI (30081) uart: Recieved SUCCESFULLY,   88<0x1b>[0m
<0x1b>[0;32mI (30431) uart: Recieved SUCCESFULLY,   48<0x1b>[0m
I (30531) wifi: state: init -> auth (b0)
I (30531) wifi: state: auth -> assoc (0)
I (30551) wifi: state: assoc -> run (10)
```

thanks !!

hchaudhary1, 2019-01-21:
@koobest 
1. I will check on RTS
2. looks like you are not sending data in sequence. If you send packets in sequence (eg: 1, 2, 3 .... 1000). You will find that there will be packets that are missed.

hchaudhary1, 2019-01-21:
Sorry, I forgot to mention the test pattern to send over UART is:
[1][0][0][0][0][0][0][0][0].....[0] //4000 bytes
[2][0][0][0][0][0][0][0][0].....[0] //4000 bytes
[3][0][0][0][0][0][0][0][0].....[0] //4000 bytes
.
.
.

The first byte should increase, the reset should be 0.
Can you please test again with this pattern?

hchaudhary1, 2019-02-11:
@koobest I have tested the RTS scenario. RTS is always high, except for transmissions. Tested using wrover boards.  
See attached pic:

[image]

Please test in accordance with the test sequence above, you will see the issue.

hchaudhary1, 2019-02-11:
RTS pin behaves same during and/or before failure. There is no difference seen.

koobest, 2019-02-15:
Hi, @hchaudhary1 
 GPIO9 and GPIO10 are used by flash,  do you want to wake up the chip with UART?

hchaudhary1, 2019-02-15:
@koobest
We are using PICO, so the pins are free.
Also, we would like to wake on UART, but had issues with this.

wimdewitte, 2019-02-21:
We observer similar issues. In our application also RX data is missed when the wifi/networking stack is reconnecting.

Alvin1Zhang, 2019-03-14:
@koobest Hi, would you help to take a relook at this issue? Thanks.

negativekelvin, 2019-03-14:
Are you running uart install on core 1?

hchaudhary1, 2019-03-15:
I tried all combinations:
-UART pinned to core 1, Wifi Pinned to 0
-UART noaffinity, Wifi noaffinity
etc...  issue still exists

negativekelvin, 2019-03-15:
Did you try using uart events to see if any errors are reported

hchaudhary1, 2019-03-15:
hmm... I can try that

Alvin1Zhang, 2019-05-16:
@hchaudhary1 Thanks for reporting the issue. Would you please help share if any updates for the issue? Thanks.

hchaudhary1, 2019-05-17:
@Alvin1Zhang I was unable to solve it the issue.

koobest, 2019-07-19:
Hi, @hchaudhary1 
I tried  your example with the pattern, and tested for a few minutes, but can not reproduce your issue.

```
Recieved SUCCESFULLY,   101
Recieved SUCCESFULLY,   102
Recieved SUCCESFULLY,   49
Recieved SUCCESFULLY,   50
Recieved SUCCESFULLY,   51
Recieved SUCCESFULLY,   52
I (21434) wifi: flush txq
I (21434) wifi: stop sw txq
I (21434) wifi: lmac stop hw txq
I (21444) wifi: mode : sta (18:fe:34:6d:74:d1)
retry to connect to the AP
connect to the AP fail
Recieved SUCCESFULLY,   53
Recieved SUCCESFULLY,   54
Recieved SUCCESFULLY,   55
Recieved SUCCESFULLY,   56
Recieved SUCCESFULLY,   57
Recieved SUCCESFULLY,   97
I (23504) wifi: flush txq
I (23504) wifi: stop sw txq
I (23504) wifi: lmac stop hw txq
I (23514) wifi: mode : sta (18:fe:34:6d:74:d1)
retry to connect to the AP
connect to the AP fail
Recieved SUCCESFULLY,   98
Recieved SUCCESFULLY,   99
Recieved SUCCESFULLY,   100
Recieved SUCCESFULLY,   101
Recieved SUCCESFULLY,   102
Recieved SUCCESFULLY,   49
I (25574) wifi: flush txq
I (25574) wifi: stop sw txq
I (25574) wifi: lmac stop hw txq
I (25584) wifi: mode : sta (18:fe:34:6d:74:d1)
retry to connect to the AP
connect to the AP fail
Recieved SUCCESFULLY,   50
Recieved SUCCESFULLY,   51
Recieved SUCCESFULLY,   52
Recieved SUCCESFULLY,   53
Recieved SUCCESFULLY,   54
I (27644) wifi: flush txq
I (27644) wifi: stop sw txq
I (27644) wifi: lmac stop hw txq
Recieved SUCCESFULLY,   55
I (27654) wifi: mode : sta (18:fe:34:6d:74:d1)
retry to connect to the AP
connect to the AP fail
```
The strange thing is that every time WiFi is reconnected, The UART task can receive 5 to 6 packets. But in the your log, I only saw 2 packets. So can you try the latest master branch ?

thanks !!

hchaudhary1, 2019-07-19:
Ok, will retest in a few weeks

ammaree, 2019-07-24:
We have a similar but not exactly same problem losing characters transmitted through the UART. Background is as follows:
We have 3 separate tasks enabling access to our device using UART, HTTP and Telnet.
All 3 tasks use the same command interpreter with out going into an 8KB buffer.
If the command came in from UART then the buffered response is returned via UART, similarly with TNET & HTTP.

Executing the same command, listing all running tasks and associated stats, produces 19 tabulated lines of output about 1515 -> 5120 bytes in total.
Running the request via HTTP or TNET produces 100% correct output no matter how many times it is run.
Running the request via UART clearly shows (through the broken tabulation) 1 or more missing characters every 3rd or 4th time the command is run.
Our UART speed is 2Mbps but the same occurs at 115200bps.

Alvin1Zhang, 2020-05-28:
@hchaudhary1 @ammaree Thanks for reporting and updates. We have been trying to reproduce this issue but failed to reproduce, will close for now, feel free to reopen with more updates/details. Thanks.

closed 2020-05-28 by Alvin1Zhang