espressif/esp-idf #2966: bootloader fails to map flash segments that are close to 64k boundary

sinope-db, 2019-01-18:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32 16MB]
- IDF version: 3.0.7 https://github.com/espressif/esp-idf/commit/7e0ea6b05db8ad5d1a8dc183414ca8cd456ec934
- Build System:         Make
- Compiler version: 1.22.0-61-gab8375a9
- Operating System:     Windows
- Power Supply:         external 3.3V

## Problem Description

When the size of DROM segment is within 32 bytes of a 64k boundary, the last bytes are not mapped to cache.

If I build an application with a DROM segment (vaddr=0x3f400020) size of 0xbfffc, constructors (.ctors section) are not mapped and the function do_global_ctors() will call a wrong pointer and crash. Building the same application with a DROM segment size of 0xbffdc works. If DROM is 0xc0000 or larger it also works.

I think the issue is that set_cache_and_start_app() function does not take into account the start offset when calculating the number of pages required.

https://github.com/espressif/esp-idf/blob/7e0ea6b05db8ad5d1a8dc183414ca8cd456ec934/components/bootloader/subproject/main/bootloader_start.c#L613

should probably be something like:

```cpp
uint32_t drom_page_count = (drom_size + (drom_load_addr & (64*1024-1)) + 64*1024 - 1) / (64*1024); 
```
Same for IROM segment.

### Expected Behavior

Since the DROM segment is loaded at address 0x3f400020, a segment size of to 0xbfffc would require to be mapped over 13 pages (segment end address = 0x3f400020 + 0xbfffc = 0x3f4c001c)

### Actual Behavior

set_cache_and_start_app() only map 12 pages for size 0xbfffc.

### Steps to repropduce

1. embed a dummy binary file in the project to make DROM segment size 0xbfffc
2. check in .map file that .ctors section is at the beginning of last page (0x3f4c0018). 

## Debug Logs

```
I (11) boot: ESP-IDF v3.0.7 2nd stage bootloader
I (11) boot: compile time 09:05:55
I (11) boot: Enabling RNG early entropy source...
I (15) qio_mode: Enabling default flash chip QIO
I (20) boot: SPI Speed      : 40MHz
I (24) boot: SPI Mode       : QIO
I (28) boot: SPI Flash Size : 16MB
I (32) boot: Partition Table:
I (36) boot: ## Label            Usage          Type ST Offset   Length
I (43) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (51) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (58) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (65) boot:  3 AtpData          unknown          40 00 00010000 00001000
I (73) boot:  4 static_fs        Unknown data     01 82 00020000 00020000
I (80) boot:  5 nvs2             WiFi data        01 02 00040000 00040000
I (88) boot:  6 dynamic_fs       Unknown data     01 82 00080000 00380000
I (95) boot:  7 factory          factory app      00 00 00400000 00400000
I (103) boot:  8 ota_0            OTA app          00 10 00800000 00400000
I (110) boot:  9 ota_1            OTA app          00 11 00c00000 00400000
I (118) boot: End of partition table
I (122) boot: Defaulting to factory image
I (127) esp_image: segment 0: paddr=0x00400020 vaddr=0x3f400020 size=0xbfffc (786428) map
I (369) esp_image: segment 1: paddr=0x004c0024 vaddr=0x3ffb0000 size=0x03158 ( 12632) load
I (374) esp_image: segment 2: paddr=0x004c3184 vaddr=0x40080000 size=0x00400 (  1024) load
I (377) esp_image: segment 3: paddr=0x004c358c vaddr=0x40080400 size=0x0c878 ( 51320) load
I (404) esp_image: segment 4: paddr=0x004cfe0c vaddr=0x400c0000 size=0x00064 (   100) load
I (404) esp_image: segment 5: paddr=0x004cfe78 vaddr=0x50000000 size=0x00014 (    20) load
I (410) esp_image: segment 6: paddr=0x004cfe94 vaddr=0x00000000 size=0x0017c (   380)
I (419) esp_image: segment 7: paddr=0x004d0018 vaddr=0x400d0018 size=0xa246c (664684) map
I (633) boot: Loaded app from partition at offset 0x400000
I (633) boot: Disabling RNG early entropy source...
I (634) cpu_start: Pro cpu up.
I (637) cpu_start: Starting app cpu, entry point is 0x400810f8
I (0) cpu_start: App cpu up.
I (648) heap_init: Initializing. RAM available for dynamic allocation:
I (655) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (661) heap_init: At 3FFC4F20 len 0001B0E0 (108 KiB): DRAM
I (667) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (673) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (682) heap_init: At 4008CC78 len 00013388 (76 KiB): IRAM
I (686) cpu_start: Pro cpu start user code
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited)
. Exception was unhandled.
Register dump:
PC      : 0xffffffff  PS      : 0x00060f30  A0      : 0x8008104f  A1      : 0x3ffe3bd0
A2      : 0x3f4c0014  A3      : 0x000af122  A4      : 0x3ff000cc  A5      : 0x3ff00020
A6      : 0xffffffff  A7      : 0xfffffff0  A8      : 0x800d089e  A9      : 0x3ffe3bb0
A10     : 0x3ffafc88  A11     : 0x00000000  A12     : 0x00001000  A13     : 0x00000000
A14     : 0x00000016  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000014
EXCVADDR: 0xfffffffc  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x7fffffff:0x3ffe3bd0 0x4008104c:0x3ffe3bf0 0x4008129d:0x3ffe3c20 0x40078d12:0x3ffe3c40 0x40078dc5:0x3ffe3c70 0x4007903a:0x3ffe3cb0 0x40079182:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...

Backtrace: 
0x7fffffff: ?????
0x4008104c: start_cpu0_default at /components/esp32/./cpu_start.c:328
0x4008129d: call_start_cpu0 at /components/esp32/./cpu_start.c:203
0x40078d12: set_cache_and_start_app at /components/bootloader/subproject/main/./bootloader_start.c:637
0x40078dc5: unpack_load_app at /components/bootloader/subproject/main/./bootloader_start.c:584
0x4007903a: bootloader_main at /components/bootloader/subproject/main/./bootloader_start.c:546 (discriminator 1)
0x40079182: call_start_cpu0 at /components/bootloader/subproject/main/./bootloader_start.c:128
```

and another run with more debug:

```
I (249) esp_image: segment 0: paddr=0x00400020 vaddr=0x3f400020 size=0xbfffc (786428) map
I (491) esp_image: segment 1: paddr=0x004c0024 vaddr=0x3ffb0000 size=0x03158 ( 12632) load
I (496) esp_image: segment 2: paddr=0x004c3184 vaddr=0x40080000 size=0x00400 (  1024) load
I (499) esp_image: segment 3: paddr=0x004c358c vaddr=0x40080400 size=0x0c8a8 ( 51368) load
I (526) esp_image: segment 4: paddr=0x004cfe3c vaddr=0x400c0000 size=0x00064 (   100) load
I (526) esp_image: segment 5: paddr=0x004cfea8 vaddr=0x50000000 size=0x00014 (    20) load
I (533) esp_image: segment 6: paddr=0x004cfec4 vaddr=0x00000000 size=0x0014c (   332) 
I (541) esp_image: segment 7: paddr=0x004d0018 vaddr=0x400d0018 size=0xa246c (664684) map
I (755) boot: Loaded app from partition at offset 0x400000
I (755) boot: Disabling RNG early entropy source...
D (756) boot: Mapping segment 0 as IROM
D (759) boot: seg addr: 00400020 load_addr: 3f400020 size: 000bfffc
D (766) boot: Mapping segment 7 as DROM
D (770) boot: seg addr: 004d0018 load_addr: 400d0018 size: 000a246c
D (776) boot: calling set_cache_and_start_app
D (780) boot: configure drom and irom and start
D (785) boot: d mmu set paddr=004d0000 vaddr=400d0000 size=664684 n=12
D (792) boot: rc=0
D (794) boot: rc=0
D (796) boot: i mmu set paddr=00400000 vaddr=3f400000 size=786428 n=12
D (802) boot: rc=0
D (805) boot: rc=0
D (807) boot: start: 0x40081158
```

sinope-db, 2019-01-18:
I'd also like to point out that according to the Reference Manual section 1.3.1, the naming used for DROM and IROM seem to be reversed (drom_xxx objects used for IROM address range and vice-versa). It does not seem to cause a problem, it's just confusing.

https://github.com/espressif/esp-idf/blob/7e0ea6b05db8ad5d1a8dc183414ca8cd456ec934/components/bootloader/subproject/main/bootloader_start.c#L561

```cpp
// This is IROM address range
        if (header->load_addr >= SOC_IROM_LOW && header->load_addr < SOC_IROM_HIGH) {  
            if (drom_addr != 0) {
                ESP_LOGE(TAG, MAP_ERR_MSG, "DROM");
            } else {
                ESP_LOGD(TAG, "Mapping segment %d as %s", i, "DROM");
            }
            drom_addr = data->segment_data[i];
            drom_load_addr = header->load_addr;
            drom_size = header->data_len;
        }
// This is DROM address range
        if (header->load_addr >= SOC_DROM_LOW && header->load_addr < SOC_DROM_HIGH) { 
            if (irom_addr != 0) {
                ESP_LOGE(TAG, MAP_ERR_MSG, "IROM");
            } else {
                ESP_LOGD(TAG, "Mapping segment %d as %s", i, "IROM");
            }
            irom_addr = data->segment_data[i];
            irom_load_addr = header->load_addr;
            irom_size = header->data_len;
        }
```

igrr, 2019-01-19:
Hi @sinope-db, a fix for both of these issues has been prepared and will be merged into the master branch some time next week, then backported to v3.1/v3.2 branches. Do you also need it to be in v3.0 branch, or you plan to update to v3.1?

projectgus, 2019-01-23:
The fix we have includes an update to esptool so it will not generate images that will crash, even on older bootloaders.

While waiting for us to merge and backport the fix to v3.0, you can update the [esptool submodule to to v2.6](https://github.com/espressif/esptool/releases/tag/v2.6) in your IDF installation. This should work without other modifications to IDF, and the .bin images will be padded so that they don't trigger the bootloader bug.

igrr, 2019-01-24:
Fixed on master branch with https://github.com/espressif/esp-idf/commit/07645955a269a8db0504952d11f67d08f46af4cd.

closed 2019-01-24 by igrr