espressif/esp-idf #2970: SPI Master: received data is shifted

mtmsweb, 2019-01-19:
Using SPI Master driver received data is always 1bit shifted (0x80 instead 0x00).
For tests: slave device is TI's ADC- ADS1220. Datasheet link: http://www.ti.com/lit/ds/symlink/ads1220.pdf Timing about page 38. Device works in SPI Mode 1 and I am testing 1MHz speed.
I am writing test registers and then reading their value. Received bytes are 1 bit shifted.
Without reseting sleve device, reading registers using the same chip but programmed with arduino, I receive valid data.
Test code:

	spi_device_handle_t spi;
	char * rx_buffer;

	void spi_init() {
		spi_bus_config_t buscfg = {
			.miso_io_num=PIN_NUM_SPI_MISO,
			.mosi_io_num=PIN_NUM_SPI_MOSI,
			.sclk_io_num=PIN_NUM_SPI_CLK,
			.quadwp_io_num=-1, .quadhd_io_num=-1,
		};
		spi_device_interface_config_t devcfg = {
			.spics_io_num=PIN_NUM_SPI_ADC_CS,
			.clock_speed_hz=1*1000*1000, // 1MHz
			.mode=1, // SPI MODE 1
			.command_bits=8, // 8-bit command length
			.address_bits=0, // no address
			.queue_size=20,
			//.input_delay_ns = 25 // minimum of td(SCCS) = 25ns
		};
		ESP_ERROR_CHECK(spi_bus_initialize(VSPI_HOST, &buscfg, 1/*dma_chan - 0 no DMA*/));
		ESP_ERROR_CHECK(spi_bus_add_device(VSPI_HOST, &devcfg, &spi));
	}
	void writeSingleRegister(uint8_t address, const uint8_t value) {
		spi_transaction_t transaction;
		memset(&transaction, 0, sizeof(transaction));
		transaction.cmd = WREG | (address<<2);
		transaction.length = 8;
		transaction.tx_buffer = &value;
		assert(spi_device_polling_transmit(spi, &transaction) == ESP_OK);
	}
	void readSingleRegister(uint8_t address) {
		spi_transaction_t transaction;
		memset(&transaction, 0, sizeof(transaction));
		transaction.cmd = RREG | (address<<2);
		transaction.length = 8;
		transaction.rx_buffer = rx_buffer;
		assert(spi_device_polling_transmit(spi, &transaction) == ESP_OK);
		ESP_LOG_BUFFER_HEX("readSingleRegister:", rx_buffer, 1 );
	}

	void spi_task(void *pvParameters) {
		rx_buffer = heap_caps_malloc(4, MALLOC_CAP_DMA);

		spi_init();
		while (1) {
			ESP_LOGI(TAG, "DEFAULT_CONFIG_REG.. : ");
			writeSingleRegister(0, DEFAULT_CONFIG_REG0);
			writeSingleRegister(1, DEFAULT_CONFIG_REG1);
			writeSingleRegister(2, DEFAULT_CONFIG_REG2);
			writeSingleRegister(3, DEFAULT_CONFIG_REG3);
			vTaskDelay(1000 / portTICK_RATE_MS);

			ESP_LOGI(TAG, "Read: ");
			readSingleRegister(0);
			readSingleRegister(1);
			readSingleRegister(2);
			readSingleRegister(3);
			vTaskDelay(5000 / portTICK_RATE_MS);
		}
	}

malbrook, 2021-04-19:
I recently had this problem using two ESP32 devices to work with an FRAM for data transfer
CPU1 uses vspi settings and writes to the FRAM using SPI_MODE0, CPU2 reads from the FRAM using HSPI settings, good data goes into CPU1 and comes out garbaged on CPU2, eventually using a logic analyser I could see that the data leaving CPU1 going into the FRAM  and the data coming out and going into CPU2 were all good, (used tristate drivers between CPU and FRAM) but discovered the garbage was actually 1 bit left shift on data displayed by CPU2.
Solution was actually realy simple, changed just CPU2 to SPI_MODE1 and data is now fine at all points, so to me it looks like all this 1 bit shift messages on the Internet for ESP32 is down to a failure to correctly implement the MODES on the HSPI interface.
  
VSPI is 18,19 and 23 with select on 5
HSPI is 12,13 and 14 with CS on 15.
Enclosed images show the three sets of SPI Data from CPU1 and CPU2 and the FRAM behind the tristate drivers, these were taken before resetting the MODE on CPU2 and show the data is correct for all three interfaces.

[image]
[image]
[image]

7heaven, 2023-06-08:
I encountered same issue myself, I change mode to 2 and it works