espressif/esp-idf #2972: ESP32 i2s_write - Idle1 task starved?

uncle-yong, 2019-01-20:
Hello ESP-IDF team,

I have a ported a little program to ESP32 and one issue on this is I have no control over when I can put in more samples to the i2s_write after the batch of buffers are being streamed to the I2S's codec.

Worse of all, one of the Idle tasks had been starved and it kept shouting to me the WDT hasn't been reset. Shutting the Idle WDT doesn't help much here as I have other stuff at the background to worry about. 

For example:
> // Generate Samples here at Task A in Core no. 1:
> while (1)
>     {
>         GenerateSamplesHere(audio_buffer);
>         i2s_write(I2S_NUM_0, audio_buffer, sizeof(audio_buffer), &bytesWritten, portMAX_DELAY);
>     }
> 
Console output after about 6~8 seconds:
> [0;31mE (14243) task_wdt: Task watchdog got triggered. The following tasks did 
> not reset the watchdog in time:[0m                                           
> [0;31mE (14243) task_wdt:  - IDLE0 (CPU 1)[0m                               
> [0;31mE (14243) task_wdt: Tasks currently running:[0m                       
> [0;31mE (14243) task_wdt: CPU 0: TaskB[0m                         
> [0;31mE (14243) task_wdt: CPU 1: TaskA[0m
> 

Could it be possible if the task or function would yield or wait until the I2S has almost finished streaming the samples, and then place in some more samples instead of just continuously pushing into the I2S buffers?

uncle-yong, 2019-02-02:
Hi there,

> Is it possible that there is another task running which is also monopolizing the CPU?

I have a summary of available tasks at that time:
1.) Task No.1 processing and pushing samples to the I2S module.
- The processing takes 2.217ms for 256 samples.

2.) Task No.2 has a function that is being polled every 1ms (MIDI requirements).

Would (2) actually affect this too? Even if I remove (2) the Idle1 task was still starved.

wilhelmw201, 2020-07-22:
Hi, sorry for the late response, is your problem solved now? 

`while (1)
{
GenerateSamplesHere(audio_buffer);
i2s_write(I2S_NUM_0, audio_buffer, sizeof(audio_buffer), &bytesWritten, portMAX_DELAY);
}`
If (quote) generating the samples is taking longer than transmitting them, then the loop will be problematic. This loop may also be problematic if you have too large a buffer, I guess? Then i2s_write never fills the buffer and blocks before the watchdog causes a reset. (Because the idle task only runs if all other tasks on that CPU are blocked. If this loop never blocks at i2s_write, then the CPU will never go in idle task.) You can call explicitly vtaskdelay if you become too annoyed.

To avoid stutters you need to make data quickly so that the i2s doesn't have to use old data. It depends on your settings and you can calculate it yourself. For example, if you are using 2 channels and 16 bits and 36kHz sample rate, you need to make 36k*2*16bits per second. 
You can also set tx_desc_auto_clear in the config struct to 1 to make the i2s (try) to send nothing, instead of sending noise, when it runs out of data.

nyh-workshop, 2020-07-26:
Hello @wilhelmw201 and ESP team,

Thanks for the assistance and apologies for the late reply. I have not been working on this issue for the time being - I highly suspect that the generating samples part require more computational power (preferably a Raspberry Pi) and could not be done in the ESP32 at a shorter time. I have already found an alternative in the meantime.

closed 2020-07-26 by nyh-workshop