espressif/esp-idf #2973: RTC retention registers

ammaree, 2019-01-20:
We need a small space to store a crash restart counter in order to keep track of the number of times the running firmware has rebooted. We use this value to try make an informed decision regarding possibly reverting to a previous set of firmware.

We are currently using nvs_set_u32() to store this value but in extreme cases this can lead to excessive flash writes. We have looked at and tested using the RTC_CNTL_STORE0_REG to store this value and it works perfectly so far. We have also searched the TRM as well as the IDF and cannot find where/how this register might be used since it is indicated as "reserved"

We do not necessarily require a 32bit location (although it would be nice), even an 8 bit size would be sufficient. What is critical is that the value is retained across any type of restart or sleep, other than a power off/on restart.

Can any clarity please be provided on whether this register can be used by applications. If not, any alternative location that can be used?

Thanks

KonstantinKondrashov, 2019-01-22:
I think you can use a `RTC_DATA_ATTR ` attribute like in this example: https://github.com/espressif/esp-idf/blob/master/examples/protocols/sntp/main/sntp_example_main.c#L49. Is it suitable for you?

Regarding using a `RTC_CNTL_STORE0_REG ` I found the note: `Some Rtc memory and registers are used, in ROM or in internal library. Please do not use reserved or used rtc memory or registers.` https://github.com/espressif/esp-idf/blob/master/components/esp32/include/rom/rtc.h#L40

AmrutaCh, 2019-01-22:
@ammaree Implemented this feature last week only. 
First I tried `RTC_NOINIT_ATTR` solution suggested by @igrr, but ran into some issues. I don't remember the issue now. But [this](https://www.esp32.com/viewtopic.php?t=4931) solution worked for me and I could store last 6 program counter values before crash with some additional info like crash reason, reboot count, etc

Alvin1Zhang, 2019-03-18:
@ammaree Hi, would you share if any updates for this issue or your issue has been resolved? Thanks.

ammaree, 2019-03-18:
@Alvin1Zhang 

The suggestion to use RTC_DATA_ATTR or RTC_NOINIT_ATTR are things we were aware of. We are already using the full remaining 0x1000 for 2 purposes:
a) preserving M90E26 sensor data across reboots, so as not to lose energy readings taken before reboot
b) console output buffer to preserve last syslog messages across reboots

We were looking for some additional storage in RTC registers but it seems this is not available.

Unless some additional info on the use of some RTC registers can be made available, you can close this issue.

ammaree, 2019-05-03:
@Alvin1Zhang @igrr 

The suggestion to use RTC_NOINIT_ATTR is not a solution. When the device reboots as a result of a WATCHDOG timeout, the reset count is cleared and thus the objective cannot be achieved.

So the questions comes back to an RTC register. The RTCCNTL.rtc_store{0-3} and RTCCNTL.store{4-5} registers seem to be in use somewhere, RTCCNTL.{6-7} registers seem unused.

a) Can either of these registers be made available, reserved for application use?
b) Alternatively, can a bit-mapped portion of 1 of the other registers be made available, just need 4 bits to store a reboot counter value 0->F that we use to control firmware revert decisions..

But the register value MUST be preserved across anything except a power failure or brownout.

Thanks

igrr, 2019-05-06:
Hi @ammaree, I have checked that RTC_NOINIT_ATTR variables are preserved across the following kinds of watchdog reset: task watchdog, interrupt watchdog (both software triggered and hardware triggered), RTC watchdog. Can you please provide a code example which demonstrates the issue you are observing?

Alvin1Zhang, 2019-05-16:
@ammaree Hi, would you please help provide a code example which demonstrates the issue you are observing? Thanks.

ammaree, 2019-05-17:
I Have reverted my code to the version using the RTC Slow RAM as opposed to trying RTC registers. The code snippets are provided below. I will now monitor device restarts to verify the operation and will provide feedback as soon as I have.

`
/* Used to keep track of unplanned/spurious restarts. Handling is based on following logic
 * ------ Reason ------|---- Action ----|
 * Firmware upgrade		reset to 0
 * Scheduled restart	unchanged
 * Unplanned restart	increment
 *
 * RTC_CNTL_STORE6_REG	Deep Sleep handler address
 * RTC_CNTL_STORE7_REG	RTC memory CRC
 */

RTC_NOINIT_ATTR	uint32_t	RebootCounter ;

uint32_t halSTORAGE_RebootCounterRead(void) { return RebootCounter ; }

uint32_t halSTORAGE_RebootCounterUpdate(int32_t xValue) { return (RebootCounter = (xValue != 0) ? (RebootCounter + xValue) : 0) ; }

void halSTORAGE_Test(void) {
	cprintf("Counter Before=%u  ", halSTORAGE_RebootCounterRead()) ;
	cprintf("Counter After=%u\n", halSTORAGE_RebootCounterUpdate(1)) ;
	vRtosSetStatus(flagAPP_RESTART) ;
}
`

Code calling the above functions:
---------------------------------
`
void	halFOTA_UpdateBootInfo(void) {
	part_info_t	sPI ;
	size_t		SizeBlob = sizeof(part_info_t) ;
	halSTORAGE_ReadBlob(halSTORAGE_STORE, halSTORAGE_KEY_PART, &sPI, &SizeBlob) ;
	esp_reset_reason_t xReason = esp_reset_reason() ;
	uint32_t CurPart = halFOTA_GetRunningNumber() ;
	uint32_t Counter = halSTORAGE_RebootCounterRead() ;
	// TEMPORARY FIX to get the variable initialized...
	if (Counter > halSTORAGE_REVERT_THRESHOLD) {
            Counter = halSTORAGE_RebootCounterUpdate(0) ;
        }
	if ((sPI.xx[CurPart].version != DEV_FW_VER_NUM) || (sPI.xx[CurPart].tBuild != BuildSeconds)) {
		sPI.xx[CurPart].version	= DEV_FW_VER_NUM ;		// new version FW, first (successful) boot after upgrade
		sPI.xx[CurPart].tBuild	= BuildSeconds ;
		sPI.xx[CurPart].tBoot	= xTimeStampAsSeconds(sTSZ.usecs) ;
		uint32_t PrevPart = CurPart > 1 ? CurPart - 1 : fotaMAX_OTA_PARTITIONS-1 ;
		/* Using as dynamic storage for restart count. When new firmware is run 1st time
		 * we need to take the current dynamic counter value and save that as the static
		 * counter value for the previous firmware, then reset dynamic & static counters */
		sPI.xx[PrevPart].cBoot	= Counter ;
		halSTORAGE_WriteBlob(halSTORAGE_STORE, halSTORAGE_KEY_PART, &sPI, sizeof(part_info_t)) ;
		// reset counter for current (new) FW to 0
		sPI.xx[CurPart].cBoot	= halSTORAGE_RebootCounterUpdate(0) ;
		SL_NOT("Boot P#%u new FW v" DEV_FW_VER_STR " built %R", CurPart, xTimeMakeTimestamp(BuildSeconds, 0)) ;

	} else if ((xReason != ESP_RST_SW) && (xReason != ESP_RST_POWERON)) {
		/* unscheduled restart (HW, Watchdog or similar) thus increment dynamic
		 * counter then log current boot statistics as a warning */
		Counter = halSTORAGE_RebootCounterUpdate(1) ;		// increment counter
		uint64_t	tSinceFirstBoot		= sTSZ.usecs - xTimeMakeTimestamp(sPI.xx[CurPart].tBoot,0) ;
		uint64_t	tAvgSinceFirstBoot	= tSinceFirstBoot / (uint64_t) Counter ;
		SL_WARN("P#%u  ReBoot#%u  %s  tTotal=%!R  tAvg=%!R",
									CurPart,
									Counter,
									halMCU_GetResetReason(),
									tSinceFirstBoot,
									tAvgSinceFirstBoot) ;
		xHttpClientCoredumpUpload(NULL) ;

		/* Add logic to make accurate assessment of the stability of the running software.
		 * In order for calculations to work SNTP must have run to ensure time is correct !!! */
		if (Counter == halSTORAGE_REVERT_THRESHOLD) {
			// before we change the partition, update the info for current running partition.
			sPI.xx[CurPart].cBoot	= Counter ;
			halSTORAGE_WriteBlob(halSTORAGE_STORE, halSTORAGE_KEY_PART, &sPI, sizeof(part_info_t)) ;
			CurPart = halFOTA_RevertToPreviousFirmware() ;
		}
	} else {
		// normal restart due to ESP_RST_SW or ESP_RST_POWERON, no need to update restart counter
	}
}
`

ammaree, 2019-07-13:
Thanks, got this one sorted out.
Closing now.

closed 2019-07-13 by ammaree

cyberman54, 2021-03-21:
I found that RTC_NOINIT_ATTR variables loose their values and are not initialized, after an OTA update (app1 partition written) and esp_restart().

This makes it hard to detect a firmware update after a soft (non power cycled) restart.

/cc @igrr

igrr, 2021-03-21:
@cyberman54 Please open a new issue and fill in the issue template.

Note that if you are re-building the application, then the addresses of RTC_NOINIT_ATTR variables may change between builds. So RTC_NOINIT_ATTR is not suitable for passing data between different builds of an application (such as two different OTA versions). You can check the addresses of these variables using the .map file found in the build directory.

cyberman54, 2021-03-22:
@igrr thanks for this explanation! It saved me a lot of time, because i did not think of this while debugging. Thus, it's not a bug, and so i shouldn't open issue, right?

But maybe this should be addressed in the ESP32 documentation?

I need to detect OTA update after a (software) reset, so need to set a flag which survives a reset between different builds.

This could be done using NVRAM to hold the flag.

Would using registers in RTC memory, like shown above, also be an appropriate solution for this?