espressif/esp-idf #2976: WDT reset  with WiFi

jstar88, 2019-01-21:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    v3.0.7
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Power Supply:         [external 3.3V]

## Problem Description
While WiFi is active and connected to MQTT broker, a WDT reset occur after a while. (5minuts)
Seems this behavior happens only 1 time after external reset or start-up

## Debug Logs
```
I (2482921) system: free heap = 91820 bytes
I (2483921) system: free heap = 91664 bytes
I (2484922) system: free heap = 91820 bytes
I (2485922) system: free heap = 91820 bytes
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4548
load:0x40078000,len:0
load:0x40078000,len:13092
entry 0x40078d24
W (29) boot: PRO CPU has been reset by WDT.
W (29) boot: WDT reset info: PRO CPU PC=0x40081737
0x40081737: .check_dport_access_end at /home/nicola/esp/esp-idf/components/esp32/./dport_panic_highint_hdl.S:182

W (29) boot: WDT reset info: APP CPU PC=0x4008155a
0x4008155a: esp_dport_access_stall_other_cpu_start at /home/nicola/esp/esp-idf/components/esp32/./dport_access.c:102

I (363) cpu_start: Pro cpu up.
I (364) cpu_start: Starting app cpu, entry point is 0x40081070
```

KonstantinKondrashov, 2019-01-22:
Could you provide more information about it? (Your application if it is possible or a description of how we can reproduce it, the sdkconfig file will be useful as well). Maybe you have some changes in IDF?

When it happens do you see only this log or you also see other logs?
```
...
0x40081737: .check_dport_access_end at /home/nicola/esp/esp-idf/components/esp32/./dport_panic_highint_hdl.S:182
...
0x4008155a: esp_dport_access_stall_other_cpu_start at /home/nicola/esp/esp-idf/components/esp32/./dport_access.c:102
```

jstar88, 2019-01-23:
i don't see anyother log.
My IDF differs only about MQTT library 
I uploaded sdkconfig in first topic, sorry i don't know actually how to reproduce it

jstar88, 2019-01-29:
@KonstantinKondrashov 
i updated first message, as you can see the issue is not heap size-related.
Indeed I checked uxTaskGetStackHighWaterMark() for each thread every 100ms and printed the minimum value.
 free heap is retrived with esp_get_free_heap_size()

KonstantinKondrashov, 2019-02-04:
Hi @jstar88.
Could you try this path to figure out a root cause of reset? This path adds a counter of time into the dport interrupt. If operation with dport is so long (lockup) then it will finish waiting and call the panic handler before IWDT happen, it will print a backtrace. I hope we can see more information about this issue.
[0001-esp32-Add-counter-in-handle_dport_access_int-to-cont.zip](https://github.com/espressif/esp-idf/files/2826775/0001-esp32-Add-counter-in-handle_dport_access_int-to-cont.zip)

jstar88, 2019-02-07:
I just discovered a "break" condition in loop of my wifi thread.
In such case, there wasn't any vTaskDelay() and so after a while (some conditions?) a wdt reset occur.
I solved this problem adding the delay

KonstantinKondrashov, 2019-02-07:
Thanks for the reply. Do you mean a function like this https://github.com/espressif/esp-mqtt/blob/master/examples/mqtt_ws/main/app_main.c#L73. Do you have yourself implementation this function? Could you provide your wifi thread? 

Did you use the patch above? Was it helpful to understand the reason or not?

jstar88, 2019-02-07:
In my case, i have a specific thread "wifi_manager"
wifi_init() is called from this thread, and there is a loop where i manage commands 
pretty similar to this 
[https://github.com/tonyp7/esp32-wifi-manager/blob/master/wifi_manager/main/wifi_manager.c](url) 

i solved before try your patch

closed 2019-02-11 by jstar88