espressif/esp-idf #2978: I2S documentation refers to non-existent i2s_adc_start/stop() functions

detly, 2019-01-21:
The [documentation for I2S](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/i2s.html), specifically the [`is2_read()`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/i2s.html#_CPPv28i2s_read10i2s_port_tPv6size_tP6size_t10TickType_t) function, refers to functions `i2s_adc_start` and `i2s_adc_stop`:

> **Note:** If the built-in ADC mode is enabled, we should call `i2s_adc_start` and `i2s_adc_stop` around the whole reading process, to prevent the data getting corrupted.

These functions don't appear anywhere in the source code (for I2S or the whole IDF library as far as I can tell).

itaisod, 2019-02-18:
Indeed.

mehmetcanbalci-Notrino, 2020-01-28:
They are doing deliberately

giPeteR, 2020-04-15:
I'm having a hard time getting the **I2S ADC DMA** to work in >4.0.
Confusing documentation still mentioning _i2s_adc_start and i2s_adc_stop_ doesn't help:
https://docs.espressif.com/projects/esp-idf/en/release-v4.1/api-reference/peripherals/i2s.html

The example:
_\examples\peripherals\i2s_adc_dac_
works in **4.0**, but in **4.2-dev** the ADC works about every 4'th reset, DAC never works.

koobest, 2020-04-15:
Hi,@detly
We will update the DOC
@giPeteR 
Can you please open a new issue?

thanks !!

closed 2020-06-12 by igrr