espressif/esp-idf #2980: Data formats and API for I2S direct-from-ADC could be better documented

detly, 2019-01-22:
## Environment

- Development Kit: Wemos Lolin D32 
- Module or chip used:  ESP32-WROOM-32
- IDF version: `v3.3-beta1-223-ga62cbfec9`
- Build System: Cmake
- Compiler version: `1.22.0-80-g6c4433a`
- Operating System: Linux (Ubuntu 18.04)
- Power Supply: USB

## Problem Description

This is a little more vague than a more technical problem/bug, because I'm not sure whether there's really any bug in the ESP IDF libraries, whether the documentation should be clearer, or whether I've misunderstood certain things. So I'm happy for feedback that helps me narrow down the scope of what I'm reporting here. I've **bolded** specific questions or suggestions.

I'm trying to use I2S direct-from-ADC functionality via the ESP IDF libraries, specifically `driver/i2s.h`. I'm having a lot of trouble figuring out how I'm meant to use the library functions.

Let's start with the sample format itself. I was using all 12 bits of the ADC resolution, so I would have expected the DMA data to range from `0x0000` to `0x0fff` (after decoding from MSB or LSB format, whichever I selected). However, I wasted a lot of time trying to figure out why I was getting values like `0x6abc` until I realised it was always a `0x6` in the highest four bits, and a bit of digging found this (from p583 of the [ESP32 Technical Reference Manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf)):

> ESP32 supports up to a 12-bit SAR ADC resolution. The 16-bit data
> in DMA is composed of the ADC result and some necessary information
> related to the scanning mode:
>  - For single mode, only 4-bit information on channel selection is
>    added.

**Suggestion:** I think this information needs to be in the IDF docs. I'm quite happy to be referred to technical documentation from an API, but the reference needs to be made, in my opinion. Otherwise I feel like I can only approach the IDF docs after having read and absorbed all the more technical documentation, which detracts from having the library in the first place. But in *this specific case*, I think it's an odd enough feature that it deserves to be mentioned in the high-level docs.

Secondly: I began by testing with a single microphone, so I selected `I2S_CHANNEL_FMT_ONLY_LEFT`. The signal I decoded on my client was effectively halved in frequency, so I quickly figured out I still needed to deinterleave the stereo signal (ie. take every second decoded 16-bit value, after masking with `0x0fff`). I have no idea what was in the "other" channel, since I was only using one microphone and ADC, but it looked very similar to my primary signal (not identical though!). Same with `I2S_CHANNEL_FMT_ALL_LEFT`, except now half my signal was zeroed. (Disclaimer: I may have these switched around, I'm going from memory here.) Changing from `LEFT` to `RIGHT` seemed to make absolutely no difference.

There are tables on p308 of the [technical reference manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) that describe what the register `I2S_TX_CHAN_MOD` does — **question:** is this what corresponds to the `i2s_channel_fmt_t` `enum`? **Suggestion:** could that also please be documented more explicitly in the IDF docs? **Question:** can I actually do mono ADC reads and transmission, to avoid transmitting unnecessary bytes in mono mode?

Thirdly: I now want to use stereo conversion. My first **question:** how do I set the second ADC for the second microphone? There's only [`i2s_set_adc_mode()`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/i2s.html#_CPPv216i2s_set_adc_mode10adc_unit_t14adc1_channel_t) which takes a single ADC index. How do I populate stereo data from two ADCs? Again, if I need to do this directly in the registers, it should be referenced from the IDF docs.

Finally, a couple of little **questions: what are the `dma_buf_count` and `dma_buf_length` members of `i2s_config_t`? I can't find any reference to this in the technical reference manual. Does the `sample_rate` member need to be the rate at which I want to sample each channel with the ADCs, or the total sample rate being put into DMA (ie. so it doubles if I switch to stereo)?

I have a [test project](https://gitlab.com/detly/i2s_test) that's public on Gitlab that I'm using to test out basic I2S functionality, so if you want a code base to quickly verify something, feel free to use that.

montaguk, 2019-02-22:
I have been experimenting with the I2S interface a bit and faced many similar issues/questions as you. 
 Ultimately, my conclusion is that the I2S driver currently only supports a single channel.  The docs don't quite reflect this, but there is at least one [comment in the driver itself](https://github.com/espressif/esp-idf/blob/master/components/driver/i2s.c#L865) that indicates that only a single channel is supported.

I couldn't for the life of me figure out a way to make use of the multi-channel pattern tables via the I2S w/o modification to the adc/i2s drivers.  I don't have answers to all of your questions, but was able to piece together some info that might be useful to you so I figured I'd share it back.

From what I can tell, the main issue is with the current idf implementation of [`adc_i2s_mode_init`](https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/components/driver/rtc_module.c#L1398).  Currently it can only configure the pattern table with a single channel and defaults to 11DB attenuation.  I was able to work around this issue by exposing [`adc_set_i2s_data_len`](https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/components/driver/rtc_module.c#L1345) and [`adc_set_i2s_data_pattern`](https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/components/driver/rtc_module.c#L1360) to support calling from external code.  In my application code I set the i2s in adc mode, and then call these two functions to reconfigure the pattern table for dual channel input:
```
//install and start i2s driver
i2s_driver_install(i2s_num, &i2s_config, 0, NULL);
//init ADC pad
i2s_set_adc_mode(I2S_ADC_UNIT, I2S_ADC_CHANNEL_DATA);

// Expand default pattern with the clk channel at index 1
// Data is assigned to pattern index 0 by i2s_set_adc_mode
adc_set_i2s_data_len(I2S_ADC_UNIT, 2);
adc_set_i2s_data_pattern(I2S_ADC_UNIT, 0, I2S_ADC_CHANNEL_DATA, ADC_WIDTH_BIT_12, ADC_ATTEN_DB_0);
adc_set_i2s_data_pattern(I2S_ADC_UNIT, 1, I2S_ADC_CHANNEL_CLK, ADC_WIDTH_BIT_12, ADC_ATTEN_DB_0);
```
Finally, I found that when calling `i2s_adc_enable(EXAMPLE_I2S_NUM);` the pattern table is reconfigured by the library's [`_i2s_adc_mode_recover`](https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/components/driver/i2s.c#L1183), so (for now) I've just commented this call out in the library.

> Thirdly: I now want to use stereo conversion. My first **question:** how do I set the second ADC for the second microphone? There's only [...]

The pattern table seems to be the intended way to do this from the ESP32 TRM.  The workaround described above for the pattern table should let you configure up to sixteen channels for sequential sampling.

> Finally, a couple of little **questions:** what are the `dma_buf_count` and `dma_buf_length` members of `i2s_config_t`? I can't find any reference to this in the technical reference manual.

I believe these are used to configure/allocate the actual memory buffers used by the DMA to transfer data from the i2s controller to RAM.  In my testing, it appears that the DMA interrupt isn't fired until a given DMA buffer is full, at which point the entire buffer is copied into the driver's queue to be read by `i2s_read`.  A larger buffer results in a longer duration between interrupts, more data to process at once, and increased latency due to the interrupt coming later in larger buffers.

For my specific application, I need to process/respond to incoming ADC data quickly, so I have a larger number of small buffers.  If latency isn't an issue for you (ie. recording audio to disk) you'd probably get better overall performance by having a fewer number of large buffers.  Note that the library requires that you have at [least two and no more than 128](https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/components/driver/i2s.c#L1033).

One other thing I found is that the data returned from `i2s_read` seems to be inverted and all samples must be inverted before processing.  I tried calling [adc_set_data_inv](https://github.com/espressif/esp-idf/blob/master/components/driver/i2s.c#L865) with both TRUE and FALSE but neither seemed to have any impact on the resulting data, I still had to XOR every sample with 0xFFF.

detly, 2019-03-04:
@montaguk That is incredibly useful, great work deducing all of that.

**Espressif** ­— it would be very nice to have some answers to the above questions, even if it's not going to work out of the box any time soon, I need to be able to account for missing functionality and communicate the reason to other engineers.

detly, 2019-05-01:
ESP devs - it's been months now. I'd like to know whether this device is fit for the purpose I used it for, or whether I need to change my design to use a different device. Could you please shed some light on this?

detly, 2019-06-25:
@montaguk In your example code, I assume that `I2S_ADC_CHANNEL_DATA` stands in for one of the ADC channels being used, but what is `I2S_ADC_CHANNEL_CLK`?

montaguk, 2019-06-25:
Those names are a bit confusing w/o any context. I'm actually using the i2s ADCs to decode a proprietary protocol that has a DATA and a CLK signal.  Fom the standpoint of the i2s, it's just two separate input channels that could be used for any purpose.  The DATA and CLK only have meaning within the context of my application.

Hopefully that helps!

detly, 2019-06-25:
Thanks, that makes sense! Oh, two more things, in case you know — do you know if the sample rate in the initial `i2s_config_t` struct needs to be double its mono-mode value (to account for the extra channel), and whether it matters what the channel format is set to?

detly, 2019-06-27:
I finally got around to testing this but cannot get the channel order to make any sense. See #3686 .

Espressif devs — could we *please* have some info on this?

Splediferous, 2019-12-07:
I guess it is still sampling at half the sample rate. A mono channel @ 44.1Khz would actually be 22.05KHz after de-interleaving.

I made the same observation as you a year or so ago (https://github.com/espressif/esp-idf/issues/1911). I wanted to get 1MHz sampling rates but the best one can do is 1/2 that. What with the other issues and slow response, I was forced to go to a different device in the end.

Now coming to look at this again for a new project, I see improvements have been made somewhat but  it's obviously still interleaving a single signal.

Alvin1Zhang, 2021-07-27:
Thanks for reporting and sorry for late reply, this issue has been fixed, please help update to see if your issue gets resolved, will close now, feel free to reopen.

closed 2021-07-27 by Alvin1Zhang

wudicgi, 2021-12-15:
> Thanks for reporting and sorry for late reply, this issue has been fixed, please help update to see if your issue gets resolved, will close now, feel free to reopen.

How is this problem solved? Could you please provide the commit hash or release notes?

wudicgi, 2021-12-15:
> Thanks for reporting and sorry for late reply, this issue has been fixed, please help update to see if your issue gets resolved, will close now, feel free to reopen.

@Alvin1Zhang How is this problem solved? I didn't see the adc.c was modified
Could you please provide the commit hash or release notes?