espressif/esp-idf #2981: issue about select uart

jade111, 2019-01-22:
Hi,
I have tested the uart with open file way like linux. But I can't find how to config Termios. I am care about the INLCR\IGNCR\ICRNL and so on. Can you add this feature?
Best regards,
jade

jade111, 2019-02-15:
Hi dobairoland,
How about it?
Best regards,

dobairoland, 2019-02-19:
Hi @jade111. Can you please provide more information as it is suggested by the issue template (https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? What IDF version are you using? Can you share a piece of code explaining what is wrong with the current behavior and what behavior do you expect?

The flags you are mentioning are already implemented (see https://github.com/espressif/esp-idf/blob/master/components/vfs/vfs_uart.c#L481).

dobairoland, 2019-04-03:
I'm closing this for lack of activity. Please feel free to reopen it if this is still an issue.

closed 2019-04-03 by dobairoland