espressif/esp-idf #2984: I cannot use AVRCP mode without initializing A2DP

Marc-Aurele, 2019-01-23:
I want to use only the AVRCP mode so i started with the following example : a2dp_gatts_coex
When i flash it onto my board, here is the ouput of bluetoothctl :

```
[bluetooth]# info 20:47:DA:FB:93:65
Device 20:47:DA:FB:93:65 (public)
	Name: ESP_COEX_A2DP_DEMO
	Alias: ESP_COEX_A2DP_DEMO
	Class: 0x002c0414
	Icon: audio-card
	Paired: no
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	RSSI: -50
```
That is working nicely enough so far - but now I want to get rid of the A2DP-part. My intuition was to just remove the esp_a2d_

I would have expected to now only see the UUIDs for "A/V Remote Control" - but there is just no UUIDs at all.

Regards,

Aurélien

Marc-Aurele, 2019-01-25:
Hi,

Is there someone who has already worked on a workaround for this issue. I really need AVRCP without AD2P ? If you have any idea, feel free to share :)

Best regards,

Aurélien

Marc-Aurele, 2019-01-28:
Here is the kind of configuration i want :
[image]
Do you think it is possible with the current library ?

Weijian-Espressif, 2019-02-27:
@blueMoodBHD
Please help answer this question, thanks.

blueMoodBHD, 2019-02-27:
@Marc-Aurele Sorry to tell you that we can't use AVRCP without A2DP, now.

Marc-Aurele, 2019-02-27:
Hi @blueMoodBHD 
Thank you very much for the reply. Indeed, we were suspecting that, as a result, we have switched on HID over GATT profil.

closed 2019-03-14 by Alvin1Zhang

jotkali, 2022-10-06:
Is it still the case that the ESP family cannot act as the AVRCP remote controller without consuming the audio as well - implementing BT spec 2.3.1.1 mentioned above? thx