espressif/esp-idf #2985: UDP Broadcast Performance is Worse in AP mode than STA Mode

sumps, 2019-01-24:
## Environment

- Module or chip used:  ESP32-WROVER and ESP32-WROVER-I

- IDF version V3.1.2
  
- Build System:         Make
- Compiler version V5.2.0

- Operating System:     Linux

- Power Supply:         external 3.3V

## Problem Description

We have a relatively simple ESP32 application ESP-IDF (FreeRTOS) that converts Serial Data in to TCP or UDP Packets. By default we are using the ESP32 in AP mode but also allow the user to select STA mode.

A TCP connection works perfectly in AP and STA mode but we have found that there are a large number of UDP packets that are not transmitted in AP mode (25%) and a limited number perhaps 1-2% not transmitted in STA mode.

We have just one Mac connected to the ESP32 in AP Mode, we are using a Baud rate of 38400 and we are sending a burst of messages each with a counter and then a UDP client program, running on the Mac, uses the counter number to record which packets are not transmitted. We have also confirmed that the packets are not getting out by running Wireshark on the Mac.

We are using the SDK configuration settings that you used for the IPerf Example in which you recorded 30Mbit UDP throughput, which would be great if we can ensure all packets are transmitted.

We are using a UDP broadcast address of 255.255.255.255

This might be related to issue #1865 which was never resolved

TimXia, 2019-01-28:
@sumps Are you sure that there is enough memory for AP to send UDP packets which means `sendto` does not return `ENOMEM` ?

sumps, 2019-01-28:
@TimXia thanks for the reply, we have the same memory set for AP and STA mode, so are you worried that the WiFi TX Buffer is getting filled up while a task that is only carried out in AP mode is in progress.

Which memory setting should we change? Is it....

CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=64

This is the value we currently have set, what value would you recommend?

TimXia, 2019-01-28:
@sumps I think it is enough. Did you check the return value of `sendto`? Does it return -1 sometimes?

sumps, 2019-01-28:
@TimXia I will ask one of our developers to check this and report back ASAP

sumps, 2019-01-29:
@TimXia sorry for the delay in getting feedback to you, but my developer had been dragged off to work on another project - he has promised to look at this tonight. Are there any other things he should check as I would prefer for him to do everything at the same time to save time.

TimXia, 2019-01-30:
@sumps If call esp_wifi_set_ps(WIFI_PS_NONE) after calling esp_wifi_init() on the station side, will the issue happen?

sumps, 2019-01-30:
@TimXia my understanding is that the WiFi power save mode is only active in STA mode and in AP mode there is no power save. It is in AP mode where we are seeing the poor performance so are you suggesting that we test turning off power save mode when in STA mode (after esp_wifi_init()) to see if this causes worse performance in STA mode and therefore might identify that it is something in the Power Save NONE function that is causing the problem.

Sorry just trying to understand what we are trying to test for.

TimXia, 2019-01-30:
@sumps Sorry, I misunderstood you. Your scenario is that ESP32 which is in softap mode sends UDP broadcast packets to Mac. I'm not sure whether Mac is in power save mode. Would you please check the performance of UDP unicast packets? Maybe ESP32 is poor performance with UDP broadcast packets.

sumps, 2019-01-30:
@TimXia I thought there may have been a misunderstanding, but now I think you understand correctly our scenario. We are sending serial data to the ESP32 which it is converting to either TCP or UDP (broadcast) packets. Performance in TCP mode is fine and performance in UDP mode when the ESP-32 is in STA mode is acceptable (maybe 1-2% packet loss) but it is UDP in AP mode where we are seeing this "Stuttering" performance, with the ESP32 failing to output UDP packets every few seconds (around 20-25% packet loss). 

Our current testing is using a MAC in STA mode connected to the ESP32 in AP mode, but we are seeing other devices, LINUX, Android, WIndows, iOS have the same performance, so it is not a MAC issue. Please note that we have had products using an RN131C module since 2011, so we are experienced in this type of Serial to TCP/UDP communication.

TimXia, 2019-01-30:
@sumps So, performance of UDP unicast packets in AP mode is also poor?

sumps, 2019-01-30:
@TimXia I don't believe we have actually tested for this, as our product does not have a use case for Unicast. My understanding is a UDP Unicast and Broadcast packet is transmitted in the exact same way but the Destination Address in the header is the networks broadcast address i.e. 255.255.255.255 rather than a specific address of one device on the network.

Is there any reason that I have missed as to why the ESP32 should be slower at sending UDP Broadcast packets than Unicast?

TimXia, 2019-01-30:
@sumps If the UDP receiver has power save mode, ESP32 does not buffer broadcast UDP packets for it. If ESP32 sends UDP broadcast packets when receiver is in sleep state, the receiver will lose the packets. But ESP32 does buffer unicast UDP packets, the receiver will not lose any unicast packets. So, the performance of broadcast and unicast UDP packets may be different. It can be proved by a case for unicast.

TimXia, 2019-01-30:
@sumps Power save mode works on MAC layer. A device with power save mode informs AP whether it is in power save mode by sending NULL data frame with power management bit set or clear to AP. If AP know station is in sleep state, it will buffer both unicast and broadcast packets and send them when station wakes up.

sumps, 2019-01-30:
@TimXia in our scenario, the ESP32 in AP mode is transmitting UDP packets in one second bursts. What is a normal power save mode timeout on a receiver in STA mode? I would expect with the amount of UDP data we are sending that the receiver will never go to sleep.

I have confirmed with my developer that he saw no difference in UDP performance with the ESP32 set to send in Broadcast and Unicast AP modes, both modes gave the poor performance.

Is it not more likely to be another task like SSID transmission, Beacon, Mesh, etc. that is causing the UDP performance to intermittently stop?

sumps, 2019-01-30:
@TimXia we have done further tests looking at what the sendto function returns and it is always the number of bytes, never -1.

sumps, 2019-01-30:
@TimXia more feedback from my developer...

To be absolutely sure that the fault was not being caused by our serial port receiving code, I modified the code to generate the data strings rather than take them from the serial port. Injecting a message every 100ms, it was fine in TCP continuous messages flowing ever 100ms, but with UDP it was very erratic sometimes with bursts of 3-10 messages, then 300-400ms with nothing being transmitted and then it would start transmitting again but with those messages lost  - each message has a sequence number so we can see which ones are lost.

With the same code running in STA mode, the UDP and TCP are both fine.

So the messages are definitely getting on to the stack (no -1 returned by sendto) but in AP mode the messages are regularly being lost - never transmitted.

sumps, 2019-01-31:
@TimXia Happy New Year to you and your team at Espressif and look forward to continuing our discussions when you return from your holiday.

TimXia, 2019-02-01:
@sumps It has nothing to do with SSID transmission, Beacon, Mesh, etc. Could you please provide a demo code to us to reproduce the issue? Thus we can find the root cause soon.

TimXia, 2019-02-01:
@sumps Thanks. If you could provide a demo code, we can investigate it these two days before holiday.

sumps, 2019-02-01:
@TimXia do you need the sources or will a binary file be enough? How can I get this to you privately?

TimXia, 2019-02-01:
@sumps Please provide the source code and attach it here.

sumps, 2019-02-01:
@TimXia I have emailed you directly

akshar001, 2019-09-28:
Same issue

MaxwellAlan, 2019-11-22:
@akshar001 IDF V3.3 has solved this issue, you can update IDF version to V3.3 and do the test again, thanks.

Alvin1Zhang, 2019-12-02:
@sumps Thanks for reporting. @akshar001 Thanks for the updates. Did you have any chance to try IDF v3.3 and help share updates? Thanks.

Alvin1Zhang, 2020-01-20:
@sumps Thanks for reporting and @akshar001 for sharing the updates. We have fixed this issue, feel free to reopen if the issue still happens. Thanks.

closed 2020-01-20 by Alvin1Zhang