espressif/esp-idf #2986: Lock up in sdmmc_host_start_command causes Task WDT kick and abort

mill1000, 2019-01-24:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32
- IDF version: esp-idf v3.1.2 Release
- Build System: Make
- Compiler version: 
xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System: Windows
- Power Supply: Battery

## Problem Description
Lock up in sdmmc_host_start_command causes a Task WDT kick and abort (as configured) due to the following loop.
https://github.com/espressif/esp-idf/blob/ed1304146bb379c0a8dcf141710b42364feb4734/components/driver/sdmmc_host.c#L262-L264
This occurs occasionally when SDIO_DAT0 is randomly shorted to ground. Although not expected to occur in normal operation, it appeared during robustness tests to card errors. Suspect the SDMMC peripheral is getting into a strange state, the previous command is never sent and start_command is never cleared.

### Expected Behavior
sdmmc_host_start_command should timeout eventually and return an appropriate error code.

### Actual Behavior
sdmmc_host_start_command remains locked in busy loop until the watchdog aborts.

## Debug Logs
Here is the monitor leading up to the abort:
```
E (222201) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0xffffffff
E (222204) diskio_sdmmc: sdmmc_read_blocks failed (-1)
E (223211) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (223214) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (223216) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (223219) SDCARD: Failed to open monitor file /sdcard/data/M0124198.log. Errno: 2
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: SDCardTask
Aborting.
abort() was called at PC 0x400d2f3c on core 0
0x400d2f3c: task_wdt_isr at esp-idf/components/esp32/task_wdt.c:236

Backtrace: 0x4008ff83:0x3ffb05b0 0x400900db:0x3ffb05d0 0x400d2f3c:0x3ffb05f0 0x4008243d:0x3ffb0610 0x40173977:0x00000000
0x4008ff83: invoke_abort at esp-idf/components/esp32/panic.c:649

0x400900db: abort at esp-idf/components/esp32/panic.c:649

0x400d2f3c: task_wdt_isr at esp-idf/components/esp32/task_wdt.c:236

0x4008243d: _xt_lowint1 at esp-idf/components/freertos/xtensa_vectors.S:1105

0x40173977: esp_vApplicationWaitiHook at esp-idf/components/esp32/freertos_hooks.c:66
```

## Other items if possible

Here is the backtrace of the task in question:
```
#0  sdmmc_host_start_command (slot=1, cmd=..., arg=16384)
    at esp-idf/components/driver/sdmmc_host.c:262
#1  0x400f28bf in sdmmc_host_do_transaction (slot=1, cmdinfo=0x3ffc8fa0)
    at esp-idf/components/driver/sdmmc_transaction.c:155
#2  0x40175005 in sdmmc_send_cmd (card=0x3ffdc3bc, cmd=0x3ffc8fa0)
    at esp-idf/components/sdmmc/sdmmc_cmd.c:460
#3  0x4014f4b6 in sdmmc_read_sectors_dma (card=0x3ffdc3bc,
    dst=<optimized out>, start_block=<optimized out>, block_count=1)
    at esp-idf/components/sdmmc/sdmmc_cmd.c:965
#4  0x4014f890 in sdmmc_read_sectors_dma (block_count=1, start_block=16384,
    dst=0x3ffe43c4, card=0x3ffdc3bc)
    at esp-idf/components/sdmmc/sdmmc_cmd.c:945
#5  sdmmc_read_sectors (card=0x3ffdc3bc, dst=0x3ffe43c4, start_block=16384,
    block_count=1)
    at esp-idf/components/sdmmc/sdmmc_cmd.c:917
#6  0x4012eaa3 in ff_sdmmc_read (pdrv=<optimized out>, buff=0x3ffe43c4 "",
    sector=16384, count=1)
    at esp-idf/components/fatfs/src/diskio_sdmmc.c:39
#7  0x4012de7d in ff_disk_read (pdrv=0 '\000', buff=0x3ffe43c4 "",
    sector=16384, count=1)
    at esp-idf/components/fatfs/src/diskio.c:69
#8  0x4012bc50 in move_window (fs=0x3ffe438c, sector=16384)
    at esp-idf/components/fatfs/src/ff.c:1067
#9  0x4012c7a8 in dir_find (dp=0x3ffc90c0)
    at esp-idf/components/fatfs/src/ff.c:2465
#10 follow_path (dp=0x3ffc90c0, path=0x3ffe53cc "")
    at esp-idf/components/fatfs/src/ff.c:3090
#11 0x4012c8e8 in f_open (fp=0x3ffe5bd0, path=0x3ffe53c6 "", mode=18 '\022')
    at esp-idf/components/fatfs/src/ff.c:3565
#12 0x4012e548 in vfs_fat_open (ctx=0x3ffe436c, path=0x3ffe53c4 "",
    flags=521, mode=<optimized out>)
    at esp-idf/components/fatfs/src/vfs_fat.c:313
#13 0x400d7cc8 in esp_vfs_open (r=0x3ffc97d8, path=<optimized out>,
    flags=521, mode=438)
    at esp-idf/components/vfs/vfs.c:386
#14 0x4000bd6a in _open_r ()
#15 0x400d4885 in _fopen_r (ptr=0x3ffc97d8,
    file=0x3ffdc4d4 <error: Cannot access memory at address 0x3ffdc4d4>,
    mode=<optimized out>) at ../../../.././newlib/libc/stdio/fopen.c:141
#16 0x400d490f in fopen (
    file=0x3ffdc4d4 <error: Cannot access memory at address 0x3ffdc4d4>,
    mode=0x3f4102da "ab") at ../../../.././newlib/libc/stdio/fopen.c:181
#17 0x400ea8c5 in sdCardWriteMonitorBufferToFile (filename=...)
    at pingESP32/main/Storage/sdcard.cpp:711
#18 0x400ebee8 in sdCardTask (pvParameters=<optimized out>)
    at pingESP32/main/Storage/sdcard.cpp:1129
#19 0x4008d42c in vPortTaskWrapper (pxCode=0x400eba4c <sdCardTask(void*)>,
    pvParameters=0x0)
    at esp-idf/components/freertos/port.c:141
```

mill1000, 2019-04-24:
Saw a related lock up in sdmmc_host_clock_update_command in the following loop:
https://github.com/espressif/esp-idf/blob/ed1304146bb379c0a8dcf141710b42364feb4734/components/driver/sdmmc_host.c#L168-L183

This occurred after a software reset triggered by esp_restart(). I suspect the restart occurs during a SD transfer and leaves the host in a unexpected state. Again, these loops should have some timeout condition.

Resetting the peripheral before initialization seems like an acceptable work around.
```
periph_module_reset(PERIPH_SDMMC_MODULE);
```

ryankurte, 2020-11-08:
We're also seeing this on `v4.2-dev-2182-gdddcc2ede` and had identified the same forever-loop, would it be reasonable to put a retry counter in there to fail at SD init instead of hitting the watchdog?

closed 2023-04-13 by adokitkat