espressif/esp-idf #2988: http_server, server_port overide settings Not working

versamodule, 2019-01-25:
when I change the server_port from the default vaule of 80, it does not work on that port, but yet works on port 80 no matter what I set it to. 

httpd_handle_t web_server = NULL;
httpd_config_t config = HTTPD_DEFAULT_CONFIG();
config.server_port = 8080;
if (httpd_start(&web_server, &config) == ESP_OK) {...}

mahavirj, 2019-01-25:
@versamodule Could you please send across device log and also IDF commit/version (as per issue template)?

versamodule, 2019-01-25:
I wanted to but I am using VisualGDB and I do not have access to the terminal to issue commands to find this information out.  They say they are using 3.2. 

[image]

mahavirj, 2019-01-29:
@versamodule Unfortunately I am unable to reproduce this, if we walk through code, `server_port` is definitely being used,https://github.com/espressif/esp-idf/blob/18118a6d5ceebef59127ab3d2319f616c2924880/components/esp_http_server/src/httpd_main.c#L242

Can you check around this through debug logs at your end?

mahavirj, 2019-02-06:
@versamodule Did you get chance to look into this?

versamodule, 2019-02-06:
Hello, I'll create a short demo to show it works when set to port 80, but when I change the port it stops.

versamodule, 2019-02-06:
I am speechless. I took the time to prove this was not working before I posted the problem to avoid anyone wasting their time. I changed "config.server_port " from 80 to 8080 and back several times. Even doing a clean build. It was always repeatable. 
I created a new project today, and same code worked find on both ports. I just don't know what to say other than thank you for your time, and I am sorry I wasted it.

closed 2019-02-06 by versamodule