espressif/esp-idf #2989: Imposible to connect in AP mode to a hidden SSID in channel 13

Daverkex, 2019-01-26:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v3.3-beta1-268-g5c88c5996
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

Imposible to connect in AP mode to a hidden SSID in channel 13.
Googling I found a similar issue reported in ESP32 forum: https://www.esp32.com/viewtopic.php?t=4880
I can confirm the forum bug.

If the SSID is unhidden the ESP32 connect whithout problems.
if the SSID is move to other channel the ESP32 connect whithout problems.

### Expected Behavior
Connect correctly to SSID hidden in channel 13

### Actual Behavior
Error to connect with SYSTEM_EVENT_STA_DISCONNECTED and reason WIFI_REASON_NO_AP_FOUND

### Steps to repropduce

1. Create a AP with hidde SSID in channel 13
2. Try to connect the ESP32

### Code to reproduce this issue

https://gist.github.com/Daverkex/93719871cc53659a9de4d60db32b6f47

## Debug Logs

```
...
I (85135) wifi station: retry to connect to the AP
I (85135) wifi station: 201
I (87545) wifi station: retry to connect to the AP
I (87545) wifi station: 201
I (89955) wifi station: retry to connect to the AP
...
```

liuzfesp, 2019-02-02:
HI @Daverkex, it's expected behavior because it's passive scan from channel 12 to 13 by default.

If you want to find the hidden from channel 12 to 13, you can configure the country to:
wifi_country_t count = {
.cc = "CN",
.schan = 1,
.nchan = 13,
.policy = WIFI_COUNTRY_POLICY_MANUAL,
};
esp_wifi_set_country(&country);

Alvin1Zhang, 2019-02-14:
@Daverkex Hi, would you help share if any updates for this issue? Thanks.

Daverkex, 2019-02-14:
Hi @Alvin1Zhang and @liuzfesp 
Sorry for being late. I had problems in my personal life.

I tried the @liuzfesp code and it work fine. I forgot to change the **policy flag to manual mode**.

This solution works as expected.

Thank you.

closed 2019-02-14 by Daverkex