espressif/esp-idf #2991: HTTP_CLIENT: Connection failed, sock < 0

kosukekurimoto, 2019-01-27:
Hello.
I created a code to send an http request with reference to [this.](https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/esp_http_client.c)
But it will not be able to send with errors in a few hours. This happens every time.
Is there any way to solve it?

```
I (20121) HTTP_CLIENT: Start send signal

I (20121) HTTP_CLIENT: Start send signal: {"format":"us","freq":32,"data":[8895,4493,556,601,577,578,577,579,581,577,581,575,580,580,555,600,577,1655,580,1658,575,581,576,1660,578,578,559,596,578,582,575,1658,575,1658,576,580,577,1656,578,581,576,1660,576,1656,558,596,578,585,577,1653,576,1657,575,583,575,1656,577,579,585,570,577,1658,558,1673,578,581,575,52371,15711,665,17074,665,3137,665,5686,1924,611,648,619,1916,602,1931,622,645,605,663,618,1915,620,1915,623,643,620,1917,621,646,618,650,616,1917,620,1915,619,647,625]}

I (20411) HTTP_CLIENT: HTTP Status = 200, content_length = -1
I (20411) HTTP_CLIENT: Finish send signal
```
↓
a few hours later
↓
```
I (31241941) HTTP_CLIENT: Start send signal

I (31241941) HTTP_CLIENT: Start send signal: {"format":"us","freq":37,"data":[8911,4533,580,583,581,582,578,585,577,582,582,582,578,585,575,584,579,1666,577,1669,576,582,580,1665,580,583,578,581,581,583,577,1666,577,1668,576,1668,577,1664,579,583,576,1665,579,1663,580,584,577,587,574,1665,578,589,573,584,579,1660,582,582,577,586,575,1667,577,1663,580,584,576,52588,10075,5099,620,650,621,652,620,1925,621,652,623,649,620,655,618,652,624,1922,627,1917,26055,1943,604,669,4447,643,4439,1928,618,651,630]}

E (31241991) TRANS_TCP: Error create socket
E (31241991) HTTP_CLIENT: Connection failed, sock < 0
E (31241991) HTTP_CLIENT: Error perform http request ESP_ERR_HTTP_CONNECT
I (31242001) HTTP_CLIENT: Finish send signal
```

Code
```
#include "natureremo.h"
#include <stdio.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "wifi_setup.h"

#include "esp_http_client.h"

static const char* TAG = "HTTP_CLIENT";
static int usage_core_id = 0;

/* アプライアンスとシグナルIDの定義 */
static natureremo_ceiling_light_config_t ceiling_lights[] = {
    {
    .on = false,
    .name = "Living Light\0",
    .signal_id_on = "{\"format\":\"us\",\"freq\":32,\"data\":[8895,4493,556,601,577,578,577,579,581,577,581,575,580,580,555,600,577,1655,580,1658,575,581,576,1660,578,578,559,596,578,582,575,1658,575,1658,576,580,577,1656,578,581,576,1660,576,1656,558,596,578,585,577,1653,576,1657,575,583,575,1656,577,579,585,570,577,1658,558,1673,578,581,575,52371,15711,665,17074,665,3137,665,5686,1924,611,648,619,1916,602,1931,622,645,605,663,618,1915,620,1915,623,643,620,1917,621,646,618,650,616,1917,620,1915,619,647,625]}\0",
    .signal_id_off = "{\"format\":\"us\",\"freq\":30,\"data\":[8906,4492,573,585,580,577,575,583,575,580,578,583,576,582,554,601,577,1656,578,1660,575,585,574,1661,573,585,575,584,576,580,575,1664,571,1659,576,585,575,583,557,598,575,1665,574,579,559,602,573,585,571,584,577,1656,578,1661,574,1658,579,581,558,1675,578,1660,575,1661,577,1660,571,52411,20824,641,6944,1936,6966,641,620,1918,619,648,624,645,621,647,619,651,624,1912,618,650,620,651,616,653,620,645,621,1919,616,1918,618,1919,624,645,602,1934,619,1920,617,1918,619,1920,617]}\0",
    },{
    .on = false,
    .name = "Kitchen Light\0",
    .signal_id_on = "{\"format\":\"us\",\"freq\":37,\"data\":[8911,4533,580,583,581,582,578,585,577,582,582,582,578,585,575,584,579,1666,577,1669,576,582,580,1665,580,583,578,581,581,583,577,1666,577,1668,576,1668,577,1664,579,583,576,1665,579,1663,580,584,577,587,574,1665,578,589,573,584,579,1660,582,582,577,586,575,1667,577,1663,580,584,576,52588,10075,5099,620,650,621,652,620,1925,621,652,623,649,620,655,618,652,624,1922,627,1917,26055,1943,604,669,4447,643,4439,1928,618,651,630]}\0",
    .signal_id_off = "{\"format\":\"us\",\"freq\":36,\"data\":[8909,4536,577,583,578,586,579,584,576,585,576,586,578,585,574,587,574,1666,580,1669,577,584,575,1666,561,603,576,587,578,581,584,1660,578,1666,575,1666,579,584,586,577,549,1695,576,586,575,588,573,587,575,588,575,584,577,1664,579,1666,587,577,575,1668,585,1660,579,1664,575,1669,576,52613,10081,5108,618,651,621,653,620,1928,620,654,618,652,603,672,620,653,622,1927,625,1920,625,649,618,655,623,652,620,1927,627,648,620,1924,622,652,621,1928,620,653,624,647,626,1924,620,652,606,666,624,656,616,653,624,651,620,1927,621,1926,620,652,622,1929,600,1946,623,1924,620,1927,620]}\0",
    }
};

esp_err_t _http_event_handler(esp_http_client_event_t *evt) {
    switch(evt->event_id) {
        case HTTP_EVENT_ERROR:
            ESP_LOGD(TAG, "HTTP_EVENT_ERROR");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_CONNECTED");
            break;
        case HTTP_EVENT_HEADER_SENT:
            ESP_LOGD(TAG, "HTTP_EVENT_HEADER_SENT");
            break;
        case HTTP_EVENT_ON_HEADER:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
            break;
        case HTTP_EVENT_ON_DATA:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
            if (!esp_http_client_is_chunked_response(evt->client)) {
                // Write out data
                // printf("%.*s", evt->data_len, (char*)evt->data);
            }

            break;
        case HTTP_EVENT_ON_FINISH:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_FINISH");
            break;
        case HTTP_EVENT_DISCONNECTED:
            ESP_LOGD(TAG, "HTTP_EVENT_DISCONNECTED");
            break;
    }
    return ESP_OK;
}

static void send_signal(char* signal_id) {
    ESP_LOGI(TAG, "Start send signal: %s\n", signal_id);
    esp_http_client_config_t config = {
        .url = "http://192.168.10.20/messages",
        .event_handler = _http_event_handler,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);

    esp_http_client_set_method(client, HTTP_METHOD_POST);
    esp_http_client_set_header(client, "X-Requested-With", "curl");
    esp_http_client_set_header(client, "accept", "application/json");
    esp_http_client_set_post_field(client, signal_id, strlen(signal_id));
    esp_err_t err = esp_http_client_perform(client);

    if (err == ESP_OK) {
        ESP_LOGI(TAG, "HTTP Status = %d, content_length = %d",
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
        if (esp_http_client_get_status_code(client) == 400) {
            vTaskDelay(2000 / portTICK_PERIOD_MS);
            send_signal(signal_id);
        }
    } else {
        ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(client);
}

/* httpリクエストで使用するCPUコアを切り替える */
static void switch_core(){
    usage_core_id = usage_core_id == 0 ? 1 : 0;
}

static void send_signal_task(void* pvParameters){
    app_wifi_wait_connected();
    char* signal_id = (char*)pvParameters;
    ESP_LOGI(TAG, "Start send signal\n");
    send_signal(signal_id);
    ESP_LOGI(TAG, "Finish send signal\n");
    vTaskDelete(NULL);
}

void natureremo_ceiliing_light_on(int light_no) {
    switch_core();
    void* signal_id = ceiling_lights[light_no].signal_id_on;
    ceiling_lights[light_no].on = true;
    xTaskCreatePinnedToCore(&send_signal_task, "send_signal_task", 8192, signal_id, 5, NULL, usage_core_id);
}

void natureremo_ceiliing_light_off(int light_no) {
    switch_core();
    void* signal_id = ceiling_lights[light_no].signal_id_off;
    ceiling_lights[light_no].on = false;
    xTaskCreatePinnedToCore(&send_signal_task, "send_signal_task", 8192, signal_id, 5, NULL, usage_core_id);
}

bool natureremo_ceiliing_light_status(int light_no) {
    ESP_LOGI(TAG, "Get ceiling light status(No=%d): Status=%d\n", light_no, ceiling_lights[light_no].on);
    return ceiling_lights[light_no].on;
}

```

kosukekurimoto, 2019-02-04:
I published the entire source code including this problem.
https://github.com/kosukekurimoto/esp32-homekit-natureremo

keesus, 2019-03-14:
@kosukekurimoto 

I guess your issue is same as mine.

mine occurs when I make more than ten http requests regardless of method.

have you found any solutions yet?

liuzfesp, 2019-03-17:
@espxiehang please take a look at this issue.

espxiehang, 2019-03-18:
hi，@kosukekurimoto ，Please tell me how to reproduce this issue and what version are you using ?Thank you.

burkulesomesh43, 2019-07-09:
Hello everyone,
I am also getting this issue. I am using esp idf v3.2.

flyingraijin9, 2019-07-24:
hello,@kosukekurimoto , I tried your code and the issue seems to be that number of sockets used is greater  than the number of  maximum configured sockets.  The number of maximum configured sockets is 10 by default in IDF.You can check the same in your sdkconfig file. You can modify this value through "make menuconfig",To find the option go to Component config -> LWIP ->Max Number of Open Sockets. Set them to the more than the number of sockets you require.
@burkulesomesh43 you can try this as well and see if it solves your problem.

mahavirj, 2019-08-01:
Closing due to lack of response. Please do give a try with tuning config option `Component config -> LWIP ->Max Number of Open Sockets`

closed 2019-08-01 by mahavirj

stanislau-arkhipenka, 2024-03-06:
Having exactly same issue. My app doing get requests every couple seconds, and after few minutes of runtime - I get an error, that indicates that too many sockets in use

AdityaHPatwardhan, 2024-03-06:
Hi @stanislau-arkhipenka 
1) As indicated above, have you checked the value of ` Component config -> LWIP ->Max Number of open sockets` and the current open sockets that you have?
2) Can you ensure that the sockets that are created for the request are closed appropriately ?

stanislau-arkhipenka, 2024-03-06:
Hey @AdityaHPatwardhan, thanks for reply!
I checked the config - and I used default value 10. I tried to increase up to 16, but issue doesn't go away.
Here is a function I call once every 2 seconds (sorry, it's quite ugly):
```
cJSON* do_get(std::string link) {
    cJSON *def_response = cJSON_CreateObject();
    char output_buffer[MAX_HTTP_OUTPUT_BUFFER] = {0}; 
    esp_http_client_config_t config = {
        .url = link.c_str(),
        .method = HTTP_METHOD_GET,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    esp_err_t err = esp_http_client_open(client, 0); 

    if (err == ESP_OK) {
        int content_length = esp_http_client_fetch_headers(client); 
         if (content_length < 0) { 
             ESP_LOGE(TAG, "HTTP client fetch headers failed"); 
         } else { 
             int data_read = esp_http_client_read_response(client, output_buffer, MAX_HTTP_OUTPUT_BUFFER); 
             if (data_read >= 0) { 
                int content_length = esp_http_client_get_content_length(client);
                ESP_LOGI(TAG, "HTTP GET Status = %d, content_length = %d", 
                esp_http_client_get_status_code(client), 
                content_length); 
                cJSON *root = cJSON_Parse(output_buffer);
                if (root != NULL) {
                    def_response = root;
                    ESP_LOGD(TAG, "Successfully parsed json");
                } else {
                    ESP_LOGW(TAG, "Unable to parse json");
                } 
             } else { 
                 ESP_LOGW(TAG, "Failed to read response");
             }
         }         
    } else {
        ESP_LOGI(TAG, "HTTP GET request failed: %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(client); 
    return def_response;
}
```
TLDR:
esp_http_client_init()
....
esp_http_client_cleanup()

stanislau-arkhipenka, 2024-03-06:
Allrighty, it seems problem is unrelated to http client. 
In my particular case: I wasn't calling cJSON_Delete() on my root object, resulting in **memory leak**. Over time I ran out of memory, and because of that - was unable to open new socket. 
From log standpoint it looked like I run out of sockets, but actually I run out of memory to open sockets