espressif/esp-idf #2993: mqtt: TLS: test.mosquitto.org issue

vikrant8051, 2019-01-28:
Broker: test.mosquitto.org
CA file: http://test.mosquitto.org/ssl/mosquitto.org.crt

For more details: https://mosquitto.org/blog/2012/06/ssl-support-on-test-server/

I tried to edit esp-idf/examples/protocols/mqtt/ssl but getting following error on terminal:

```
E (4536) esp-tls: mbedtls_ssl_handshake returned -0x2700                                                                          
I (4536) esp-tls: Failed to verify peer certificate!                                                                              
I (4536) esp-tls: verification info:   ! The certificate is signed with an unacceptable key (eg bad curve, RSA too short).
```

I have  copy mosquitto.org.crt in esp-idf/examples/protocols/mqtt/ssl/main & renamed it as iot_eclipse_org.pem.

Is it correct way to add CA certs details ?

david-cermak, 2019-01-28:
Hi @vikrant8051 

The error description actually tells us what the problem is: `The certificate is signed with an unacceptable key (eg bad curve, RSA too short).`

If you look at the certificate `openssl x509 -in  Downloads/mosquitto.org.crt -text`, you may find that

-  Signature Algorithm: sha1WithRSAEncryption
-  RSA key len is 1024

This is considered unsafe and the certificate is not accepted by default mbed-tls configuration.

It is however possible to override these settings directly in mbedtls 
(either by enabling `MBEDTLS_TLS_DEFAULT_ALLOW_SHA1_IN_CERTIFICATES` and lowering `rsa_min_bitlen` or calling api for user cert profile `mbedtls_ssl_conf_cert_profile`, but neither option is exposed from idf/mqtt api)

The best option is probably to choose a broker with uptodate security certificates. Please note the warning from mbedtls:  SHA-1 is considered a weak message digest and its use constitutes
 a security risk. If possible, we recommend avoiding dependencies on it, and considering stronger message digests instead.

PS: yes, it is a correct way to add a certificate to just include/embed as text file (another option is to place it directly into source as string/char* variable)

vikrant8051, 2019-02-07:
@david-cermak Thanks for clarification.

david-cermak, 2019-03-14:
Hi @vikrant8051 

Thank you for letting us know!

Would you please help closing this issue as that is not an issue on esp-idf side?

closed 2019-03-14 by Alvin1Zhang