espressif/esp-idf #2994: Wifi AP not accepting iPhones' connections

eftimarco, 2019-01-28:
Hello,

With 3.1.2 esp-idf version, I'm struggling to connect iPhones to Wifi AP while it is working fine with Androids' phones.

Wifi is configured in APSTA (see configuration below), and I answering DNS requests (using captdns).
Iphones never get an IP address... esp side, system_event_id goes to SYSTEM_EVENT_AP_STACONNECTED but no dhcps log (even with DHCPS_DEBUG = 1 in dhcpserver.c).
Android phones never get the problem.

Does anyone face the same issue ? Any tricks ?

```
wifi_config_t wifi_config = {
	.ap = {
		.ssid = WIFI_AP_SSID,
		.ssid_len = strlen(WIFI_AP_SSID),
		.password = WIFI_AP_PWD,
		.authmode = WIFI_AUTH_WPA2_PSK,
		.max_connection = 4,
		.beacon_interval = 100,
		.channel = 5
	}
};
if (strlen(WIFI_AP_PWD) == 0) {
	wifi_config.ap.authmode = WIFI_AUTH_OPEN;
}
esp_wifi_set_storage(WIFI_STORAGE_RAM);
esp_wifi_set_mode(WIFI_MODE_APSTA);
```

TimXia, 2019-01-28:
@eftimarco Which iphone did you use? Can you tell us the IOS version?

eftimarco, 2019-01-28:
We tried on 3 different phones :
  iPhone 5 - OS 12.1
+ iPhone 6 - OS 12.1
+ iPhone 7 - OS 12.1

TimXia, 2019-01-29:
@eftimarco Can you try the latest IDF 3.1? It has already fixed the issue.

TimXia, 2019-01-31:
@eftimarco We have found the root cause and will fix it soon.

eftimarco, 2019-01-31:
@TimXia One thing seems to work : if any connection (working, from an Android, or not, from an iPhone) is ongoing, a second one (from another iPhone) works...
I hope that can help !

TimXia, 2019-02-01:
@eftimarco Good news. We have found the root cause. And it will be fixed in IDF before Chinese New Year.

StefanStsc, 2019-02-07:
@TimXia when will you provide a fix for this? I guess CNY already started.

TimXia, 2019-02-14:
@StefanStsc We have provide the fix. Please update the IDF.

TimXia, 2019-02-14:
@StefanStsc The fix is on the way and will be provided soon.

TimXia, 2019-02-15:
@StefanStsc The fix is provided. Please update the IDF branch.

liuzfesp, 2019-02-15:
@StefanStsc this fix is already merged into IDF, but I think it still need 2 days before it can be automatically sync to github.

eftimarco, 2019-02-18:
@TimXia It does not seem to work... I upgrade from v3.1.2-95-gcfe10825c to v3.1.2-99-gcf5dbadf4
Is that the version in which the fix has been merged ?

TimXia, 2019-02-27:
@eftimarco Please upgrade IDF again. It has been merged.

eftimarco, 2019-02-27:
@TimXia Yes, I was testing yesterday evening (europe time) v3.1.2-114-g57118e2f1 : it is working with the first 2 iphones ! I'll make more tests today, I tell you if something goes wrong but it looks like ok !

BPR800, 2019-02-28:
I tried idf 3.1.2 it when you connect with Android there is a continuous dynamic ip address (tried two devices). Then I updated part of the files with 0c5f3d45865423caa1f3069ecff905420641220b
now Android devices are working, but when you connect ios, the watchdog is triggered and the processor freezes. Also tried idf 3.1.3 Android device not getting ip address. Examples of Idf in its pure form is also not working

TimXia, 2019-03-01:
@BRP800 Did you update the submodule? Can you provide the whole log?

BPR800, 2019-03-01:
Need update WiFi submodule? I am update whole idf

TimXia, 2019-03-01:
@BPR800 What example did you use?

BPR800, 2019-03-01:
now I'm trying to work with my program. Everything works with Android device. When you use an ios device, the watchdog triggers when you open a web page.

TimXia, 2019-03-01:
@BPR800 Can you provide the whole log and elf file?

BPR800, 2019-03-01:
`Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4013b9b6  PS      : 0x00060834  A0      : 0x8008bc68  A1      : 0x3ffbe790
A2      : 0x00000020  A3      : 0x00000000  A4      : 0x3ffb4548  A5      : 0x3ffb2464
A6      : 0x3ffb4870  A7      : 0x3ffb4618  A8      : 0x3ffb4564  A9      : 0x3ffbe770
A10     : 0x00000000  A11     : 0x7fffffff  A12     : 0x8008c900  A13     : 0x3ffbe690
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4013b9b6:0x3ffbe790 0x4008bc65:0x3ffbe7b0

Core 1 register dump:
PC      : 0x4000c271  PS      : 0x00060034  A0      : 0x8008c05a  A1      : 0x3ffc2b00
A2      : 0x3ffc0bf0  A3      : 0x3f4033ee  A4      : 0x00000014  A5      : 0x3ffb1fc8
A6      : 0x3ffb2010  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x3f4033ef
A10     : 0x000000a5  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x00000001
A14     : 0x0000abab  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x0000001c

Backtrace: 0x4000c271:0x3ffc2b00 0x4008c057:0x3ffc2b20 0x4008d338:0x3ffc2b50 0x4008d2ee:0x3ffb9c24

CPU halted.
`

TimXia, 2019-03-01:
@BPR800 The whole log please, and the elf file.

BPR800, 2019-03-01:
Sorry, I don't know how to show you .elf file ... it's big

TimXia, 2019-03-01:
@BPR800 You can compress it and attach it here.

BPR800, 2019-03-01:
here

BPR800, 2019-03-01:
https://dropmefiles.com/UND6Z

BPR800, 2019-03-01:
I have now compiled the project with IDF 3.1.3 
Android devices can't get ip address
IOS gets ip. But again it triggers the watchdog when you try to open the file.
`I (3273) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (3273) wifi: station: dc:2b:61:af:23:2a join, AID=1, bgn, 20
I (3381) tcpip_adapter: softAP assign IP to station,IP is: 192.168.5.2
I (6343) wifi: station: dc:2b:61:af:23:2a leave, AID = 1
I (6343) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (20585) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (20585) wifi: station: 5c:0a:5b:95:93:5b join, AID=1, bgn, 20
I (51349) wifi: station: 5c:0a:5b:95:93:5b leave, AID = 1
I (51349) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (54225) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (54225) wifi: station: 5c:0a:5b:95:93:5b join, AID=1, bgn, 20
I (84995) wifi: station: 5c:0a:5b:95:93:5b leave, AID = 1
I (84997) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1

`

jack0c, 2022-01-05:
This bug should be fixed after version v3.3

closed 2022-01-05 by jack0c