espressif/esp-idf #2995: fopen + fclose result in memory leak

adherent88, 2019-01-28:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 3.1.2
- Build System:         [Make]
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
fopen directly followed by a fclose will result in 160 bytes heap leak. 

### Expected Behavior
heap size should recover after fclose is called

### Actual Behavior
160 bytes in heap is gone forever

### Steps to repropduce

### Code to reproduce this issue

```cpp
    printf("--------> before, free heap size = %d\n" , xPortGetFreeHeapSize());
        
    ESP_ERROR_CHECK( heap_trace_init_standalone(trace_record, NUM_RECORDS) );
    ESP_ERROR_CHECK( heap_trace_start(HEAP_TRACE_LEAKS) );

    FILE* adc = fopen("/sdcard/project/common/0.wav", "r");
    fclose(adc);

    ESP_ERROR_CHECK( heap_trace_stop() );
    heap_trace_dump();

    printf("--------> after, free heap size = %d\n" , xPortGetFreeHeapSize());

```

## Debug Logs
--------> after, free heap size = 80362
2 allocations trace (100 entry buffer)
80 bytes (@ 0x3ffebbfc) allocated CPU 1 ccount 0xe35bea18 caller 0x4008f216:0x40
08f447:0x4008380d:0x40083833:0x40083978:0x4000be38:
0x4008f216: xQueueGenericCreate at C:/msys32/home/db/esp/esp-idf/components/free
rtos/queue.c:2519

0x4008f447: xQueueCreateMutex at C:/msys32/home/db/esp/esp-idf/components/freert
os/queue.c:2519

0x4008380d: lock_init_generic at C:/msys32/home/db/esp/esp-idf/components/newlib
/locks.c:80

0x40083833: lock_acquire_generic at C:/msys32/home/db/esp/esp-idf/components/new
lib/locks.c:133

0x40083978: _lock_acquire_recursive at C:/msys32/home/db/esp/esp-idf/components/
newlib/locks.c:169

80 bytes (@ 0x3ffebca4) allocated CPU 0 ccount 0xe4fc38b0 caller 0x4008f216:0x40
08f447:0x4008380d:0x40083833:0x40083968:0x400e2c3c:0x4000bd6a:
0x4008f216: xQueueGenericCreate at C:/msys32/home/db/esp/esp-idf/components/free
rtos/queue.c:2519

0x4008f447: xQueueCreateMutex at C:/msys32/home/db/esp/esp-idf/components/freert
os/queue.c:2519

0x4008380d: lock_init_generic at C:/msys32/home/db/esp/esp-idf/components/newlib
/locks.c:80

0x40083833: lock_acquire_generic at C:/msys32/home/db/esp/esp-idf/components/new
lib/locks.c:133

0x40083968: _lock_acquire at C:/msys32/home/db/esp/esp-idf/components/newlib/loc
ks.c:165

0x400e2c3c: esp_vfs_open at C:/msys32/home/db/esp/esp-idf/components/vfs/vfs.c:9
70 (discriminator 1)

160 bytes 'leaked' in trace (2 allocations)
total allocations 10 total frees 8
--------> after, free heap size = 80192

igrr, 2019-01-28:
Are you seeing twice the amount of memory leaked after repeating fopen+fclose twice? At the moment we rely on lazy allocation of locks (mutexes) in a few places, and 160 bytes seems to be about the size of one mutex. Any further VFS operation should use the same lock, so we don't consider this as a real leak.

adherent88, 2019-01-28:
> Are you seeing twice the amount of memory leaked after repeating fopen+fclose twice? At the moment we rely on lazy allocation of locks (mutexes) in a few places, and 160 bytes seems to be about the [...]

No, if I keep doing fopen and fclose, only 160 bytes heap is forever gone.

But if I first open several files(leave them open), then close them afterwards, e.g.
```
        FILE* aaa = fopen("/sdcard/puppet/ag/tutorial/common/0.wav", "r");
        FILE* bbb = fopen("/sdcard/puppet/ag/tutorial/common/1.wav", "r");
        FILE* ccc = fopen("/sdcard/puppet/ag/tutorial/common/2.wav", "r");
        FILE* ddd = fopen("/sdcard/puppet/ag/tutorial/common/3.wav", "r");
        FILE* eee = fopen("/sdcard/puppet/ag/tutorial/common/4.wav", "r");
        FILE* fff = fopen("/sdcard/puppet/ag/tutorial/common/5.wav", "r");

        fclose(aaa);
        fclose(bbb);
        fclose(ccc);
        fclose(ddd);
        fclose(eee);
        fclose(fff);
```

1026 bytes are gone forever, please see the heap trace log:
```
--------> before, free heap size = 80220
4 allocations trace (100 entry buffer)
80 bytes (@ 0x3ffebc98) allocated CPU 1 ccount 0x52f069b0 caller 0x4008f216:0x40
08f447:0x4008380d:0x40083833:0x40083978:0x4000be38:
0x4008f216: xQueueGenericCreate at C:/msys32/home/db/esp/esp-idf/components/free
rtos/queue.c:2519

0x4008f447: xQueueCreateMutex at C:/msys32/home/db/esp/esp-idf/components/freert
os/queue.c:2519

0x4008380d: lock_init_generic at C:/msys32/home/db/esp/esp-idf/components/newlib
/locks.c:80

0x40083833: lock_acquire_generic at C:/msys32/home/db/esp/esp-idf/components/new
lib/locks.c:133

0x40083978: _lock_acquire_recursive at C:/msys32/home/db/esp/esp-idf/components/
newlib/locks.c:169

80 bytes (@ 0x3ffebd40) allocated CPU 0 ccount 0x5490ff60 caller 0x4008f216:0x40
08f447:0x4008380d:0x40083833:0x40083968:0x400e2c50:0x4000bd6a:
0x4008f216: xQueueGenericCreate at C:/msys32/home/db/esp/esp-idf/components/free
rtos/queue.c:2519

0x4008f447: xQueueCreateMutex at C:/msys32/home/db/esp/esp-idf/components/freert
os/queue.c:2519

0x4008380d: lock_init_generic at C:/msys32/home/db/esp/esp-idf/components/newlib
/locks.c:80

0x40083833: lock_acquire_generic at C:/msys32/home/db/esp/esp-idf/components/new
lib/locks.c:133

0x40083968: _lock_acquire at C:/msys32/home/db/esp/esp-idf/components/newlib/loc
ks.c:165

0x400e2c50: esp_vfs_open at C:/msys32/home/db/esp/esp-idf/components/vfs/vfs.c:9
70 (discriminator 1)

428 bytes (@ 0x3ffebd94) allocated CPU 0 ccount 0x54915304 caller 0x400839bc:0x4
000bbc5:
0x400839bc: _malloc_r at C:/msys32/home/db/esp/esp-idf/components/newlib/syscall
s.c:38

428 bytes (@ 0x3ffecab0) allocated CPU 0 ccount 0x54b2f4dc caller 0x400839bc:0x4
000bbc5:
0x400839bc: _malloc_r at C:/msys32/home/db/esp/esp-idf/components/newlib/syscall
s.c:38

1016 bytes 'leaked' in trace (4 allocations)
total allocations 16 total frees 12
--------> after, free heap size = 79184
```

adherent88, 2019-01-28:
the first two 80 bytes could be due to the mutex, that is ok. But what about the last two 428 bytes lost? 

Thanks for the answer.

igrr, 2019-01-29:
Could you please increase [`CONFIG_HEAP_TRACING_STACK_DEPTH`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-heap-tracing-stack-depth) to get a slightly better backtrace for that 428 byte allocation? I'm asking because I am not able to reproduce the same effect here (probably due to not having exactly the same files layout).

adherent88, 2019-01-29:
> Could you please increase [`CONFIG_HEAP_TRACING_STACK_DEPTH`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-heap-tracing-stack-depth) to get a slightly [...]

I have set the depth = 10 since beginning, that's why for the 80 bytes lost case we have the rather complete call stack.

But for the 428 bytes lost case, the backtrace has only a single entry: syscalls.c which is doing malloc. I also want to know where does this call originate....

closed 2021-07-20 by igrr