espressif/esp-idf #2997: Possible coredump in hostap_input with _WindowOverflow4 exception

mill1000, 2019-01-28:
## Environment
- Development Kit: none
- Module or chip used: ESP32-WROOM-32
- IDF version: v3.1 22da5f6 with 3ba63a5 cherry-picked
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System: Windows
- Power Supply: Battery

## Problem Description
European customers have been reporting WiFi disconnects with our product integrating an ESP32. Our team has been unable to replicate this behavior in the USA. 

Multiple coredumps taken have the following backtrace.
```
==================== CURRENT THREAD STACK =====================
#0  0x40000000 in _WindowOverflow4 ()
#1  0x400f3859 in hostap_input ()
#2  0x400f4b0e in ap_rx_cb ()
#3  0x4011145c in ppRxPkt ()
#4  0x400878a5 in ppTask ()
#5  0x4008c5bc in vPortTaskWrapper (pxCode=0x400877b0 <ppTask>, pvParameters=0x0) at esp-idf/components/freertos/port.c:141
```

Other coredumps have caused GDB to crash when loading the ROM elf, possibly indicating a stack corruption problem.

Oddly, we have also seen the issue on esp-idf 28cb562. Another series of test to determine if the problem occurs on esp-idf v3.1.2 will occur in the next day or two.

### Expected Behavior
The device continues to operate.

### Actual Behavior
The device coredumps and reboots.

## Debug Logs

info_coredump on one instance. Full coredump, ELF and sdkconfig are attached.
```
espcoredump.py v0.2-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

================== CURRENT THREAD REGISTERS ===================
pc             0x40000000	0x40000000 <_WindowOverflow4>
lbeg           0x4000c349	1073791817
lend           0x4000c36b	1073791851
lcount         0xffffffff	4294967295
sar            0x10	16
ps             0x60020	393248
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x400f3859	1074739289
a1             0x3ffd5170	1073566064
a2             0x0	0
a3             0x37	55
a4             0xffffffa0	-96
a5             0x3ffd52f8	1073566456
a6             0xc	12
a7             0x0	0
a8             0x80104ca8	-2146415448
a9             0x3ffd5160	1073566048
a10            0x3ffde8b0	1073604784
a11            0x37	55
a12            0xffffffa0	-96
a13            0x3ffd5204	1073566212
a14            0xd173	53619
a15            0xd173	53619

==================== CURRENT THREAD STACK =====================
#0  0x40000000 in _WindowOverflow4 ()
#1  0x400f3859 in hostap_input ()
#2  0x400f4b0e in ap_rx_cb ()
#3  0x4011145c in ppRxPkt ()
#4  0x400878a5 in ppTask ()
#5  0x4008c5bc in vPortTaskWrapper (pxCode=0x400877b0 <ppTask>, pvParameters=0x0) at esp-idf/components/freertos/port.c:141

======================== THREADS INFO =========================
  Id   Target Id         Frame 
  14   process 13        0x40082052 in esp_crosscore_int_send_yield (core_id=0) at esp-idf/components/esp32/crosscore_int.c:112
  13   process 12        xEventGroupWaitBits (xEventGroup=0x3ffd7664, uxBitsToWaitFor=16777215, xClearOnExit=1, xWaitForAllBits=0, xTicksToWait=<optimized out>) at esp-idf/components/freertos/event_groups.c:445
  12   process 11        0x4008ce3c in xQueueGenericReceive (xQueue=0x3ffaea30 <r_ip_funcs+800>, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at esp-idf/components/freertos/queue.c:1591
  11   process 10        0x4008ce3c in xQueueGenericReceive (xQueue=0x3ffcb640, pvBuffer=0x3ffc5d60, xTicksToWait=4294967295, xJustPeeking=0) at esp-idf/components/freertos/queue.c:1591
  10   process 9         0x4008ce3c in xQueueGenericReceive (xQueue=0x3ffc0a28, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at esp-idf/components/freertos/queue.c:1591
  9    process 8         0x4008ce3c in xQueueGenericReceive (xQueue=0x3ffc0ff8, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at esp-idf/components/freertos/queue.c:1591
  8    process 7         0x4008bb7c in vTaskDelay (xTicksToDelay=<optimized out>) at esp-idf/components/freertos/tasks.c:1491
  7    process 6         0x4008ce3c in xQueueGenericReceive (xQueue=0x3ffdc714, pvBuffer=0x0, xTicksToWait=1000, xJustPeeking=0) at esp-idf/components/freertos/queue.c:1591
  6    process 5         xEventGroupWaitBits (xEventGroup=0x3ffd8310, uxBitsToWaitFor=16777215, xClearOnExit=1, xWaitForAllBits=0, xTicksToWait=<optimized out>) at esp-idf/components/freertos/event_groups.c:445
  5    process 4         0x4008ce3c in xQueueGenericReceive (xQueue=0x3ffca994, pvBuffer=0x3ffcb410, xTicksToWait=99, xJustPeeking=0) at esp-idf/components/freertos/queue.c:1591
  4    process 3         0x4008d5ad in prvProcessTimerOrBlockTask (xListWasEmpty=0, xNextExpireTime=<optimized out>) at esp-idf/components/freertos/timers.c:588
  3    process 2         0x4014f4b2 in esp_vApplicationWaitiHook () at esp-idf/components/esp32/freertos_hooks.c:66
  2    process 1         0x4014f4b2 in esp_vApplicationWaitiHook () at esp-idf/components/esp32/freertos_hooks.c:66
* 1    <main task>       0x40000000 in _WindowOverflow4 ()

======================= ALL MEMORY REGIONS ========================
Name   Address   Size   Attrs
.rtc.text 0x400c0000 0x0 RW  
.rtc_noinit 0x50000000 0x0 RW  
.iram0.vectors 0x40080000 0x400 R XA
.iram0.text 0x40080400 0x10f64 R XA
.dram0.data 0x3ffb0000 0x3870 RW A
.noinit 0x3ffb3870 0x0 RW  
.flash.rodata 0x3f400020 0x339a4 RW A
.flash.text 0x400d0018 0x80f20 R XA
.coredump.tasks.data 0x3ffd53a8 0x16c RW 
.coredump.tasks.data 0x3ffd50b0 0x2f0 RW 
.coredump.tasks.data 0x3ffc3754 0x16c RW 
.coredump.tasks.data 0x3ffc35b0 0x19c RW 
.coredump.tasks.data 0x3ffc31e0 0x16c RW 
.coredump.tasks.data 0x3ffc3040 0x198 RW 
.coredump.tasks.data 0x3ffc45c4 0x16c RW 
.coredump.tasks.data 0x3ffc4440 0x17c RW 
.coredump.tasks.data 0x3ffcb4d0 0x16c RW 
.coredump.tasks.data 0x3ffcb2b0 0x218 RW 
.coredump.tasks.data 0x3ffc83e8 0x16c RW 
.coredump.tasks.data 0x3ffc80e0 0x300 RW 
.coredump.tasks.data 0x3ffca55c 0x16c RW 
.coredump.tasks.data 0x3ffca200 0x354 RW 
.coredump.tasks.data 0x3ffc2be8 0x16c RW 
.coredump.tasks.data 0x3ffc29b0 0x230 RW 
.coredump.tasks.data 0x3ffc1458 0x16c RW 
.coredump.tasks.data 0x3ffc12b0 0x1a0 RW 
.coredump.tasks.data 0x3ffc0e88 0x16c RW 
.coredump.tasks.data 0x3ffc0ce0 0x1a0 RW 
.coredump.tasks.data 0x3ffc16d8 0x16c RW 
.coredump.tasks.data 0x3ffc5c60 0x3f0 RW 
.coredump.tasks.data 0x3ffafa90 0x16c RW 
.coredump.tasks.data 0x3ffaf8e0 0x1a8 RW 
.coredump.tasks.data 0x3ffc1848 0x16c RW 
.coredump.tasks.data 0x3ffc7620 0x1a4 RW 
.coredump.tasks.data 0x3ffd3d18 0x16c RW 
.coredump.tasks.data 0x3ffd3b20 0x1f0 RW 
.coredump.rom.text 0x40000000 0x170 R X
.coredump.rom.text 0x40000180 0x8 R X
.coredump.rom.text 0x400001c0 0x8 R X
.coredump.rom.text 0x40000200 0x8 R X
.coredump.rom.text 0x40000240 0x8 R X
.coredump.rom.text 0x40000280 0xc R X
.coredump.rom.text 0x400002c0 0x4 R X
.coredump.rom.text 0x40000300 0x8 R X
.coredump.rom.text 0x40000340 0x18 R X
.coredump.rom.text 0x400003c0 0x8 R X
.coredump.rom.text 0x40000400 0x160 R X
.coredump.rom.text 0x40000560 0xd10c R X
.coredump.rom.text 0x40010000 0x54ef4 R X

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

## Additional files

[ESP32.zip](https://github.com/espressif/esp-idf/files/2804743/ESP32.zip)

liuzfesp, 2019-02-02:
@mill1000 could you help to provide the crash log (not the coredump)?

mill1000, 2019-02-03:
@liuzfesp, do you mean the monitor logs? Unfortunately I can't at this time. We have been unable to replicate the crash locally and the devices don't have any persistent storage available for the logs.

It's tentative but the problem seems to have been resolved in v3.1.2

mill1000, 2019-02-07:
As a follow up note, the sample size is small but we have not received any reports of WiFi disconnects when built against esp-idf v3.1.2. Perhaps this was an manifestation of the softAP crash that was mentioned in the release notes.

liuzfesp, 2019-02-15:
@mill1000 I think this issue is fixed by commit 9edde5e8af3f2329fb7e0012fa806d9f54696ee7 which is contained in v3.1.2. 

 0x4000_0000 is an exception vector, and the PS register does not indicate that the CPU is in an exception mode: 0x60020. Furthermore, when IDF starts up, exception vectors are relocated to IRAM, so CPU will never jump to ROM vectors at 0x4000_0000.
This may be either an issue where we read invalid literal value from flash (flash read error) and then use this value as jump target, or this is the case that a function pointer is used, and it's value was corrupted.

mill1000, 2019-02-15:
@[liuzfesp](/liuzfesp), I agree. We've been having more and more successful reports of devices running using esp-idf v3.1.2

Alvin1Zhang, 2019-03-04:
@mill1000 Glad to hear that you have more and more successful reports, will close this issue for now and feel free to reopen, thanks.

closed 2019-03-04 by Alvin1Zhang