espressif/esp-idf #2999: AP+STA+WPS co-existence

huming2207, 2019-01-29:
## Environment

- Development Kit: TTGO Zero
- Kit version: Unkown
- Module or chip used:  ESP32-WROVER
- IDF version: v3.3-beta1-266-g2eabed161
- Build System: CMake
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB

## Problem Description

Hi there,

I've found an issue that the WPS doesn't work in AP+STA mode. I have followed these two examples:

- [https://esp32.com/viewtopic.php?t=538](https://esp32.com/viewtopic.php?t=538)
- [https://github.com/tidyjiang8/esp32-projects/blob/master/ap_sta/main/ap_sta.c](https://github.com/tidyjiang8/esp32-projects/blob/master/ap_sta/main/ap_sta.c)

I combined those two examples together by adding in a empty WiFi STA config and set it with `ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifiStaConfig));`. But there is no AP showing up on my phones and laptops. I've tried removing those STA config lines but it didn't help. So I would like to know that is it technically not possible for AP+STA+WPS co-existence? Or have I done something wrong in my code?

Regards,
Jackson

### Expected Behavior

AP, STA and WPS works just fine

### Actual Behavior

AP didn't shows up, not sure for WPS yet as I don't have a working WiFi router with WPS function for now.

### Steps to repropduce

1. follow those two examples
2. combines those two examples 

### Code to reproduce this issue

The full code is not available as it's confidential.

```cpp
            // Fill in WPS config here
            esp_wps_config_t wpsConfig = {};
            memset(&wpsConfig, 0, sizeof(wpsConfig));
            wpsConfig.crypto_funcs = &g_wifi_default_wps_crypto_funcs;
            wpsConfig.wps_type = WPS_TYPE_PBC;

            wifi_config_t wifiApConfig = {};
            wifi_config_t wifiStaConfig = {};
            memset(&wifiApConfig, 0, sizeof(wifiApConfig));
            memset(&wifiStaConfig, 0, sizeof(wifiStaConfig));

            // Fill AP config
            std::strcpy((char*)wifiApConfig.ap.ssid, ssid);

            wifiApConfig.ap.authmode = authMode;
            wifiApConfig.ap.ssid_len = std::strlen(ssid);
            wifiApConfig.ap.channel = channel;
            wifiApConfig.ap.max_connection = maxConn;

            if(passwd == nullptr || std::strlen(passwd) < 1) {
                wifiApConfig.ap.authmode = WIFI_AUTH_OPEN;
            } else {
                std::strcpy((char*)wifiApConfig.ap.password, passwd);
                wifiApConfig.ap.authmode = authMode;
            }

            ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
[... 211 lines cut ...]
D (3383) wpa: WPS: attr type=0x104a len=1
D (3383) wpa: WPS: attr type=0x1044 len=1
D (3383) wpa: WPS: attr type=0x104a len=1
D (3383) wpa: WPS: attr type=0x1044 len=1
D (3863) wpa: WPS: attr type=0x104a len=1
D (3863) wpa: WPS: attr type=0x1044 len=1
D (3863) wpa: WPS: attr type=0x1049 len=6
D (3863) wpa: WPS: WFA subelement id=0 len=1
D (3863) wpa: WPS: attr type=0x104a len=1
D (3873) wpa: WPS: attr type=0x1044 len=1
D (3873) wpa: WPS: attr type=0x1049 len=6
D (3873) wpa: WPS: WFA subelement id=0 len=1
D (3883) wpa: WPS: attr type=0x104a len=1
D (3883) wpa: WPS: attr type=0x1044 len=1
D (3893) wpa: WPS: attr type=0x1047 len=16
D (3893) wpa: WPS: attr type=0x103c len=1
D (3893) wpa: WPS: attr type=0x1049 len=6
D (3903) wpa: WPS: WFA subelement id=0 len=1
D (3903) wpa: WPS: attr type=0x104a len=1
D (3913) wpa: WPS: attr type=0x1044 len=1
D (3913) wpa: WPS: attr type=0x1047 len=16
D (3923) wpa: WPS: attr type=0x103c len=1
D (3923) wpa: WPS: attr type=0x1049 len=6
D (3923) wpa: WPS: WFA subelement id=0 len=1
```

TimXia, 2019-02-01:
@huming2207 Please test it with the latest IDF. We have already fixed the WPS issue.

huming2207, 2019-02-01:
@TimXia  Thanks. I will arrange a test later on.

Alvin1Zhang, 2019-05-16:
@huming2207 Thanks for reporting the issue. Would you please help share if any updates on your testing? Thanks.

huming2207, 2019-05-16:
Hello @Alvin1Zhang ,

Sorry for the delay. I was very busy at the moment.

There is no issue with my router, which is Linksys WRT32X with Marvell 88W8964 wireless chipset, plus OpenWrt 18.06.1 with backported mainline Marvell wireless drivers. 

I will conduct another detailed test next week if possible.

Regards,
Jackson

Alvin1Zhang, 2019-05-16:
@huming2207 Thanks

Alvin1Zhang, 2019-10-11:
@huming2207 Thanks for reporting, would you please help share if any updates for the issue? Any testing feedback? Thanks.

huming2207, 2019-10-11:
Hello @Alvin1Zhang ,

Sorry forgot to tell you. It works fine with Netgear R6300v2. I will close this issue now. Please re-open it when necessary.

Regards,
Jackson

closed 2019-10-11 by huming2207