espressif/esp-idf #3000: Option to require an IDF version in project config

MightyPork, 2019-01-29:
I'm looking for a way to make a project demand a particular IDF version and refuse to compile when the version differs.

apologies if this is written somewhere in the documentation.

This is similar to how you specify versions e.g. in `package.json` or `composer.json`, `cargo.toml` etc.

When developing on different computers, or having a separate build server, this would be very useful to keep the SDK in sync. I can't use the stable version because of the very slow release cycle, so I'm stuck with master and checking the version manually.

Ideally this would be an entry in sdkconfig or similar

mill1000, 2019-01-30:
You can accomplish this with some additions to your project Makefile.

The following makefile will throw an error if the esp-idf commit does not match CONFIG_REQUIRED_IDF
```
#
# This is a project Makefile. It is assumed the directory this Makefile resides in is a
# project subdirectory.
#
PROJECT_NAME := ESP32
include $(IDF_PATH)/make/project.mk

IDF_COMMIT := "$(shell git -C $$IDF_PATH rev-parse HEAD)"

.PHONY: check-idf
check-idf:
ifneq ($(IDF_COMMIT),$(CONFIG_REQUIRED_IDF))
	$(error esp-idf commit $(IDF_COMMIT) is not required $(CONFIG_REQUIRED_IDF))
endif

# Add check-idf as a prereq for all builds
all_binaries: check-idf
```

CONFIG_REQUIRED_IDF is set via `make menuconfig` by adding a `Kconfig.projbuild` to your `main/` directory.

```
menu "Project Configuration"

config REQUIRED_IDF
    string "Required esp-idf commit"
    default ""
    help
        esp-idf commit required for successful build. Makefile will abort if commits mismatch.
		
endmenu
```

The only thing I wish I knew how to fix was to throw the error before the components are built. Currently it will waste time building all the components, and then error.

MightyPork, 2019-01-30:
Thanks, though I'm using cmake - idf.py (and consider the makefile option close to deprecated). Maybe a similar hack can be ported to cmake :thinking:

mill1000, 2019-01-30:
Ah well I am not familiar with cmake but I imagine the general concept still applies.

Based on the docs I've got until esp-idf v5.0 before I have to worry about my Makefile

renzbagaporo, 2019-02-01:
If checking from the main component CMakeLists.txt is acceptable (not quite the project level as you originally asked), then you can actually just create the `Kconfig.projbuild` as @mill1000 suggested and do a comparison to `IDF_VER`.

```
if (NOT CONFIG_REQUIRED_IDF STREQUAL IDF_VER)
    message(FATAL_ERROR "Invalid ESP-IDF version")
endif()
```

MightyPork, 2019-02-01:
@renzbagaporo thanks, this works

main/KConfig.projbuild
```kconfig
menu "Project Configuration"

config REQUIRED_IDF
    string "Required esp-idf commit"
    default ""
    help
        esp-idf commit required. Run git describe in IDF_PATH to get the right format
		
endmenu
```

main/CMakeLists.txt
```cmake
...

register_component()

if (NOT CONFIG_REQUIRED_IDF STREQUAL IDF_VER)
    message(FATAL_ERROR "Invalid IDF version\nconfig \"${CONFIG_REQUIRED_IDF}\"\nactual \"${IDF_VER}\"")
endif()
```

closed 2019-02-01 by MightyPork