espressif/esp-idf #3002: Export bss.gWpaSm symbol to aid debugging Wi-Fi problems

mikaelkanstrup, 2019-01-29:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): v3.3-beta1-268-g5c88c59
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [external 5V]

## Problem Description

This is rather a feature/improvement request than an issue.

To investigate Wi-Fi related problems the internals of wpa_supplicant gives valuable information. Would it be possible for you to export the gWpaSm symbol of components/esp32/lib/libwpa.a so that one can write something like this from the code:

extern struct wpa_sm *gWpaSm;

To access the wpa_supplicant internals.

As an example. In a project we run we connect to a WPA2 Enterprise Wi-Fi network. To be able to decrypt the wifi traffic we need to get hold of the PMK generated during the EAP process / Wi-Fi connection establishment.

Having this symbol available in the elf-file makes possible to access the pmk array of wpa_sm struct. With the pmk content we can use for example Wireshark to decrypt and analyze traffic captured with a Wi-Fi sniffer.

The following hack makes this possible already today but it's error prone due to offsets in binary might change between builds so would be great if you can consider exporting the symbol instead.

```
xtensa-esp32-elf-objcopy --globalize-symbol=.bss.gWpaSm $IDF_PATH/components/esp32/lib/libwpa.a && \
make && \
PMK_ADDR=`xtensa-esp32-elf-objdump -t build/tracking-observer.elf  | grep gWpaSm | cut -f 1 -d " "` && \
make EXTRA_CFLAGS=-DDEBUG_PMK=0x$PMK_ADDR flash
```

In wifi_event_handler:
```
#ifdef DEBUG_PMK
#define PMK_LEN 32
struct wpa_sm
{
    uint8_t pmk[PMK_LEN];
    size_t pmk_len;
    /* more */
};
static struct wpa_sm *wifi_wpa_sm = (struct wpa_sm *)DEBUG_PMK;
#endif /* DEBUG_PMK */

...

#ifdef DEBUG_PMK
        case SYSTEM_EVENT_STA_CONNECTED:
            ESP_LOG_BUFFER_HEXDUMP("KEY", wifi_wpa_sm->pmk,
                                   wifi_wpa_sm->pmk_len,
                                   ESP_LOG_WARN);
            break;
#endif /* DEBUG_PMK */

```

### Expected Behavior

gWpaSm symbol is possible to access from code.

### Actual Behavior

gWpaSm symbol is hidden

liuzfesp, 2019-02-02:
HI @mikaelkanstrup ESP WPS implementation is different than the original WPS in wpa_supplicant, so I don't think it's helpful to export gWpaSm for your debugging. Anyway, we plan to open source the whole wpa_supplicant in 2019, just wait for it. 
Call @sagb2015 for help.

mikaelkanstrup, 2019-02-04:
> ESP WPS implementation is different than the original WPS in wpa_supplicant, so I don't think it's helpful to export gWpaSm for your debugging.

gWpaSm from what I can tell is the internal state machine of wpa_supplicant (from mainline struct wpa_sm declared in /src/rsn_supp/wpa_i.h) not WPS. But yeah I can understand that it's been modified from original sources. This was kind of a quick alternative to getting the full sources of wpa_supplicant.

> Anyway, we plan to open source the whole wpa_supplicant in 2019, just wait for it.

That's great news! Thanks! I'll look forward to it.

closed 2019-02-04 by mikaelkanstrup