espressif/esp-idf #3003: Request to provide source code of EspressIf's hostap fork

mikaelkanstrup, 2019-01-29:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): v3.3-beta1-268-g5c88c59
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [external 5V]

## Problem Description

This is rather a request for improvements than an issue.

After working with ESP-IDF for a while I've discovered, debugged and presented solutions to a number of Wi-Fi related errors. With EspressIf's fork of the hostap open source project included only as binary object files it's hard/impossible to debug and solve problems still present.

For example EAP reauthentication still does not work with latest ESP-IDF from master:
https://github.com/espressif/esp-idf/issues/2324

From the code that you do provide (components/wpa_supplicant) I've noticed that the code is based on a rather old version of wpa_supplicant. Over the years plenty of security fixes including fixes for the infamous KRACK attacks have been applied to the mainline hostap. Without the source code it is not possible to tell what's been integrated in EspressIf's fork of hostap making this not only a nice to have thing but also a security concern for projects built on top of ESP-IDF.

In short this issue is a request for you to prove the full source code of EspressIf's hostap (wpa_supplicant/hostapd) fork.

mikaelkanstrup, 2019-02-04:
In issue https://github.com/espressif/esp-idf/issues/3002 @liuzfesp told me you are going to release your wpa_supplicant as open source during 2019. That's great news!

Oh and for the record I should also mention that I've read your official statement about KRACK's been handled already:
https://www.espressif.com/en/media_overview/news/espressif-releases-patches-wifi-vulnerabilities-cert-vu228519

Do you want me close this issue or can we keep it as a placeholder for the upcoming release of wpa_supplicant source code?

mikaelkanstrup, 2019-08-28:
@liuzfesp Are you able to share any details about progress and time plan for this request?

mikaelkanstrup, 2019-08-28:
@liuzfesp Thanks! I guess I was expecting to find some more EAP state related code and while quickly searching through the code I didn't notice that in fact quite some additional parts are available since:

c13968302 supplicant/esp_wifi: move supplicant to idf

liuzfesp, 2019-08-28:
@mikaelkanstrup, current IDF contains almost all of ESP supplicant. I have to say ESP supplicant is very different than the official wpa_supplicant, it doesn't contains everything in official wpa_supplicant.

sagb2015, 2019-08-28:
@mikaelkanstrup Almost all the part of supplicant and EAP state machine inherited from supplicant is already in IDF now. Code flow to act based on authentication result is still closed and will remain so in near future. We do not support "enterprise mode" with SoftAP mode yet. Closing this issue. Feel free to reopen in case of further doubts/requests.

closed 2019-08-28 by sagb2015