espressif/esp-idf #3005: Http Server: Allow closing of connection if client attempts to access invalid URI

lucascoxBAF, 2019-01-29:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version: v3.3-beta1-250-g58acac883
- Build System: CMake
- Compiler version: 1.22.0-73-ge28a011
- Operating System: Linux
- Power Supply: 3v3 board from external power supply

## Problem Description
Currently, if curl (or any other http client) attempts to put a file on an esp server at an invalid uri, the esp properly replies with a 404 not found. However, there is no user callback that an invalid URI is accessed, so I have no way to close the connection. 

So, what ends up happening is the client sees the 404 error and "aborts the transaction." However, Ubuntu's networking layer (I assume) keeps sending the file so that the ESP32 had to discard all of it. This can take a VERY long time for big files. 

Since there is no callback to user code that alerts of an invalid URI access, I have no way to close the connection so that it doesn't continue to spam the HTTP server. 

The actual uri for my http server is **192.168.4.1:80/fileEndpoint**

I type the following command:

curl -T changelog.txt 192.168.4.1:80/invalidUri

#### Output of curl

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   109  100    22  100    87     17     69  0:00:01  0:00:01 --:--:--    87
This URI doesn't exist

## Debug Logs

D (301802) httpd: httpd_server: processing listen socket 54
D (301802) httpd: httpd_accept_conn: newfd = 59
D (301802) httpd_sess: httpd_sess_new: fd = 59
D (301802) httpd: httpd_accept_conn: complete
D (301802) httpd: httpd_server: doing select maxfd+1 = 60
D (301812) httpd: httpd_server: processing socket 59
D (301812) httpd_sess: httpd_sess_process: httpd_req_new
D (301822) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (301832) httpd_txrx: httpd_recv_with_opt: received length = 128
D (301832) httpd_parse: read_block: received HTTP request block size = 128
D (301842) httpd_parse: cb_url: message begin
D (301842) httpd_parse: cb_url: processing url = /invalidUri
D (301852) httpd_parse: verify_url: received URI = /invalidUri
D (301862) httpd_parse: cb_header_field: headers begin
D (301862) httpd_txrx: httpd_unrecv: length = 103
D (301872) httpd_parse: pause_parsing: paused
D (301872) httpd_parse: cb_header_field: processing field = Host
D (301882) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (301882) httpd_txrx: httpd_recv_with_opt: pending length = 103
D (301892) httpd_parse: read_block: received HTTP request block size = 103
D (301892) httpd_parse: continue_parsing: skip pre-parsed data of size = 5
D (301902) httpd_parse: continue_parsing: un-paused
D (301912) httpd_parse: cb_header_value: processing value = 192.168.4.1:80
D (301912) httpd_parse: cb_header_field: processing field = User-Agent
D (301922) httpd_parse: cb_header_value: processing value = curl/7.58.0
D (301932) httpd_parse: cb_header_field: processing field = Accept
D (301932) httpd_parse: cb_header_value: processing value = */*
D (301942) httpd_parse: cb_header_field: processing field = Content-Length
D (301952) httpd_parse: cb_header_value: processing value = 87
D (301952) httpd_parse: cb_header_field: processing field = Expect
D (301962) httpd_parse: cb_header_value: processing value = 100-continue
D (301972) httpd_parse: parse_block: parsed block size = 103
D (301972) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (301982) httpd_txrx: httpd_recv_with_opt: received length = 4
D (301982) httpd_parse: read_block: received HTTP request block size = 4
D (301992) httpd_parse: cb_header_value: processing value =
D (302002) httpd_parse: cb_headers_complete: bytes read     = 133
D (302002) httpd_parse: cb_headers_complete: content length = zu
D (302012) httpd_parse: parse_block: parsed block size = 107
D (302012) httpd_txrx: httpd_recv_with_opt: requested length = 128
I (302412) Ble Driver: event: ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT
I (302412) Ble Driver: Stop adv successfully

I (302412) Ble Driver: event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
I (302422) Ble Driver: scan start success
D (302812) httpd_txrx: httpd_recv_with_opt: received length = 87
D (302812) httpd_parse: read_block: received HTTP request block size = 87
D (302812) httpd_txrx: httpd_unrecv: length = 87
D (302822) httpd_parse: pause_parsing: paused
D (302822) httpd_parse: cb_on_body: body begins
D (302832) httpd_parse: httpd_parse_req: parsing complete
D (302832) httpd_uri: httpd_uri: request for /invalidUri with type 4
D (302842) httpd_uri: httpd_find_uri_handler2: [0] = /invalidUri
W (302842) httpd_uri: httpd_uri: URI '/invalidUri' not found
W (302852) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI doesn't exist
D (302862) httpd_txrx: httpd_send_all: sent = 69
D (302862) httpd_txrx: httpd_send_all: sent = 2
D (302872) httpd_txrx: httpd_send_all: sent = 22
D (302872) httpd_sess: httpd_sess_process: httpd_req_delete
D (302882) httpd_txrx: httpd_req_recv: remaining length = 87
D (302882) httpd_txrx: httpd_recv_with_opt: requested length = 31
D (302892) httpd_txrx: httpd_recv_with_opt: pending length = 87
D (302902) httpd_txrx: httpd_req_recv: received length = 31
D (302902) httpd_parse: httpd_req_delete: purging data : this is a line that should be i
D (302912) httpd_txrx: httpd_req_recv: remaining length = 56
D (302922) httpd_txrx: httpd_recv_with_opt: requested length = 31
D (302922) httpd_txrx: httpd_recv_with_opt: pending length = 56
D (302932) httpd_txrx: httpd_req_recv: received length = 31
D (302932) httpd_parse: httpd_req_delete: purging data : gnored by the http server!
This
D (302942) httpd_txrx: httpd_req_recv: remaining length = 25
D (302952) httpd_txrx: httpd_recv_with_opt: requested length = 25
D (302952) httpd_txrx: httpd_recv_with_opt: pending length = 25
D (302962) httpd_txrx: httpd_req_recv: received length = 25
D (302962) httpd_parse: httpd_req_delete: purging data :  is another line...ditto

D (302972) httpd_sess: httpd_sess_process: success
D (302982) httpd: httpd_server: doing select maxfd+1 = 60
D (302982) httpd: httpd_server: processing socket 59
D (302992) httpd_sess: httpd_sess_process: httpd_req_new
D (302992) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (303002) httpd_txrx: httpd_recv_with_opt: received length = 0
D (303012) httpd_parse: read_block: connection closed
D (303012) httpd: httpd_server: closing socket 59
D (303022) httpd_sess: httpd_sess_delete: fd = 59
D (303022) httpd: httpd_server: doing select maxfd+1 = 56

anurag-kar, 2019-01-30:
@lucascoxBAF Thanks for reporting this issue. We will try to resolve this ASAP.

closed 2019-02-26 by igrr

lucascoxBAF, 2019-02-27:
@anurag-kar @igrr Thank you for adding the error handler! That's exactly what I needed!