espressif/esp-idf #3007: RMT: how do I convert RAW IR codes to short data?

keesus, 2019-01-30:
Development Kit: ESP32-DevKitC
Kit version (for WroverKit/PicoKit/DevKitC):
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v3.3-dev-132-gbb4714671
Build System: CMake
Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 5.2.0
Operating System: macOS
Power Supply: USB

Hi. I am using ESP32_IR_Remote library which returns RAW IR codes when receiving.
Because most IR codes consist of long integer array, I have no idea how to send/receive it with server.

My IR code looks like this.

unsigned int arr[] = [4700, 400, 350, 700, 1200, 450, 3200, 1000, 320, 500, 300, 100, 450, 500, 1200, 4000, 1200]

and it could reaches more than 500 indexes according to IR command. 

does anyone has a hint ?

koobest, 2019-02-15:
Hi, @keesus
Do you want to transfer the entire arr[] that might contain a lot of integer data? I think the transmitter is fine, but the receiver can only receive 31 bytes of data per transfer, unless the arr[] is transmitted multiple times. how about using WiFi or Bluetooth to transfer data (I don't know your application scenario.)?

Alvin1Zhang, 2019-03-14:
@keesus Hi, could you help share if any updates for the issue? Thanks

keesus, 2019-03-15:
@koobest @Alvin1Zhang 

I have resolved this issue by using HTTP. 

Thanks

closed 2019-03-15 by keesus

Nakul93, 2022-01-13:
How to transmit this same data using esp-idf rmt module. I'm able to receive it, I also get data in the same format:

> unsigned int arr[] = [4700, 400, 350, 700, 1200, 450, 3200, 1000, 320, 500, 300, 100, 450, 500, 1200, 4000, 1200]

elektronika-ba, 2023-02-26:
> How to transmit this same data using esp-idf rmt module. I'm able to receive it, I also get data in the same format:
> 
> > unsigned int arr[] = [4700, 400, 350, 700, 1200, 450, 3200, 1000, 320, 500, 300, 100, 450, 500, 1200, 4000, 1200]

Did you figure out how to send this bit-stream back using RMT module?