espressif/esp-idf #3008: I2S initialization breaks the semaphore from ISR

vgonet, 2019-01-30:
Hello,

## Problem Description

I'm trying to use the I2S-DAC based on https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s_adc_dac in my project to play sounds.

My project has several FreeRTOS tasks. One of my task is called by semaphore from a timer interrupt.

When I initialize the I2S, all tasks continue to run except the task called by semaphore. I think the timer interrupt is never reached. No errors are displayed in the logs.

## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version: 3.1.2
- Operating System:  Windows

### Code to reproduce this issue
```cpp
void Sound_Init(void)
{
  int16_t i2s_num = I2S_NUM;

  i2s_config_t i2s_config =
  {
    .mode                 = (I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
    .sample_rate          = I2S_SAMPLE_RATE,
    .bits_per_sample      = I2S_SAMPLE_BITS,
    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
    .channel_format       = I2S_FORMAT,
    .intr_alloc_flags     = 0,
    .dma_buf_count        = 2,
    .dma_buf_len          = 1024
  };

  // Install and start I2S driver
  i2s_driver_install(i2s_num, &i2s_config, 0, NULL);

  // Init DAC pad
  i2s_set_dac_mode(I2S_DAC_CHANNEL_RIGHT_EN);

  ESP_LOGI(Tag, "Sound is initialized");
}
```

The line of code which "kill" the timer interrupt is:
```cpp
i2s_driver_install(i2s_num, &i2s_config, 0, NULL);
```

The timer interrupt is:
```cpp
static void IRAM_ATTR ISR_TimerGroup0(void* para)
{
  static BaseType_t higherPriorityTaskWoken = pdFALSE;

  ...

  // Clear the interrupt and update the alarm time for the timer with without reload
  if ((intr_status & BIT(timer_idx)) && (timer_idx == TIMER_0))
  {
    TIMERG0.int_clr_timers.t0 = 1;

    xSemaphoreGiveFromISR(Timer125usSemaphore, &higherPriorityTaskWoken);

    if (higherPriorityTaskWoken != pdFALSE)
    {
      portYIELD_FROM_ISR();
    }
  }
}
```

Best regards,
Vincent

costaud, 2019-01-31:
Would you please provide your code which initialize the timer?

vgonet, 2019-01-31:
Hello,

When I comment the lines 129 and 145-152, the logs print the following
message:

I (424) sensors: Sensors are initialized
I (424) mode: Mode is initialized
I (434) mode: VM Mode is initialized
I (434) timer: Group 0 Timer 0 is initialized
I (444) timer: Group 0 Timer 1 is initialized
I (444) behavior: Start Behavior Task
I (444) sensors: Start Comm Task
I (454) sensors: Start Sensor Task
I (454) sensors: Run Sensor Task
I (464) sensors: Run Sensor Task
I (464) sensors: Run Sensor Task
I (464) sensors: Run Sensor Task
I (474) sensors: Run Sensor Task

So the Sensor task ir running. If the lines 129 and 145-152 are
uncommented, the sound is played but the Sensor task is not running:

I (459) sensors: Sensors are initialized
I (459) mode: Mode is initialized
I (469) mode: VM Mode is initialized
I (469) I2S: DMA Malloc info, datalen=blocksize=4092, dma_buf_count=2
I (479) I2S: DMA Malloc info, datalen=blocksize=4092, dma_buf_count=2
I (479) I2S: PLL_D2: Req RATE: 16000, real rate: 1004.000, BITS: 16, CLKM:
83, BCK: 60, MCLK: 83.333, SCLK: 32128.000000, diva: 64, divb: 21
I (499) sound: Sound is initialized
I (499) timer: Group 0 Timer 0 is initialized
I (509) timer: Group 0 Timer 1 is initialized
I (509) behavior: Start Behavior Task
I (519) I2S: PLL_D2: Req RATE: 16000, real rate: 1004.000, BITS: 16, CLKM:
83, BCK: 60, MCLK: 83.333, SCLK: 32128.000000, diva: 64, divb: 21
I (529) sensors: Start Comm Task
I (539) sensors: Start Sensor Task
I (539) sensors: Run Sensor Task
I (5519) I2S: PLL_D2: Req RATE: 16000, real rate: 1004.000, BITS: 16, CLKM:
83, BCK: 60, MCLK: 83.333, SCLK: 32128.000000, diva: 64, divb: 21

The initialization of the timer and the ISR used for the notification is in
timer.c (Timer_Init())

The "notification take" is in the file sensors.c (Sensor_RunTask()).

I've try with semaphore instead of notification and I've exactly the same
behaviour.

I hope this code will help you to reproduce the issue.

Thank you
Best regards
Vincent

vgonet, 2019-01-31:
Hello,

Sorry the lines 129 and 145-152 are in the file main.c

costaud, 2020-05-26:
@vgonet  Sorry, we failed to reproduce this issue with the limited given code. Feel free to reopen it if you get further information. Thanks.

closed 2020-05-26 by costaud