espressif/esp-idf #3011: nvs_partition_gen.py crashes on writing a big blob

Oromis, 2019-01-30:
## Environment

- Development Kit: none
- Module or chip used: none
- IDF version: v3.1.2
- Build System: n/a
- Compiler version: n/a
- Operating System: Windows + Linux
- Power Supply: n/a

## Problem Description

I try to create an NVS partition using the nvs_partition_gen.py utility. According to the official docs (https://docs.espressif.com/projects/esp-idf/en/v3.1.2/api-reference/storage/nvs_flash.html), the maximum BLOB size is 1984 (without multi-page BLOB support). I want to write multiple such BLOBs using the following command line on Python 3.7:

`python nvs_partition_gen.py --input nvs_data.csv --output nvs.bin --size 983040 --version v1`

nvs_data.csv looks like this:

```
key,type,encoding,value
hotspot,namespace,,
index.html____0,data,u32,6216
index.html____1,data,base64,PCFET0NUWVBFIGh0bWw+PGh0bWw+PGhlYWQ+PG1ldGEgY2hhcnNldD0idXRmLTgiPjxtZXRhIG5hbWU9InZpZXdwb3J0IiBjb250ZW50PSJ3aWR0aD1kZXZpY2Utd2lkdGgsaW5pdGlhbC1zY2FsZT0xIj48dGl0bGUgZGF0YS1pMThuPSJ0aXRsZSI+Vm9sdFN0b3JhZ2UgYmF0dGVyeSBXaUZpIHNldHVwPC90aXRsZT48bGluayByZWw9InNob3J0Y3V0IGljb24iIGhyZWY9ImZhdmljb24ucG5nIj48bGluayByZWw9InN0eWxlc2hlZXQiIHR5cGU9InRleHQvY3NzIiBocmVmPSJtYWluLm1pbi5jc3MiPjwvaGVhZD48Ym9keSBpZD0iY29ubmVjdGlvbiIgY2xhc3M9ImRpc2Nvbm5lY3RlZCI+PGRpdiBjbGFzcz0ic3BhY2VyIHNwYWNlci0xIj48L2Rpdj48c3ZnIHhtbG5zPSJodHRwOi8vd3d3LnczLm9yZy8yMDAwL3N2ZyIgdmlld0JveD0iMCAwIDEwMCA2MC4wMTQiIGNsYXNzPSJsb2dvIj48cGF0aCBjbGFzcz0iYSIgZD0iTTU0Ljc2IDM5LjMyaC05LjFhLjg3Ljg3IDAgMCAwIDAgMS43M2g5LjFhLjg3Ljg3IDAgMSAwIDAtMS43M3pNNTAuMjMgMGExNy45MyAxNy45MyAwIDAgMC02LjQzIDM0LjY4IDE1LjM3IDE1LjM3IDAgMCAwLS45NC00LjE3IDEzLjUxIDEzLjUxIDAgMCAwLTIuMDYtMy40QTEzLjEzIDEzLjEzIDAgMCAxIDUwLjE5IDQuOGguMDZhMTMuMTMgMTMuMTMgMCAwIDEgOS4zOCAyMi4zMSAxMy41IDEzLjUgMCAwIDAtMi4wNyAzLjQxIDE1LjQxIDE1LjQxIDAgMCAwLS45NCA0LjE3QTE3LjkzIDE3LjkzIDAgMCAwIDUwLjIzLjAxek01NS45NSAzNi4xN0g0NC40OGEuODcuODcgMCAwIDAgMCAxLjczaDExLjQ3YS44Ny44NyAwIDEgMCAwLTEuNzN6TTkzLjA3IDQ5LjA3SDEwMHYxLjg5aC00Ljc1djIuNTFoMy44MXYxLjg1aC0zLjgxdjIuNTdIMTAwdjEuODloLTYuOTN6TTkwLjg3IDU0LjA1djUuNTZhMjQuMTUgMjQuMTUgMCAwIDEtMy42NS4zNnEtMi41IDAtMy40NC0xLjM0YTcuNTQgNy41NCAwIDAgMS0uOTQtNC4yNSA3LjA3IDcuMDcgMCAwIDEgMS00LjIgMy45IDMuOSAwIDAgMSAzLjMyLTEuMjkgMTYuNjUgMTYuNjUgMCAwIDEgMy4xNi4zM2wuNTcuMTEtLjA2IDEuNjhhMzAuNDIgMzAuNDIgMCAwIDAtMy4wOS0uMiAzLjYzIDMuNjMgMCAwIDAtMS43NC4zIDEuNzEgMS43MSAwIDAgMC0uNzIgMSAxMC4zOSAxMC4zOSAwIDAgMC0uMjMgMi42MSA1LjI4IDUuMjggMCAwIDAgLjQ2IDIuNTkgMi4wNiAyLjA2IDAgMCAwIDEuODkuNzRsMS4zMS0uMDZ2LTIuMDVoLTF2LTEuODl6TTc5LjQ2IDQ5LjExbDIuNDEgMTAuN2gtMi4xOGwtLjQyLTEuOTRoLTMuNzVsLS40MiAxLjk0aC0yLjE3bDIuNC0xMC43em0tLjYgNi44OGwtMS4wOS01LjA5SDc3bC0xLjA5IDUuMDl6TTcwLjA2IDU1LjY4bDEuNTMgNC4wOWgtMi4zOEw2OCA1Ni4xOWgtMi4xM3YzLjU5SDYzLjdWNDkuMTFoNC4xNnEzLjc4IDAgMy43OCAzLjUzYTMuMzkgMy4zOSAwIDAgMS0xLjU4IDMuMDR6bS0xLTEuODNhMS45NCAxLjk0IDAgMCAwIC4zOC0xLjI0IDEuOCAxLjggMCAwIDAtLjQtMS4yMyAxLjQ2IDEuNDYgMCAwIDAtMS4xNi0uNDZoLTJ2My40aDJhMS4zOSAxLjM5IDAgMCAwIDEuMTItLjQ2ek02MC44OSA1MC4yNmE3LjE5IDcuMTkgMCAwIDEgMSA0LjIxIDcgNyAwIDAgMS0xIDQuMTYgNCA0IDAgMCAxLTMuNDQgMS4zM0E0IDQgMCAwIDEgNTQgNTguNjRhNyA3IDAgMCAxLTEtNC4xNiA3LjE5IDcuMTkgMCAwIDEgMS00LjIxIDQgNCAwIDAgMSAzLjQ0LTEuMzcgNCA0IDAgMCAxIDMuNDUgMS4zNnptLTEuNyA3YTYgNiAwIDAgMCAuNDctMi43NyA2LjMxIDYuMzEgMCAwIDAtLjQ3LTIuODMgMi4xNCAyLjE0IDAgMCAwLTMuNDYgMCA2LjMxIDYuMzEgMCAwIDAtLjQ3IDIuODMgNiA2IDAgMCAwIC40NyAyLjc3IDIuMjQgMi4yNCAwIDAgMCAzLjQ4IDB6TTQ5LjMxIDU5Ljc4aC0yLjE3VjUxaC0yLjg2di0xLjkzaDcuODdWNTFoLTIuODR6TTQyLjUzIDU0LjU5YTIuNyAyLjcgMCAwIDEgLjYzIDIgMy4xNiAzLjE2IDAgMCAxLTEgMi41NQ==
index.html____2,data,base64,IDQgNCAwIDAgMS0yLjY4Ljg3IDE0Ljg3IDE0Ljg3IDAgMCAxLTMtLjM4bC0uNTctLjExLjIyLTEuNjlhMjYuMDkgMjYuMDkgMCAwIDAgMy4yMy4yOHExLjY0IDAgMS42NC0xLjQyYS45Mi45MiAwIDAgMC0uMzgtLjc5IDUuNTEgNS41MSAwIDAgMC0xLjU0LS42MSA2LjIzIDYuMjMgMCAwIDEtMi41Ny0xLjI0IDIuNjIgMi42MiAwIDAgMS0uNzUtMiAyLjg5IDIuODkgMCAwIDEgMS0yLjM5IDQuMTYgNC4xNiAwIDAgMSAyLjY4LS43OCAxNy45MSAxNy45MSAwIDAgMSAzIC4zMWwuNTYuMTEtLjE3IDEuNzJjLTEuNDgtLjE3LTIuNTMtLjI1LTMuMTMtLjI1cS0xLjcyIDAtMS43MiAxLjEzYS44Ni44NiAwIDAgMCAuNDMuNzcgMTEgMTEgMCAwIDAgMiAuNzcgNS41NiA1LjU2IDAgMCAxIDIuMTIgMS4xNXoiPjwvcGF0aD48ZyBjbGFzcz0iYiI+PHBhdGggY2xhc3M9ImEiIGQ9Ik0zMS44MSA3OS42N2gtMi4xN3YtOC43OWgtMi44N3YtMS45Mmg3Ljg4djEuOTJoLTIuODR6TTIwLjY3IDY4Ljk2aDIuMTd2OC43OWgzLjkydjEuOTJoLTYuMDl6TTE3LjggNzAuMTVhNy4yMSA3LjIxIDAgMCAxIDEgNC4yMSA3IDcgMCAwIDEtMSA0LjE2IDQgNCAwIDAgMS0zLjQ0IDEuMzMgNCA0IDAgMCAxLTMuNDQtMS4zMyA3IDcgMCAwIDEtMS00LjE2IDcuMTkgNy4xOSAwIDAgMSAxLTQuMjEgNCA0IDAgMCAxIDMuNDQtMS4zNyA0IDQgMCAwIDEgMy40NCAxLjM3em0tMS43IDdhNiA2IDAgMCAwIC40Ny0yLjc3IDYuMzEgNi4zMSAwIDAgMC0uNDctMi44MyAxLjc5IDEuNzkgMCAwIDAtMS43My0uODggMS43OSAxLjc5IDAgMCAwLTEuNzMuODggNi4zMSA2LjMxIDAgMCAwLS40NyAyLjgzIDYgNiAwIDAgMCAuNDYgMi43NyAxLjgyIDEuODIgMCAwIDAgMS43NC44MyAxLjgyIDEuODIgMCAwIDAgMS43Mi0uODR6TTAgNjguOTZoMi4yOGwxLjg2IDguODJoLjcxbDEuODYtOC44MmgyLjI4TDYuNSA3OS42N0gyLjQ5eiIgdHJhbnNmb3JtPSJ0cmFuc2xhdGUoMCAtMTkuODkpIj48L3BhdGg+PC9nPjwvc3ZnPjxkaXYgY2xhc3M9InNwYWNlciBzcGFjZXItMSI+PC9kaXY+PGRpdiBjbGFzcz0iY29udGVudCI+PGRpdiBpZD0iZ3JhcGhpYyIgY2xhc3M9ImludmlzaWJsZSI+PGgxIGlkPSJjb25uZWN0aW9uLXN0YXR1cyI+PC9oMT48ZGl2IGNsYXNzPSJjb25uZWN0aW9uLWdyYXBoaWMiPjxzdmcgeG1sbnM9Imh0dHA6Ly93d3cudzMub3JnLzIwMDAvc3ZnIiB2aWV3Qm94PSIwIDAgNDUuNiA1OC43IiBjbGFzcz0iY29ubmVjdC1pY29uIj48cGF0aCBjbGFzcz0ic3ltYm9sIiBkPSJNMjUuMyAxOC4ydi0zLjZoLTUuOHYzLjZoLTMuNnYzMmgxM3YtMzJ6Ii8+PHBhdGggY2xhc3M9InN5bWJvbCIgZD0iTTQzLjkgNTdIMS43VjE4LjVMMjIuOCAyLjJsMjEuMSAxNi4zeiIvPjxwYXRoIGNsYXNzPSJzeW1ib2wiIGQ9Ik0xOS44IDQ1LjNIMjUiLz48cGF0aCBjbGFzcz0ic3ltYm9sIiBkPSJNMTkuOCA0MC44SDI1Ii8+PHBhdGggY2xhc3M9InN5bWJvbCIgZD0iTTE5LjggMzYuNEgyNSIvPjxwYXRoIGNsYXNzPSJzeW1ib2wiIGQ9Ik0xOS44IDMxLjhIMjUiLz48cGF0aCBjbGFzcz0ic3ltYm9sIiBkPSJNMTkuOCAyNy40SDI1Ii8+PC9zdmc+IDxzdmcgeG1sbnM9Imh0dHA6Ly93d3cudzMub3JnLzIwMDAvc3ZnIiB2aWV3Qm94PSIwIDAgMTAwIDEwMCIgY2xhc3M9ImNvbm5lY3QtaWNvbiBpbnRlcm5ldC1pY29uIj48ZyBjbGFzcz0ic3ltYm9sIj48Y2lyY2xlIHI9IjQ3LjA4NyIgY3k9IjUwIiBjeD0iNTAiLz48ZWxsaXBzZSByeT0iNDcuMDAxIiByeD0iMjcuMDAxIiBjeT0iNTAiIGN4PSI1MCIvPjxwYXRoIGQ9Ik0xMS42NjcgMjIuNzQ5YzExLjk3NyA2LjIzMyAyNS44NjYgNi43NCAzNy44MyA2LjcxNyAxMi43NzIuMDQ5IDI2Ljk4OC0xLjQxNCAzOC4xODQtNy43NzgiLz48cGF0aCBkPSJNMTIuMDIxIDc2LjMwMmMxMS45NzctNi4yMzMgMjUuODY2LTYuNzQgMzcuODMtNi43MThDNjIuNjIzIDY5LjUzNSA3Ni44NCA3MSA4OC4wMzUgNzcuMzYzIi8+PA==
index.html____3,data,base64,cGF0aCBkPSJNNTAuMDMgMi45MDhsLS4wMDQgOTQuMjI2Ii8+PHBhdGggZD0iTTk2Ljk1NyA1MGwtOTMuOTE0LS4wMDQiLz48L2c+PC9zdmc+PGRpdiBjbGFzcz0iY29ubmVjdGlvbi1hbmltIj48L2Rpdj48L2Rpdj48L2Rpdj48L2Rpdj48ZGl2IGNsYXNzPSJzcGFjZXIgc3BhY2VyLTEiPjwvZGl2Pjxub3NjcmlwdD48aDIgZGF0YS1pMThuPSJub3NjcmlwdCI+UGxlYXNlIGVuYWJsZSBKYXZhU2NyaXB0IGluIHlvdXIgYnJvd3NlciE8L2gyPjwvbm9zY3JpcHQ+PGRpdiBjbGFzcz0iY29udGVudCI+PGRpdiBpZD0iZm9ybS1jb250ZW50Ij48aDEgZGF0YS1pMThuPSJmb3JtLmhlYWRpbmciPjwvaDE+PGZvcm0gaWQ9IndpZmktZm9ybSI+PGRpdiBjbGFzcz0iaW5wdXQtcm93Ij48c2VsZWN0IGlkPSJzc2lkIiBuYW1lPSJzc2lkIj48L3NlbGVjdD4gPGxhYmVsIGZvcj0ic3NpZCIgZGF0YS1pMThuPSJmb3JtLnNzaWQiPjwvbGFiZWw+PC9kaXY+PGRpdiBjbGFzcz0iaW5wdXQtcm93Ij48aW5wdXQgaWQ9InB3IiB0eXBlPSJwYXNzd29yZCIgbmFtZT0icGFzcyI+IDxsYWJlbCBmb3I9InB3IiBkYXRhLWkxOG49ImZvcm0ucHciPjwvbGFiZWw+PC9kaXY+PGRpdiBjbGFzcz0iYWN0aW9uLXJvdyI+PGJ1dHRvbiB0eXBlPSJzdWJtaXQiIGNsYXNzPSJidG4iIGRhdGEtaTE4bj0iZm9ybS5zdWJtaXQiPjwvYnV0dG9uPjwvZGl2PjwvZm9ybT48L2Rpdj48ZGl2IGlkPSJzdWNjZXNzIiBjbGFzcz0iaW52aXNpYmxlIj48aDIgZGF0YS1pMThuPSJzdWNjZXNzLmhlYWRpbmciPjwvaDI+PHAgZGF0YS1pMThuPSJzdWNjZXNzLmJvZHkiPjwvcD48ZGl2IGNsYXNzPSJhY3Rpb24tcm93IGlucHV0LXJvdyI+PGEgaHJlZj0iaHR0cHM6Ly9hcHAudm9sdHN0b3JhZ2UuY29tIiB0YXJnZXQ9Il9ibGFuayI+Vm9sdFN0b3JhZ2UgQXBwPC9hPjwvZGl2PjwvZGl2PjxkaXYgaWQ9ImZhaWwiIGNsYXNzPSJpbnZpc2libGUiPjxoMiBkYXRhLWkxOG49ImZhaWwuaGVhZGluZyI+PC9oMj48cCBkYXRhLWkxOG49ImZhaWwuYm9keSI+PC9wPjxkaXYgY2xhc3M9ImFjdGlvbi1yb3cgaW5wdXQtcm93Ij48YnV0dG9uIHR5cGU9ImJ1dHRvbiIgaWQ9ImJhY2siIGRhdGEtaTE4bj0iZmFpbC5hY3Rpb24iPlRyeSBhZ2FpbjwvYnV0dG9uPjwvZGl2PjwvZGl2PjxkaXYgY2xhc3M9ImFjdGlvbi1yb3cgbGFuZy1yb3ciPjxidXR0b24gdHlwZT0iYnV0dG9uIiBjbGFzcz0iaWNvbiIgaWQ9ImxhbmciPjxkaXYgaWQ9ImxhbmctZW4iPjxzdmcgeG1sbnM9Imh0dHA6Ly93d3cudzMub3JnLzIwMDAvc3ZnIiB2aWV3Qm94PSIwIDAgNjAgMzAiIHdpZHRoPSI0NSIgaGVpZ2h0PSIzMCI+PGNsaXBQYXRoIGlkPSJ0Ij48cGF0aCBkPSJNMzAsMTUgaDMwIHYxNSB6IHYxNSBoLTMwIHogaC0zMCB2LTE1IHogdi0xNSBoMzAgeiIvPjwvY2xpcFBhdGg+PHBhdGggZD0iTTAsMCB2MzAgaDYwIHYtMzAgeiIgZmlsbD0iIzAwMjQ3ZCIvPjxwYXRoIGQ9Ik0wLDAgTDYwLDMwIE02MCwwIEwwLDMwIiBzdHJva2U9IiNmZmYiIHN0cm9rZS13aWR0aD0iNiIvPjxwYXRoIGQ9Ik0wLDAgTDYwLDMwIE02MCwwIEwwLDMwIiBjbGlwLXBhdGg9InVybCgjdCkiIHN0cm9rZT0iI2NmMTQyYiIgc3Ryb2tlLXdpZHRoPSI0Ii8+PHBhdGggZD0iTTMwLDAgdjMwIE0wLDE1IGg2MCIgc3Ryb2tlPSIjZmZmIiBzdHJva2Utd2lkdGg9IjEwIi8+PHBhdGggZD0iTTMwLDAgdjMwIE0wLDE1IGg2MCIgc3Ryb2tlPSIjY2YxNDJiIiBzdHJva2Utd2lkdGg9IjYiLz48L3N2Zz48L2Rpdj48c3BhbiBpZD [... cut ...] yYz0iYXBwLm1pbi5qcyIgY2hhcnNldD0idXRmLTgiPjwvc2NyaXB0PjwvYm9keT48L2h0bWw+
```

The command fails with `IndexError: array index out of range` in `nvs_partition_gen.py", line 119`.

### Expected Behavior

The command completes without error and generates the NVS partition binary.

### Actual Behavior

The command crashes, no binary is created.

### Steps to repropduce

1. Copy above CSV file to a local file on your machine
2. Run the above command (adjusting paths accordingly)
3. Observe the error

LaurentLouf, 2019-02-01:
I have a problem related to this one (I think) using `hex2bin` on larges hexadecimal strings (512 characters, so 256 bytes, which is not a lot). The complete error log : 

```
Traceback (most recent call last):
  File "components/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 709, in <module>
    main()
  File "components/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 705, in main
    nvs_part_gen(input_filename, output_filename, key_gen, is_encrypt_data, key_file, version_no)
  File "components/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 638, in nvs_part_gen
    write_entry(nvs_obj, row["key"], row["type"], row["encoding"], row["value"])
  File "components/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 571, in write_entry
    nvs_instance.write_entry(key, value, encoding)
  File "components/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 503, in write_entry
    self.cur_page.write_varlen_data(key, value, encoding, self.namespace_idx,self)
  File "components/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 387, in write_varlen_data
    self.write_single_page_entry(entry_struct, data, datalen, data_entry_count, nvs_obj)
  File "components/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 328, in write_single_page_entry
    self.write_entry_to_buf(data, data_entry_count, nvs_obj)
  File "components/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 208, in write_entry_to_buf
    self.write_bitmaparray()
  File "components/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 109, in write_bitmaparray
    self.bitmap_array[byte_idx] &= mask
IndexError: array index out of range
```

I guess it wouldn't hurt to put larger objects in the test cases here https://github.com/espressif/esp-idf/tree/master/components/nvs_flash/nvs_partition_generator , that all have pretty small data sizes.

shivani-tipnis, 2019-02-20:
Hi,
The fix for this is currently under development. Should be available soon.

closed 2019-03-26 by igrr