espressif/esp-idf #3013: esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: native_ota_example/build/partitions_two_ota.bin

borch84, 2019-01-31:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (DevKitC): [v1]
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v3.3-beta1-316-g4b2feb316
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
     1.22.0-80-g6c4433a
- Operating System:     [linux]
- Power Supply:         [USB]

## Problem Description
When trying to flash my esp32 I got these message:
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: /home/borch/Documents/Robotica/esp/esp32-reedswitch/native_ota_example/build/partitions_two_ota.bin
/home/borch/Documents/Robotica/esp/esp-idf-master/esp-idf/components/esptool_py/Makefile.projbuild:62: recipe for target 'flash' failed
make: *** [flash] Error 2

### Expected Behavior
make should flash correctly the example. 

### Actual Behavior
make flash fails

### Steps to repropduce

1. git clone lastest version of IDF
2. copied the native_ota_example folder to another separate location
3. run make menuconfig and setup the wireless setting and file URI
4. run make builds the app successfully
5. make erase_flash executes successfully.
6. make flash fails with stated error.

### Code to reproduce this issue
This is the native_ota_example without modifications

```cpp
/* OTA example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
[... 189 lines cut ...]
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        // OTA app partition table has a smaller NVS partition size than the non-OTA
        // partition table. This size mismatch may cause NVS initialization to fail.
        // If this happens, we erase NVS partition and initialize NVS again.
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err );

    initialise_wifi();
    xTaskCreate(&ota_example_task, "ota_example_task", 8192, NULL, 5, NULL);
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2815805/sdkconfig.txt)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

KonstantinKondrashov, 2019-01-31:
Try increase this `CONFIG_PARTITION_TABLE_OFFSET`  option. 0x8000 -> 0x10000. It will help you. 

(or change `CONFIG_LOG_BOOTLOADER_LEVEL` from VERBOSE to INFO. It also resolves it)

borch84, 2019-01-31:
Hi @KonstantinKondrashov, I changed from VERBOSE to INFO and it was able to flash esp. I was struggling with project's configuration settings. I did not realize this overlapping was related to the verbosity level. How is it related to the partition table? Thank you.

KonstantinKondrashov, 2019-02-01:
Hi @borch84. 

Here is the output of the `make` command.

To flash all build output, run 'make flash' or:
python /home/virtpc/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio - flash_freq 40m --flash_size detect
0x1000 /home/virtpc/esp/esp-idf/examples/system/ota/native_ota_example/build/bootloader/bootloader.bin
0x10000 /home/virtpc/esp/esp-idf/examples/system/ota/native_ota_example/build/sntp.bin
0x8000 /home/virtpc/esp/esp-idf/examples/system/ota/native_ota_example/build/partitions_two_ota.bin

Here you can see that 3 bin files are written to the following addresses:
0x1000 bootloader.bin,
0x8000 partitions_two_ota.bin.
0x10000 native_ota_example.bin - application,

The overlap error occurs due to the fact that the bootloader did not fit into the allocated space between 0x1000-0x8000 and began to overlap with the next section of the partitions_two_ota, this happened because of the CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE option that adds a lot of code to the bootloader. Reducing the logging level allowed to flash it.

If you want to leave the logging level as VERBOSE. To do this, you need to move partitions_two_ota and native_ota_example. This is done simply by changing the CONFIG_PARTITION_TABLE_OFFSET option from 0x8000 -> 0x10000.

borch84, 2019-02-01:
Thank you @KonstantinKondrashov for the clarification. I will play with these VERBOSE level and the offset value around. Thanks.

closed 2019-02-01 by borch84