espressif/esp-idf #3016: http client send post request fail when wifi mode is WIFI_MODE_APSTA

NgJonesGitHub, 2019-02-01:
Hi,

I tried to use esp32 to run WIFI_MODE_APSTA. My application aims to run http server on softap side while regularly sending http post to a http server.  

I have tried http post task in WIFI_MODE_STA. And tried http server in WIFI_MODE_AP. Both application works fine. However, when I move forward to run both application in WIFI_MODE_APSTA. There is error when I start ssl connection with the server. Below is the err log:
-------------------
E (853714) esp-tls: couldn't get hostname for :my.test-server.com:
E (853714) esp-tls: Failed to open new connection
E (853714) TRANS_SSL: Failed to open a new connection
E (853714) HTTP_CLIENT: Connection failed, sock < 0
-------------------
Any idea what wrong with the program?

jitin17, 2019-02-12:
@NgJonesGitHub I suspect `esp_wifi_connect()` API is not being called after `esp_wifi_start()`. It would be helpful if you can post the wifi connection logic.

Alvin1Zhang, 2019-02-15:
@NgJonesGitHub Hi, would you help share the wifi connection logic? Thanks

NgJonesGitHub, 2019-02-15:
Sorry for the late reply.

I got a http post polling task which only send http post to target server when "CONNECTED_BIT" is 1. And at the first line of the while loop in that task there is a  xEventGroupWaitBits to wait until "CONNECTED_BIT" becomes 1.

And "CONNECTED_BIT" is set only at event SYSTEM_EVENT_STA_GOT_IP and clear only at event SYSTEM_EVENT_STA_DISCONNECTED. esp_wifi_connect() is called at event SYSTEM_EVENT_STA_START. 

In fact, this program works when WIFI_MODE is WIFI_MODE_STA.

If you need more information please let me know. Thank you.

ganeshlandge, 2019-07-23:
@NgJonesGitHub I tried the same on esp32 with WIFI_MODE_APSTA. It works fine for me. 
[https://github.com/ganeshlandge/apsta_mode](url)
Above is my logic. Let me know if there is an issue again.

mahavirj, 2019-07-29:
@NgJonesGitHub Can you please to try above example code and let us know the results?

mahavirj, 2019-08-01:
Closing due to lack of response, please do try above example and feel free to notify if issues still exists.

closed 2019-08-01 by mahavirj