espressif/esp-idf #3017: The variable is never updated on my phone when i read it

Marc-Aurele, 2019-02-01:
Hi,

I'm facing an issue when i want to read data from the device on my phone. I inspire from the hid demo example and i want to add a custom characteritic. In the gatts_profile_event_handler i added:
`        case ESP_GATTS_READ_EVT:
            ESP_LOGI(HID_LE_PRF_TAG, "ESP_GATTS_READ_EVT, conn_id %d, trans_id %d, handle %d\n", param->read.conn_id, param->read.trans_id, param->read.handle);
            if (param->add_attr_tab.svc_uuid.uuid.uuid16 == CUSTOM_APP_ID) {
                esp_ble_gatts_set_attr_value(param->read.handle, sizeof(my_time), my_time);
                ESP_LOGI(HID_LE_PRF_TAG, "READ EVENT: %d\n",my_time);
            }
            break;`
But the value read on the phone never change whereas "my_time" is supposed to be updated in background task.

Here is my code : 

`// Copyright 2017-2018 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#include "hidd_le_prf_int.h"
#include <string.h>
#include "esp_log.h"
#include "utils.h"

/// characteristic presentation information
struct prf_char_pres_fmt
{
    /// Unit (The Unit is a UUID)
    uint16_t unit;
    /// Description
    uint16_t description;
    /// Format
    uint8_t format;
    /// Exponent
    uint8_t exponent;
    /// Name space
    uint8_t name_space;
};

// HID report mapping table
static hid_report_map_t hid_rpt_map[HID_NUM_REPORTS];

// HID Report Map characteristic value
// Keyboard report descriptor (using format for Boot interface descriptor)
static const uint8_t hidReportMap[] = {
    0x05, 0x01,  // Usage Page (Generic Desktop)
    0x09, 0x02,  // Usage (Mouse)
    0xA1, 0x01,  // Collection (Application)
    0x85, 0x01,  // Report Id (1)
    0x09, 0x01,  //   Usage (Pointer)
    0xA1, 0x00,  //   Collection (Physical)
    0x05, 0x09,  //     Usage Page (Buttons)
    0x19, 0x01,  //     Usage Minimum (01) - Button 1
    0x29, 0x03,  //     Usage Maximum (03) - Button 3
    0x15, 0x00,  //     Logical Minimum (0)
    0x25, 0x01,  //     Logical Maximum (1)
    0x75, 0x01,  //     Report Size (1)
    0x95, 0x03,  //     Report Count (3)
    0x81, 0x02,  //     Input (Data, Variable, Absolute) - Button states
    0x75, 0x05,  //     Report Size (5)
    0x95, 0x01,  //     Report Count (1)
[... 761 lines cut ...]
      hid_rpt_map[5].type = hidReportRefLedOut[1];
      hid_rpt_map[5].handle = hidd_le_env.hidd_inst.att_tbl[HIDD_LE_IDX_BOOT_KB_OUT_REPORT_VAL];
      hid_rpt_map[5].cccdHandle = 0;
      hid_rpt_map[5].mode = HID_PROTOCOL_MODE_BOOT;

      // Boot mouse input report
      // Use same ID and type as mouse input report
      hid_rpt_map[6].id = hidReportRefMouseIn[0];
      hid_rpt_map[6].type = hidReportRefMouseIn[1];
      hid_rpt_map[6].handle = hidd_le_env.hidd_inst.att_tbl[HIDD_LE_IDX_BOOT_MOUSE_IN_REPORT_VAL];
      hid_rpt_map[6].cccdHandle = 0;
      hid_rpt_map[6].mode = HID_PROTOCOL_MODE_BOOT;

      // Feature report
      hid_rpt_map[7].id = hidReportRefFeature[0];
      hid_rpt_map[7].type = hidReportRefFeature[1];
      hid_rpt_map[7].handle = hidd_le_env.hidd_inst.att_tbl[HIDD_LE_IDX_REPORT_VAL];
      hid_rpt_map[7].cccdHandle = 0;
      hid_rpt_map[7].mode = HID_PROTOCOL_MODE_REPORT;

  // Setup report ID map
  hid_dev_register_reports(HID_NUM_REPORTS, hid_rpt_map);
}

`

Marc-Aurele, 2019-02-02:
Hello, 

Thank you very much for the reply. What do you mean by setting the value before read ? Actually i can read the first value when the variable has been intialized but then when i launch a read operation onto this variable from my phone, the variable is never updated with its last value that has been changed in a background task.

Regards,

Aurélien

Marc-Aurele, 2019-02-04:
> @Marc-Aurele, please set attribute value before read.

Weijian-Espressif, 2019-02-11:
please set the attribute value before your phone read.  When you create attr_db with `ESP_GATT_AUTO_RSP`, ESP32 automatically sends a response. when you get `ESP_GATTS_READ_EVT`, it is too late to set attribute value.
You have two ways:
1.  set the attribute value before your phone read
2.  please create  `attr_db` with `ESP_GATT_RSP_BY_APP`, when you get `ESP_GATTS_READ_EVT`, send a response using `esp_ble_gatts_send_rsp()` with your `my_time`.

Marc-Aurele, 2019-02-11:
Hi,

Thank you very much for your help. All is working fine now ;)

Regards,

Aurélien

closed 2019-02-14 by Alvin1Zhang