espressif/esp-idf #3018: esp-idf v3.2

sjoteppagol, 2019-02-01:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|
- IDF version (run ``git describe --tags`` to find it): 
    v3.2-beta1
- Build System:         [Make|
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows|
- Power Supply:         [|external 5V

[setnv_crash_backtrace.txt](https://github.com/espressif/esp-idf/files/2821764/setnv_crash_backtrace.txt)
[esp-idf-version.txt](https://github.com/espressif/esp-idf/files/2821765/esp-idf-version.txt)

## Problem Description

//Detailed problem description goes here.
We have ported our application to beta version and we are seeing multiple crashes.
**CORRUPT HEAP: multi_heap.c:429 detected at 0x3fff8e74
abort() was called at PC 0x4009640b on core 1**

igrr, 2019-02-26:
@sjoteppagol I have tried stress testing the functions mentioned in the backtrace, but haven't been able to reproduce. Could you please follow the steps listed in https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/heap_debug.html#finding-heap-corruption to isolate the part of the program which corrupts heap?

projectgus, 2019-03-12:
HI @sjoteppagol ,

Did you get any more information from following the heap debugging steps?

Angus

sjoteppagol, 2019-03-13:
Hello @igrr  and @projectgus  Sorry for late response.
I do observed the above mentioned issue in **v3.2-beta1** but recently moved to latest IDF version **v3.3-dev-264-g8a69ffc36** and you know what!, surprisingly I have not seen this issue as of now. 

Thank you
Shridhar

closed 2019-03-13 by sjoteppagol