espressif/esp-idf #3019: Garbled UART when using Power Management

BrianPugh, 2019-02-01:
I'm currently running on `master`. My problem is exactly described in #1729 where support for power management is enabled in menuconfig and Enable dynamic frequency scaling (DFS) at startup is NOT enabled. I've tested this with 160 and 240 CPU freq, as well as 1500000 and 921600 baudrates.

igrr, 2019-02-14:
@BrianPugh Do you refer to console UART output (ESP_LOG and printf), or are you using the UART driver? In the latter case make sure that [`use_ref_tick`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/uart.html#_CPPv2N13uart_config_t12use_ref_tickE) is set in `uart_config_t` ([docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/power_management.html#dynamic-frequency-scaling-and-peripheral-drivers)).

Finally, please add the IDF commit ID, sdkconfig, and some code to reproduce the issue, as requested in the issue template. Thanks.

BrianPugh, 2019-02-24:
Commit: Latest Master 1aa8e8d38ebe688c391359b5962707c1d135d07e

Just run the default "hello world" example with the following non-default menu-config:

Enable Power Management
CPU Freq -> 240MHz
Change the UART console frequency to something higher, like 921600.

This produces garbled output; it works fine at lower UART console frequencies, like 115200.

igrr, 2019-03-01:
Right, when DFS is enabled, UART will be clocked from REF_TICK clock, which is 1MHz. Normally it is clocked from APB, which is 80MHz. With 1MHz base clock, i think it is not possible to obtain accurate 921600 baud rate, hence the garbled output.

BrianPugh, 2019-03-01:
Reasonable and makes sense; could we add this to the power management documentation?

closed 2019-12-03 by BrianPugh