espressif/esp-idf #3021: How best to generate OTA certificates required to connect to AWS securely

wegunterjr, 2019-02-02:
Code esp-df v3.1
esp32 wrover 4MB PSRAM

I have tried to generate the certs as show in this example:  
https://github.com/RadialDevGroup/esp32-ota-https-example

which uses the following to generate the certs:
`openssl s_client -showcerts -connect <your-server-dns>:443 < /dev/null`
I still get an error using the simple_ota_task when trying to connect.

Thoughts? Suggestions?
Is it memory?  I connect to wifi first (mbedTLS seems to work), but then fails when trying to connect to do the OTA.
Is there a time delay recommended?

mahavirj, 2019-02-04:
@wegunterjr ESP-IDF provides sample applications to connect with `aws-iot`, please find them at https://github.com/espressif/esp-idf/tree/master/examples/protocols/aws_iot (Setup requires thing certificate and thing private key. Server certificate is already provided in examples)

Regarding `OTA` feature, ESP-IDF provides following examples https://github.com/espressif/esp-idf/tree/master/examples/system/ota (requires setting up HTTPS enabled server instance)

If you run into any problem with these examples, please feel free to raise new issue with information as per github issue template.

closed 2019-02-04 by mahavirj

wegunterjr, 2019-02-04:
it just wasn't clear - if I am using generated certs already to setup the Shadow and Connecto to AWS, do i need a special cert to do OTA?  Where should the firmware reside?

mahavirj, 2019-02-04:
@wegunterjr If you host firmware in S3 bucket in AWS, then no additional certificate should be required. You may change `url` and `server-cert` in example code at https://github.com/espressif/esp-idf/blob/master/examples/system/ota/simple_ota_example/main/simple_ota_example.c#L116. Let me know if this makes sense or you run into any issues with this.

wegunterjr, 2019-02-04:
The challenge is that when we host it in the s3 bucket, it doesn't appear
to return a https URL.
1. Do we need to do anything special when hosting it?  We are already
secure and using port 443
2.  Which certificate should i use as the server_cert?

Thanks for your help.

Walter Gunter
"The world has the habit of making room for the man whose actions show that
he knows where he is going.”  - Napoleon Hill

wegunterjr, 2019-02-06:
@mahavirj I am still not clear on how using the AWS thing certs (we are setting up things) and how that security can also be used with the OTA.
I got the cert from my server team and was able to download the OTA file (it failed, but that's another story), but at least it was able to connect to the url and do the download.
But, it is public s3? or can an s3 bucket be behind the other certs we have to generate to allow a secure device connection - seems counter intuitive, but i defer to network security geniuses...

mahavirj, 2019-02-06:
@wegunterjr 

For trying out OTA based on url from S3 bucket:
 - Create `s3` bucket, upload file. 
 - Under `s3` bucket overview section, make it public (it will be accessible to anyone with public url)
 - For getting server certificate run following command,

```
openssl s_client -showcerts -connect s3.amazonaws.com:443 < /dev/null
```
  - (Copy last certificate from output of above command, marked with `-----BEGIN CERTIFICATE-----` and `-----END CERTIFICATE-----` tags)

For production deployments you may not want this file to be publicly accessible, and hence you may create signed url [refer](https://docs.aws.amazon.com/AmazonCloudFront/latest/DeveloperGuide/private-content-signed-urls.html) or authorized rest request [refer](https://docs.aws.amazon.com/AmazonS3/latest/dev/RESTAuthentication.html) for file uploaded on `s3` bucket. 

Some additional points:
 - Thing certificate (required for mutual authentication) has no role to play here, since here we are only interested in validating identity of server from which we will be downloading update file. 
 - Ideally server certificate (`aws-root-ca.pem`) for AWS IoT should also work for `s3` but it seems that `s3` service is using different one, hence additional step as mentioned above will be required to retrieve that.

(Please note that we recommend use of esp32.com forum for `how-to` questions)

ahtashambaig, 2020-04-20:
mahavirj Sir can u guide me, what will be the url for accessing S3 bucket , also i am in production , how can i use code signing for private access bucket ota update. Kindly brief

chegewara, 2020-04-20:
https://docs.aws.amazon.com/freertos/latest/userguide/ota-code-sign-cert-esp.html
https://medium.com/the-esp-journal/esp32-ota-updates-amazon-freertos-9ffbbb8f4784

ahtashambaig, 2020-04-20:
Dont mind me asking i am a noob and i already know how to create code signing certificate... the issue is as mentioned above for implementing aws ota i just need to provide s3 bucket link and server cert (Performed this step) ,also made my binary file public ... on performing it gets connected to s3 bucket and perform which at end gets failed . i am working on a production where s3 ota update needs to be private i can create code signing certificate but no idea how to implement update
thanks and regards

ahtashambaig, 2020-04-20:
Forgot to mention i am using nina-w102 board for production where it is getting failed but on esp32 devkit it is working fine with public access

chegewara, 2020-04-20:
In that case its highly advised to start new issue, maybe even in this github repository:
https://github.com/arduino/nina-fw

ahtashambaig, 2020-04-20:
okai sure thanks , but can u help me how can i update using code signing certificate and my binary file.... i know the procedure is using creating a Job but how can i use that job with esp-idf

ahtashambaig, 2020-04-20:
can u provide me with any code  sample  for implementing aws-ota update in esp-idf

chegewara, 2020-04-20:
Sorry, but i dont know if esp-idf is supporting AWS OTA update, i am using AWSFreeRTOS with AWS OTA. Maybe @mahavirj will help you with that

AdityaHPatwardhan, 2020-05-11:
@ahtashambaig Is your issue resolved. The instructions provided by @mahavirj properly address any requirements for hosting `OTA` on a private `S3` bucket. Also, source code is same as [advanced_https_ota](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/advanced_https_ota), you will only have to change some menuconfig options as follows -
* Update the `OTA URL` with your own `S3` bucket `URL`.
you can change it with `menuconfig ->Example Configuration -> Firmware Upgrade URL`. 

* Update [ca_cert](https://github.com/espressif/esp-idf/blob/master/examples/system/ota/advanced_https_ota/server_certs/ca_cert.pem) with the cert obtained in the instructions provided [here](https://github.com/espressif/esp-idf/issues/3021#issuecomment-460917118)

I hope this resolves your query

listout, 2022-02-11:
@AdityaHPatwardhan I tried @mahavirj instruction and still can't download the binary from s3 bucket, matter of fact I cant do it from any https server. The readme of the ota suggests uploading it to something like github for test, that too is not working. I've managed to get it working only when running a https server locally, like suggested in the OTA examples root readme. I've tried downloading the binaries using curl and it works fine.

My error message is:

```
W (4641) wifi:<ba-add>idx:0 (ifx:0, e8:11:32:00:57:44), tid:0, ssn:1, winSize:64
E (5361) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (5361) esp-tls-mbedtls: Failed to verify peer certificate!
E (5361) esp-tls: Failed to open new connection
E (5371) TRANSPORT_BASE: Failed to open a new connection
E (5381) HTTP_CLIENT: Connection failed, sock < 0
E (5381) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (5391) esp_https_ota: Failed to establish HTTP connection
E (5401) advanced_https_ota_example: ESP HTTPS OTA Begin failed
```

I'm on  `ESP-IDF v5.0-dev-1536-g2522277dfe`