espressif/esp-idf #3022: ulp_main.h seems to don't exist in ESP-IDF

sch0bert, 2019-02-02:
Hello,

I'm trying to compile this example and seems that ulp_main.h does not exist in esp-idf or at least I can't find it:

https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/examples/system/ulp_adc/main/ulp_adc_example_main.c#L22

I think by mistake it was took from this example:

https://github.com/duff2013/ulptool/tree/master/ulp_examples/ulp_hall_sensor

Correct me if I'm wrong.

Regards,
Pascal.

mahavirj, 2019-02-04:
@sch0bert `ulp_main.h` gets auto-generated during build step. Reference build output which shows this:
```
[1/7] Generating esp32.ulp.ld
[2/7] Generating pulse_cnt.ulp.S
[3/7] Building ASM object CMakeFiles/ulp_main.dir/pulse_cnt.ulp.S.obj
[4/7] Linking ASM executable ulp_main
[5/7] Generating ulp_main.bin
[6/7] Generating ulp_main.sym
[7/7] Generating ulp_main.ld, ulp_main.h
```

closed 2019-02-04 by mahavirj

sch0bert, 2019-02-05:
Thanks for the explanation!