espressif/esp-idf #3024: OTA, components/bt/bt.c:1200 and Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)

an-erd, 2019-02-02:
## Environment

- Development Kit:      M5Stack-Fire
- IDF version (run ``git describe --tags`` to find it): ESP-IDF:          v3.3-beta1-dirty
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- Power Supply:         USB and/or Battery

## Problem Description

Getting 
`Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)`
when starting OTA. The code used for OTA is from `https://github.com/espressif/esp-iot-solution/tree/master/components/general/ota`, also Wifi, BLE, SPIRAM, TFT display on SPI and SD card on SPI are used. 

I get the same issue when using esp-idf with arduino as a component and using ArduinoOTA or the component in esp-iot-solution.

Overall code is in [here](https://github.com/an-erd/disPOD/tree/featurebranch).

Issue appears when calling `https://github.com/an-erd/disPOD/blob/featurebranch/main/main.cpp#L144`

### Code to reproduce this issue

I wasn't able to extract a small piece of code to reproduce. esp-idf with ArduinoOTA or with esp-iot-solution iot-ota is working fine. 

I don't know to best start further debugging. Any ideas?

## Debug Logs

```
I (109033) DISPOD: s_do_ota(): free heap size before ota: 3866840
I (109034) DISPOD: ota_task(): test mutex
I (109034) OTA: ota starting
I (109035) DISPOD: OTA progress: -1 %
I (109040) OTA: Running partition type 0 subtype 16 (offset 0x00010000)
I (109041) OTA: Erasing to partition subtype 17 at offset 0x650000
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400877e8  PS      : 0x00060034  A0      : 0x8008b670  A1      : 0x3ffbe210
0x400877e8: btdm_lpcycles_2_us at C:/msys32/home/AKAEM/esp/esp-idf/components/bt/bt.c:1200

A2      : 0x0004c478  A3      : 0x00000001  A4      : 0x3ffc240d  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3f82c598  A9      : 0xbad00bad
A10     : 0x00000002  A11     : 0x3ffb7ec0  A12     : 0x80097f30  A13     : 0x3ffb7ec0
A14     : 0x00000000  A15     : 0x400864b4  SAR     : 0x00000012  EXCCAUSE: 0x00000007
0x400864b4: spi_flash_op_block_func at C:/msys32/home/AKAEM/esp/esp-idf/components/spi_flash/cache_utils.c:71

EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x4000bfe2  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400877e8
0x400877e8: btdm_lpcycles_2_us at C:/msys32/home/AKAEM/esp/esp-idf/components/bt/bt.c:1200

Backtrace: 0x400877e8:0x3ffbe210 0x4008b66d:0x3ffbe230 0x4008be4f:0x3ffbe250 0x4008bba6:0x3ffbe270 0x4008c1af:0x3ffbe290 0x40084f09:0x3ffbe2b0 0x400864e0:0x00000000
0x400877e8: btdm_lpcycles_2_us at C:/msys32/home/AKAEM/esp/esp-idf/components/bt/bt.c:1200

0x4008b66d: r_rwbt_sleep_wakeup at ??:?

0x4008be4f: r_rwip_wakeup at ??:?

0x4008bba6: r_rwbt_isr at ??:?

0x4008c1af: r_rwbtdm_isr_wrapper at intc.c:?

0x40084f09: _xt_lowint1 at C:/msys32/home/AKAEM/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

0x400864e0: spi_flash_op_block_func at C:/msys32/home/AKAEM/esp/esp-idf/components/spi_flash/cache_utils.c:81
```

overall log file with log options set to "I"

[... 144 lines cut ...]
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x4000bfe2  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400877e8
0x400877e8: btdm_lpcycles_2_us at C:/msys32/home/AKAEM/esp/esp-idf/components/bt/bt.c:1200

Backtrace: 0x400877e8:0x3ffbe210 0x4008b66d:0x3ffbe230 0x4008be4f:0x3ffbe250 0x4008bba6:0x3ffbe270 0x4008c1af:0x3ffbe290 0x40084f09:0x3ffbe2b0 0x400864e0:0x00000000
0x400877e8: btdm_lpcycles_2_us at C:/msys32/home/AKAEM/esp/esp-idf/components/bt/bt.c:1200

0x4008b66d: r_rwbt_sleep_wakeup at ??:?

0x4008be4f: r_rwip_wakeup at ??:?

0x4008bba6: r_rwbt_isr at ??:?

0x4008c1af: r_rwbtdm_isr_wrapper at intc.c:?

0x40084f09: _xt_lowint1 at C:/msys32/home/AKAEM/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

0x400864e0: spi_flash_op_block_func at C:/msys32/home/AKAEM/esp/esp-idf/components/spi_flash/cache_utils.c:81

CPU halted.
```

## Other items if possible
[sdkconfig](https://gist.github.com/an-erd/e808fbaec27983cda5e9cc143423481f)

mywang-espressif, 2019-02-11:
Hi @an-erd , we had a bugfix for this issue: commit(SHA: 6cb37ecc) on master branch with another related  commit(SHA: 9eb22877) followed. Please have a try.

Weijian-Espressif, 2019-02-27:
@an-erd,
Has your problem been solved?

an-erd, 2019-02-27:
@Weijian-Espressif Sorry for the delay, I check on the weekend and give feedback.

an-erd, 2019-02-28:
I udated to `v3.3-beta1-504-g6c49f1924` (git describe --tags) and did `submodule update --init`, and recompiled everything.

I'm still experiencing an similar (same?) issue, complete logs below.

```

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6292
load:0x40078000,len:10248
load:0x40080400,len:6676
entry 0x40080764
I (28) boot: ESP-IDF v3.3-beta1-dirty 2nd stage bootloader
I (28) boot: compile time 08:21:28
I (28) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 16MB
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (64) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (72) boot:  2 app0             OTA app          00 10 00010000 00640000
I (79) boot:  3 app1             OTA app          00 11 00650000 00640000
I (87) boot:  4 eeprom           Unknown data     01 99 00c90000 00001000
I (94) boot:  5 spiffs           Unknown data     01 82 00c91000 0036f000
I (102) boot: End of partition table
I (106) esp_image: segment 0: paddr=0x00650020 vaddr=0x3f400020 size=0x3cd8c (249228) map
I (202) esp_image: segment 1: paddr=0x0068cdb4 vaddr=0x3ffbdb60 size=0x0325c ( 12892) load
I (207) esp_image: segment 2: paddr=0x00690018 vaddr=0x400d0018 size=0xc9488 (824456) map
0x400d0018: _flash_cache_start at ??:?

I (497) esp_image: segment 3: paddr=0x007594a8 vaddr=0x3ffc0dbc size=0x00554 (  1364) load
I (498) esp_image: segment 4: paddr=0x00759a04 vaddr=0x3ffc1310 size=0x00000 (     0) load
I (504) esp_image: segment 5: paddr=0x00759a0c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/AKAEM/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (514) esp_image: segment 6: paddr=0x00759e14 vaddr=0x40080400 size=0x1edb8 (126392) load
I (575) esp_image: segment 7: paddr=0x00778bd4 vaddr=0x400c0000 size=0x0006c (   108) load
I (575) esp_image: segment 8: paddr=0x00778c48 vaddr=0x50000000 size=0x00000 (     0) load
I (581) esp_image: segment 9: paddr=0x00778c50 vaddr=0x50000000 size=0x00000 (     0) load
I (608) boot: Loaded app from partition at offset 0x650000
I (608) boot: Disabling RNG early entropy source...
I (610) spiram: Found 32MBit SPI RAM device
I (613) spiram: SPI RAM mode: flash 40m sram 40m
I (618) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (626) cpu_start: Pro cpu up.
I (629) cpu_start: Application information:
I (634) cpu_start: Project name:     dispod
I (639) cpu_start: App version:      0.31
I (644) cpu_start: Compile time:     19:08:39
I (649) cpu_start: Compile date:     Feb  9 2019
I (654) cpu_start: ESP-IDF:          v3.3-beta1
I (659) cpu_start: Starting app cpu, entry point is 0x4008159c
0x4008159c: start_cpu0_default at C:/msys32/home/AKAEM/esp/esp-idf/components/esp32/cpu_start.c:377 (discriminator 1)

I (0) cpu_start: App cpu up.
I (1560) spiram: SPI SRAM memory test OK
I (1584) heap_init: Initializing. RAM available for dynamic allocation:
I (1584) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1586) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1592) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1599) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1605) heap_init: At 3FFC3EB0 len 0001C150 (112 KiB): DRAM
I (1611) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[... 75 lines cut ...]
A6      : 0x3ffc3060  A7      : 0x000007d0  A8      : 0x3f82c598  A9      : 0xbad00bad
A10     : 0x0000000a  A11     : 0x3ffda650  A12     : 0x800fc8b1  A13     : 0x3ffda650
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x4000bfe2  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400878cc
0x400878cc: spiWriteShortNL at C:/msys32/home/AKAEM/esp/disPOD/components/arduino/cores/esp32/esp32-hal-spi.c:543

Backtrace: 0x400878cc:0x3ffbe210 0x4008b745:0x3ffbe230 0x4008bf27:0x3ffbe250 0x4008bc7e:0x3ffbe270 0x4008c287:0x3ffbe290 0x40084fed:0x3ffbe2b0 0x400865c2:0x00000000
0x400878cc: spiWriteShortNL at C:/msys32/home/AKAEM/esp/disPOD/components/arduino/cores/esp32/esp32-hal-spi.c:543

0x4008b745: ble_dfl_txpwr_get_inter at ??:?

0x4008bf27: r_rwbt_isr at ??:?

0x4008bc7e: r_rwbt_sleep_wakeup at ??:?

0x4008c287: r_rwbt_isr at ??:?

0x40084fed: cache_sram_mmu_set at C:/msys32/home/AKAEM/esp/esp-idf/components/esp32/cache_sram_mmu.c:99 (discriminator 1)

0x400865c2: spi_flash_read at C:/msys32/home/AKAEM/esp/esp-idf/components/spi_flash/flash_ops.c:529 (discriminator 4)

CPU halted.
```

Weijian-Espressif, 2019-03-01:
@an-erd,
I want to know if it is caused by sleep. Please disable modem sleep in menuconfig and try.

                                                                                                                                      
   Location:                                                                                                                                                              -> Component config -> Bluetooth   -> Bluetooth controller -> MODEM SLEEP Options

an-erd, 2019-03-01:
@Weijian-Espressif 
I disabled the modem sleep option and did 10 tests. No problem anymore!

What is the recommended way to do OTA and have modem sleep?

mywang-espressif, 2019-03-01:
Hi @an-erd,   ESP-IDF v3.3 beta is not verified and tested, it is not ready for customers and please do not use this. Please use ESP-IDF release version v3.1.3 instead.

closed 2019-03-15 by Alvin1Zhang