espressif/esp-idf #3026: SDKCONFIG variables marked as error in Eclipse IDE

carlessole, 2019-02-04:
I'm using Ubuntu and after following all the steps given in [Build and Flash with Eclipse IDE](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/eclipse-setup.html) everything is OK and I'm able to build and flash my projects properly.
But there is an issue that is bothering me. The case is that eclipse does not find all the "variables" defined in "sdkconfig" file and so, shows as if my project is full of errors:
[image]

Is there any way to solve that?
Thanks!

igrr, 2019-02-07:
Can you please right click the source file in project tree, and use "Create parser log file" option (somewhere under Index), then attach the file here?

Also, if you add `#include "sdkconfig.h"` to your source file, and then control-click on the include line, does it open the correct sdkconfig.h file (of the project you are working with)?

carlessole, 2019-02-11:
Thanks @igrr .
I solve that issue adding this include path:

[image]

closed 2019-02-11 by carlessole