espressif/esp-idf #3027: （ESP32/ESP-IDF/platformIO）ESP_ERR_NVS_NEW_VERSION_FOUND

omkw, 2019-02-05:
Hi.
I am trouble with error about nvs. (ESP_ERR_NVS_NEW_VERSION_FOUND)
I looked for a solution, but I could not find it.
attached figure shows details.
please help me.

[enviroment info] 
OS:Windows10
IDE: VSC with PlatformIO (Home 2.0.0/Core 3.6.4)
  platform = espressif32( 1.6.0)
  framework = espidf
  board = esp32dev

[image]

AmrutaCh, 2019-02-05:
Check your esp-idf version. In version 3.1 and above, the examples already include proper nvs initialization code. (not sure about older versions.)
You can add following code at the start of `app_main()`

```
    esp_err_t err = nvs_flash_init();
    if ((ESP_ERR_NVS_NO_FREE_PAGES == err) || (ESP_ERR_NVS_NEW_VERSION_FOUND == err))
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);
```

omkw, 2019-02-07:
it works fine!
thanks so much.

in short, current version of ESP-IDF framework served by Espressif 32(v1.6.0) contains bug I posted. right?

AmrutaCh, 2019-02-07:
1. v1.6.0 is pretty old
2. Its not bug in SDK but it user code. SDK is already returning the error-code. Its up-to the application, how to handle it.

Alvin1Zhang, 2019-02-15:
@AmrutaCh Thanks for explaining. @omkw I assume your issue has been resolved, feel free to reopen if your issue still exists. Thanks

closed 2019-02-15 by Alvin1Zhang