espressif/esp-idf #3029: esp_http_client: how to catch TLS errors?

olegantonyan, 2019-02-06:
It's possible to specify `cert_pem` to verify certificates, but if there's something wrong with certicfiate I need to know that.

This is printed to a console:
```E (349004) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (349005) esp-tls: Failed to verify peer certificate!
I (349005) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA
E (349015) esp-tls: Failed to open new connection
E (349017) TRANS_SSL: Failed to open a new connection
E (349021) HTTP_CLIENT: Connection failed, sock < 0
```
But I need to know this programmatically. Via event, for example 
```
esp_err_t http_event_handle(esp_http_client_event_t *evt) {
  switch(evt->event_id) {
    case HTTP_EVENT_TLS_HANDSHAKE_ERROR:

```
something like this.

But right now esp_http_client only fires HTTP_EVENT_DISCONNECTED event and that's it. No way to distinguish between TLS and other types of errors

jitin17, 2019-02-07:
@olegantonyan Thanks for bringing this to our notice. We'll try to do something in this regard.

david-cermak, 2019-10-10:
Resolved in https://github.com/espressif/esp-idf/commit/2972f96982551c84717092e516fbd26f6c8d27dc (http client example updated to read reported errors: https://github.com/espressif/esp-idf/commit/a001eb39bf43849d6171d759ab9c925e5ce23a10)

Alvin1Zhang, 2019-10-10:
@david-cermak Thanks for the updates and @olegantonyan thanks for reporting. Feel free to reopen if the issue still exists. Thanks.

closed 2019-10-10 by Alvin1Zhang