espressif/esp-idf #3030: WPS not terminating since ESP-IDF 3.1.2

Roody75, 2019-02-06:
If WPS in station mode ist started in ESP32 first and after that in the Router (nevertheless if its using PIN or PBC method) WPS never succeeds and is never stopped anymore. If WPS is started at the router first (PBC) it works, though WPS PIN is not possible this way. I tested this with ESP-IDF 3.1.1 and it is working there in all ways so its a bug introduced by 3.1.2.

TimXia, 2019-02-15:
@Roody75 Please update IDF to 3.1.3. The bug has been fixed.

Alvin1Zhang, 2019-03-14:
@Roody75 Would you help share if any updates for the issue? Thanks.

Alvin1Zhang, 2019-05-16:
@Roody75  Thanks for reporting the issue. Would you help share if any updates for the issue? Has your issue been fixed? Thanks.

Alvin1Zhang, 2019-10-11:
@Roody75 Thanks for reporting, feel free to reopen if you still have the issue, thanks.

closed 2019-10-11 by Alvin1Zhang