espressif/esp-idf #3031: BLE Static passkey doesn't work properly with passkey of less than 6 digits

LaurentLouf, 2019-02-06:
## Environment

- Development Kit:      
- Kit version (for WroverKit/PicoKit/DevKitC):
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v3.1.2
- Build System:  Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB

## Problem Description

Trying to use `ESP_BLE_SM_SET_STATIC_PASSKEY` with 4 digits doesn't work for authentication unless I add two zeroes before the pin when entering it on my phone. 

### Expected Behavior

If it set the passkey as follow : 
```
uint32_t passkey = 1427;
esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passkey, sizeof(uint32_t));
```
I should be able to just type "1427" on my smartphone when prompted for the pin.

### Actual Behavior

When entering "1427" on my phone, I get the following (I'm using nkolban's BLE library)

> I (30947) BLEDevice: ESP_GAP_BLE_PASSKEY_NOTIF_EVT
> I (30947) BLEDevice: passKey = 1427
> I (30947) BLESecurity: On passkey Notify number:1427
> W (44307) BT_SMP: Commitment check fails
> 
> I (44317) BLEDevice: ESP_GAP_BLE_AUTH_CMPL_EVT
> W (44317) MoonaBLESecurity: Authentication of the client is a failure
> W (47307) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB
> 
> [I] BLE device disconnected.
> E (47327) BT_BTM: Device not found

But then if I enter "001427", I get the following : 

> I (144817) BLEDevice: ESP_GAP_BLE_PASSKEY_NOTIF_EVT
> I (144817) BLEDevice: passKey = 1427
> I (144817) MoonaBLESecurity: On passkey Notify number:1427
> W (152137) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
> I (152167) BLEDevice: ESP_GAP_BLE_KEY_EVT
> I (152167) BLEDevice: key type = ESP_LE_KEY_LENC
> I (152167) BLEDevice: ESP_GAP_BLE_KEY_EVT
> I (152167) BLEDevice: key type = ESP_LE_KEY_PENC
> I (152177) BLEDevice: ESP_GAP_BLE_KEY_EVT
> I (152177) BLEDevice: key type = ESP_LE_KEY_LID
> I (152187) BLEDevice: ESP_GAP_BLE_KEY_EVT
> I (152197) BLEDevice: key type = ESP_LE_KEY_LCSRK
> I (152347) BLEDevice: ESP_GAP_BLE_KEY_EVT
> I (152357) BLEDevice: key type = ESP_LE_KEY_PID
> I (152357) BLEDevice: ESP_GAP_BLE_AUTH_CMPL_EVT
> I (152357) BLESecurity: Authentication of the client is a success

### Steps to repropduce

1. Set a pin with only 4 digits
2. Enter those 4 digits when prompted to do so on your phone -> failure
3. Enter those 4 digits prefixed with "00" when prompted to do so on your phone -> success

### Code to reproduce this issue
That's the part where the security details are set up : 

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
BLESecurity* pSecurity = new BLESecurity();
pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
pSecurity->setCapability(ESP_IO_CAP_OUT);
pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
uint32_t passkey = 1427;
esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passkey, sizeof(uint32_t));

```

chegewara, 2019-02-06:
Its already been answered in other issue. Even if your passkey is 4 digit, like 1427, in most peer devices you probably have to prefix pin with 00, because pin has to be 6 digit long.

LaurentLouf, 2019-02-06:
If you could point out this issue, I haven't found any referencing the same problem going back at the date of the release of v3.1.2

chegewara, 2019-02-06:
https://github.com/espressif/esp-idf/issues/2322#issuecomment-421214524

Weijian-Espressif, 2019-02-27:
@chegewara,
thanks for your help.
@LaurentLouf ,
Do you have any other questions?

Alvin1Zhang, 2019-03-13:
@LaurentLouf Feel free to reopen if you still have the issue, thanks.

closed 2019-03-13 by Alvin1Zhang

ElenaG391, 2025-02-25:
Any update on that? Can we use 4digits instead 6 digits?